pbatard/rufus #519: Creating a WindowsToGo fails when using HDD in FANTEC DB-ALU3e-6G

MagicAndre1981, 2015-05-30:
When I use a external HDD in the FANTEC DB-ALU3e-6G (http://www.fantec.de/produkte/speicherprodukte/festplattengehaeuse/35-zoll-festplattengehaeuse/produkt/details/artikel/1693_fantec_db_alu3e_6g/) external enclosure, I can't create a WindowsToGo on a HDD in it. Rufus now always complains about that the drive is in use.

Windows shows that it uses the ASmedia ASMT1153E chip (ASmedia ASMT1153E SCSI Disk Device).

If I use an older revision of the external enclosure without the the 6G Rufus works fine (AS2105). The 6G version supports UASP and so the Default UASP driver of Windows 8.x/10 seams to cause the "in use" issue.

Any chance to fix this?

pbatard, 2015-05-31:
Rufus had been supporting UASP for some time, and it should work just fine with the standard Windows  UASP driver.

How about you provide a __full__ log from what happens when you try to create a bootable USB with the 6G?
Also, having the details of how Windows sees your enclosure connected in Device Manager would help. Especially, can you take a look at what I requested [here](https://github.com/pbatard/rufus/issues/513#issuecomment-105062950) and provide me with both a screenshot of your device, when [viewed by connections](https://github.com/pbatard/rufus/issues/513#issuecomment-105514229), as well as the `Device Stack` property?

MagicAndre1981, 2015-05-31:
Here is the log:

http://pastebin.com/UD5py6hq

pbatard, 2015-05-31:
Thanks. My guess is that you have an executable (probably a security application, as these seem to be causing [loads of exclusive access issues](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)) that keeps exclusive access of what it sees as a system drive.

Because UASP effectively makes an USB drive looks like an SCSI/SATA one, if the developers of such an application process botched their detection of system drives and used something like _"If it's listed as SCSI, nobody's going to remove that, so keep a lock on it"_ then that could explain the issue.

I do have an enhancement to try to list exactly which application is keeping an exclusive lock on a drive (#312), but I have no idea when I'll have a chance to add that feature. In the meantime, you may want to try to disable the security solutions you have, or other background applications, to see if you can identify the culprit.

MagicAndre1981, 2015-05-31:
I checked the open handles and only the Kernel has some handles open. I don't use any of the listed programs. As I said, when I use the older version of the external enclosure without the the 6G (ASmedia 2105 chip) Rufus works fine.

pbatard, 2015-05-31:
> As I said, when I use the older version of the external enclosure without the the 6G (ASmedia 2105 chip) Rufus works fine.

Yes, and that version is not UASP, which means that the drive is not seen as an SCSI device by Windows, which can make a lot of difference.

I have been testing UASP devices for some time, along with other users, and you are the first person to report such an issue. Therefore, I can only conclude that this is not a general issue with UASP support in Rufus, but something that seems specific to either your system or your hardware, which of course makes it difficult to troubleshoot...

MagicAndre1981, 2015-06-02:
I tested the UASP enclosure on different devices (Laptop, PC, 8" Tablet) with Win8/8.1 and I always get the same issue.

pbatard, 2015-06-02:
OK. Without having access to such a unit, this will be tricky to troubleshoot.

I've tried to see if there was any existing tool out there that could allow identifying which process might have the lock, but I don't think handle.exe, process explorer and similar tools will help here.

What's more, the `Volume is in use` comes straight from the internal Windows formatting API, after we have successfully managed to repartition the disk and set its MBR, so I'm starting to wonder if this could be a Windows issue. The disk access errors I tend to see, due to other applications, tend to intervene a lot sooner than that. 
What happens if you try to format Partition 2 as NTFS using the native Windows formatting tool?

For what is worth, I have tried again testing an UASP enclosure using initial conditions that are as close as possible to the ones from your log (using a Windows 10 or Window 8 image shouldn't matter when the issue is with the formatting), and everything was OK:

```
Rufus version: 2.2.668
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found UAS (USB 3.0) device 'Seagate Expansion Desk SCSI Disk Device' (0BC2:3312)
Device eliminated because it was detected as an USB Hard Drive (score 23 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found UAS (USB 3.0) device 'Seagate Expansion Desk SCSI Disk Device' (0BC2:3312)
1 device found
Disk type: Fixed, Sector Size: 4096 bytes
Cylinders: 1520, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {444F3268-48AC-4C1D-B730-80B3C238CDFE}
Max parts: 128, Start Offset: 24576, Usable = 100030193664 bytes
Partition 1:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft reserved partition'
  ID: {D138CBDE-55D8-4F0A-ABDD-C5118B77AB9F}
  Size: 128 MB (134217728 bytes)
  Start Sector: 256, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {EB4DA763-42B0-4908-9C09-AC8888CEC852}
  Size: 92.7 GB (99579949056 bytes)
  Start Sector: 33024, Attributes: 0x0000000000000000
Partition 3:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI system partition'
  ID: {9408D2D9-3AF2-43F7-8703-787AF63B0F66}
  Size: 300.2 MB (314818560 bytes)
  Start Sector: 24344535, Attributes: 0x0000000000000000
Scanning image...
Disc image is an UDF image
ISO label: IR3_CENA_X64FREV_EN-US_DV9
  Size: 4032456704 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: en_windows_8.1_enterprise_with_update_x64_dvd_4065178.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE5 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Adding MSR partition
Adding extra partition
Reserved 1220 tracks (300.2 MB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume GUID \\?\Volume{39b3993a-b365-4d73-b6fb-9c7ac3ba2747}\
\\?\Volume{39b3993a-b365-4d73-b6fb-9c7ac3ba2747}\ was already mounted as F:\
Successfully remounted Volume{39b3993a-b365-4d73-b6fb-9c7ac3ba2747}\ on F:\
Applying Windows image...
Windows To Go mode selected
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll
Mounted ISO as '\\.\CDROM1'
Opening: \\.\CDROM1\sources\install.wim:[1]
Applying Windows image...
(...)
```

All in all, I think we have to work with the information that Windows is providing, which is that, as far as it is concerned, it sees the target volume being in use by 'something'. And considering that nobody else seems to be reporting a similar issue, I would tend to believe that that 'something' is external to Rufus...

MagicAndre1981, 2015-06-02:
hm, this is really wired. Here is the log from the Tablet (USB 2.0 mode with UASP):

```
Rufus version: 2.2.668
Windows version: Windows 8.1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0407
Found UAS (USB 2.0) device 'asmedia ASMT1153E SCSI Disk Device' (174C:55AA)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as an USB Hard Drive (score 11 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
Found UAS (USB 2.0) device 'asmedia ASMT1153E SCSI Disk Device' (174C:55AA)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 30401, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {892FB900-4242-4C5B-849F-FB371F38BC61}
Max parts: 128, Start Offset: 17408, Usable = 250059315712 bytes
Partition 1:
  Type: {E3C9E316-0B5C-4DB8-817D-F92DF00215AE}
  Name: 'Microsoft reserved partition'
  ID: {512E98B3-E332-40CB-B300-636D591248A0}
  Size: 128 MB (134217728 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {5450B33C-7B09-4D2E-88C4-27AAA32F0C3D}
  Size: 232.7 GB (249819171840 bytes)
  Start Sector: 264192, Attributes: 0x0000000000000000
Partition 3:
  Type: {C12A7328-F81F-11D2-BA4B-00A0C93EC93B}
  Name: 'EFI system partition'
  ID: {7C491BEE-A4CE-4E92-BB9C-BCCC2B6D99AD}
  Size: 100 MB (104864256 bytes)
  Start Sector: 488192262, Attributes: 0x0000000000000000
Scanning image...
Disc image is an UDF image
ISO label: JM1_CENA_X86FREV_DE-DE_DV9
  Size: 2735794176 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: de_windows_10_enterprise_10122_x86_dvd.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'D:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Adding MSR partition
Adding extra partition
Reserved 3251 tracks (100 MB) for extra partition
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Volume is in use
Format error: The device is in use by another process. Please close any other process that may be accessing the device.
\\?\Volume{bf2ef33f-a066-4967-92b6-07c678ea1e78}\ was already mounted as D:\
Re-mounted volume as 'D:' after error
```

MagicAndre1981, 2015-06-03:
ok, your suggestion about doing the format in the Windows format UI did the trick. Rufus no longer fails with the volume in use error.

closed 2015-06-03 by MagicAndre1981