pbatard/rufus #522: damaged USB Drive cause of Rufus!!!

bedouijesser, 2015-06-05:
I wanted to install an windows 8.1 by burning it to my USB Drive .it was going well until an error accrued .it seemed that my version was damaged. i wanted to erase and do it again with an Win 10 preview version .but the USB Drive was no longer readable 
''Windows version: Windows 7 SP1 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x040C
Found USB device 'AX216 FLASH READER USB Device' (????:????) [ID]
Device eliminated because it appears to contain no media
0 devices found''
[image]
[image]
[image]

pbatard, 2015-06-05:
https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive and https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_contains_no_media.

USB drives are not everlasting and can fail whenever you copy data onto it. As explained __at length__ in the FAQ, formatting a drive is no more dangerous than copying files. It looks like your drive chose to fail when Rufus was copying data onto it and the same thing would have happened if you were using Windows Explorer.

There's nothing Rufus can do about hardware failures.

closed 2015-06-05 by pbatard

Tmp341, 2015-06-06:
Try chipgenius, then after you see your drive's chip vendor, if there is a low level formatter tool, try it. If that isn't working, then try downloading it's firmware and try to flash.There are lots of stuff on Google.