pbatard/rufus #524: ISO Image Extraction Failure

GageW, 2015-06-10:
I recently joined the Microsoft Insider program so i could use the tech beta for windows 10, but i dont want to get rid of my current version of windows 7 i got a flashdrive (4 gb) and downloaded the 64bit iso from there website, but when i try i get this error while "Creating master boot file"

ISO Image Extraction Failure

Here is mu FULL rufus log

Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5406)
Using autorun.inf label for drive H: 'MS_DOS'
Found USB 2.0 device 'WD Ext HDD 1021 USB Device' (1058:1021)
Device eliminated because it was detected as an USB Hard Drive (score 23 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 488, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001CD29A
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4014996992 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:125C)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5406)
Using autorun.inf label for drive H: 'MS_DOS'
Found USB 2.0 device 'WD Ext HDD 1021 USB Device' (1058:1021)
Device eliminated because it was detected as an USB Hard Drive (score 23 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 488, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001CD29A
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Size: 3.7 GB (4014996992 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an UDF image
ISO label: J_CCSA_X64FRE_EN-US_DV5
  Size: 4014569472 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: Windows10_InsiderPreview_x64_EN-US_10130.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE3 for write access
Will use 'H:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Drive has a FAT16 FreeDOS Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{e58d83ee-c44d-11e4-8572-0019b91a4dd7}\
Opened drive \\?\Volume{e58d83ee-c44d-11e4-8572-0019b91a4dd7} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{e58d83ee-c44d-11e4-8572-0019b91a4dd7}\ on H:\
Copying ISO files...
Extracting files...
Disc image is an UDF image
Extracting: H:\autorun.inf (128 bytes)
Extracting: H:\boot\bcd (256 KB)
Extracting: H:\boot\boot.sdi (3 MB)
Extracting: H:\boot\bootfix.bin (1 KB)
Extracting: H:\boot\bootsect.exe (109.5 KB)
Extracting: H:\boot\en-us\bootsect.exe.mui (16.5 KB)
  Error reading UDF file /boot/en-us/bootsect.exe.mui

Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:125C)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'SanDisk Cruzer USB Device' (0781:5406)
Found USB 2.0 device 'WD Ext HDD 1021 USB Device' (1058:1021)
Device eliminated because it was detected as an USB Hard Drive (score 23 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 488, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00126F26
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 3.7 GB (4014996992 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

GageW, 2015-06-10:
GitHub added the number #524 on my title? that must be the problem number i guesse

GageW, 2015-06-10:
Ok guys win7 usb/dvd download helper failed aswell

pbatard, 2015-06-10:
Your ISO is most likely corrupted. __100%__ of the reports I ever had with `Error reading UDF file /boot/en-us/bootsect.exe.mui` boil down to a corrupted ISO.
Please check the SHA1 and confirm that it is the same as the official one from Microsoft. If you don't know what an SHA1 is, google is your friend.

pbatard, 2015-06-13:
Since there has been no further update, I'm going to close this issue.

closed 2015-06-13 by pbatard