pbatard/rufus #525: "Could not reset partitions"

CaptainShadows, 2015-06-12:
Hello, I have been trying to bring a USB that had some formatting issues back to life. Windows simply refuses to format it, to the point that even trying to bring up the context menu nearly crashes explorer, and so I thought I might give it a try with Rufus. 

I know this is not it's intended purpose so feel free to close it without a second thought. However I would appreciate some help with this.

If you haven't closed the issue by this point, here is the error log: http://pastebin.com/58fGzfVB
And here is how Rufus looks after it errors out: http://i.imgur.com/GNVvfCr.png

Thanks for any time you have spent looking at this issue.

NaoDarkness, 2015-06-12:
You could try this:
http://hddguru.com/software/HDD-LLF-Low-Level-Format-Tool/

Click on the : Download Windows Executable, that way you don't have to install anything.

I've used this before on USB's that don't show up in windows at all (no matter what I did to fix it) or when someone messed up a formatting. (Not saying you did) 

But be careful when you use it. It WILL erase everything on the drive you select.

pbatard, 2015-06-12:
Personally, I think that trying to fix a drive that contains flash blocks that are __clearly__ defective (all signs above point to a flash hardware failure) is a complete waste of time. If you value your data, you will stop using this drive altogether and either get an RMA (if still under warranty) or recycle it.

I will close this issue now.

closed 2015-06-12 by pbatard

CaptainShadows, 2015-06-12:
Ok, thanks. I was simply doing a favor for someone, the tool didn't recognize the drive so I guess that one is dead after all. Thank you very much for your assistance.