pbatard/rufus #526: Problem with Kali Linux EFI bootable ISOs

k-su, 2015-06-12:
[image]

What could be the reason that Rufus is [failing](http://superuser.com/questions/927232/uefi-bootable-pendrive-of-kali-linux)?

pbatard, 2015-06-12:
As the message indicates, you are trying to boot in pure UEFI mode (_GPT partition scheme for UEFI_) from an ISO that does not contain UEFI boot files (only BIOS).
If you want to boot Kali, you will need to change the option to _MBR partition scheme for BIOS or UEFI.

closed 2015-06-12 by pbatard

k-su, 2015-06-14:
Kali Linux has had EFI boot support [since 1.0.8](https://www.kali.org/news/kali-1-0-8-released-uefi-boot-support/), so in this case looks like Rufus' EFI detection might not be working properly?

pbatard, 2015-06-14:
No it doesn't. It is missing a `/EFI/BOOT/bootx64.efi` file, which is what the EFI specification require (for x64).

Maybe they added a __CUSTOM__ way to boot in EFI mode, from an optical media, but they certainly did not look at the EFI specs with regards to enabling __generic__ UEFI boot.
As you will understand, if every distro maintainer out there says _Screw the specs, we're going to do it our own way_, I'm not gonna run around and try to figure out what they did.

It is VERY EASY to make a distribution media EFI bootable: You just have to add an EFI boot loader in `/EFI/BOOT/bootx64.efi` and make sure it can handle loading your content from FAT32. Why distribution (#478) after distribution (#531) think they know better and keep screwing up the __official__ UEFI boot process is beyond me.