pbatard/rufus #528: add support for dmg files

pbatard, 2015-06-13:
It looks like some tools allow converting `dmg` files to DD-images, such as [dmg2img](https://github.com/lemenkov/dmg2img) (which is also GPLv3). Would be nice to see if we could integrate this into Rufus to allow `.dmg` writing on the fly. Of course, any HFS/HFS+ partitions will be inaccessible from Windows, but this is the same limitation as the one you have when writing Linux ext# images.

pbatard, 2015-06-23:
After studying various `dmg` images, I don't think this enhancement is worth it. The hope was to produce something that might be helpful for the hackintosh community, but the `InstallESD.dmg` for the latest OS X installers would be a pain to handle. Extracting the HFS+ partition from it would be a (relatively) easy deal, but the problem is that this HFS+ partition [contains a `BaseSystem.dmg` which is the actual `dmg` you want to use as the base of your USB HFS+ partition](http://www.macworld.com/article/2056561/how-to-make-a-bootable-mavericks-install-drive.html). And to make the whole thing work, you would need to grow the `BaseSystem.dmg` HFS+ partition to copy the `Packages/` folder from the initial HFS+ partition.

In other words, unless you have the means to read/write HFS+ data on Windows, any `dmg` implementation will fall very short of doing anything useful. And while I've never been adverse to the challenge of [writing file system drivers](https://github.com/pbatard/efifs), it's just not worth the investment in time when there still exist plenty of more important features to be added to Rufus. `dmg` support is __out__!

closed 2015-06-23 by pbatard

Walkman100, 2015-06-23:
> it's just not worth the investment in time when there still exist plenty of more important features to be added to Rufus

:+1: