pbatard/rufus #530: FAT format, you can put?

SoCu, 2015-06-14:
Hello, I need to be formatted with FAT, you can put this option?

Sorry for the translation, I use a translator program.

--------------------------
Hola, necesito formatear con FAT, se puede poner esta opción?

pbatard, 2015-06-14:
The option is __already__ there.
If you don't see it, it's because your USB is bigger than 4GB. FAT cannot work with >4GB drives.

closed 2015-06-14 by pbatard

SoCu, 2015-06-14:
OK thanks,

I tested applications that do not take into account the capacity of the USB, and can format USB 16 GB,

Hoped to be formatted with "Rufus" USB over 4 GB in FAT, there are more cards to 4Gb GPS devices as they need to be in FAT

pbatard, 2015-06-14:
This is not a limitation of Rufus. This is a limitation of the FAT system designed by Microsoft. For anything >4GB, you will NEVER be able to use FAT and you have to use FAT32.
Is there a reason you MUST use FAT rather than FAT32? What is it that you are trying to do that can't use FAT32?

SoCu, 2015-06-14:
MicroSD cards bearing the TomTom GPS, but I'm seeing the features of the new models, whether these models allow formatted with FAT32.

The question I had was because I saw another application if the FAT option displays USB 16Gb

Seeing that option in the other application thought it was not a limitation of Windows.

pbatard, 2015-06-14:
> The question I had was because I saw another application if the FAT option displays USB 16Gb

If they do that, then they will limit your USB size to <4GB, which makes it next to useless. FAT is limited to 4 GB by design.

SoCu, 2015-06-14:
Ok, Now I understand, 
Thanks.