pbatard/rufus #532: Rufus could not detect my USB

rodneythu, 2015-06-16:
Here is the log:
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Could not open '\\?\usbstor#disk&ven_generic&prod_flash_disk&rev_8.07#8232094f&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found

pbatard, 2015-06-16:
> `[0x00000005] Access is denied.`

Clearly, there is __something__ on your system that is preventing access to the USB.

Unfortunately, since I don't have access to your system, and I don't remember seeing a similar issue being reported, you will have to figure it out by yourself.
- What exactly is the make and model of your USB drive?
- Does it contain a hardware security solution (for encryption or something else)?
- Can you format your USB drive using the Windows format tool?
- If you use a different USB drive, is it detected?
- Have you tried using Rufus on a different computer?
- Are you using any of the software or hardware [listed here](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)?

Tmp341, 2015-06-17:
Is your USB write-protected? Or do you know your system has a write protect setting (via registry etc.)?

chrdev, 2015-06-21:
I have just upgraded from V1.4.12 to V2.2. My **SanDisk CZ80 64GB** worked OK but now I got this

> Found USB device 'SanDisk Extreme USB Device' (????:????)
> Device eliminated because it was detected as an USB Hard Drive (score 3 > 0)
> If this device is not an USB Hard Drive, please e-mail the author of this application
> NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white > triangle)

May need a fix, thank you in advance.

pbatard, 2015-06-21:
Do you see your device when you follow the "_NOTE:_" intructions (or you can also have a look [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected)) ?
Also can you provide a log from Rufus 1.4.12?

chrdev, 2015-06-25:
Yes, I saw the device when follow the "NOTE:" intructions. Thanks.

Here is my full log from version 1.4.12.535:

> Rufus version: 1.4.12.535
> Syslinux versions: 4.07/2013-07-25, 5.10/2013-06-04
> Windows version: Windows 7 SP1 64-bit
> Locale ID: 0x0409
> Found USB 2.0 device 'Generic Ultra HS-SD/MMC USB Device' (0424:4064)
> Device eliminated because it appears to contain no media
> Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
> 1 device found
> Disk type: Fixed, Sector Size: 512 bytes
> Cylinders: 7628, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x0084E17B
> Drive has a Windows 7 Master Boot Record
> Partition 1:
>   Type: exFAT (0x07)
>   Size: 58.4 GB (62741743616 bytes)
>   Start Sector: 2048, Boot: No, Recognized: Yes
> 

My Full log from version: 2.2.668

> Rufus version: 2.2.668
> Windows version: Windows 7 SP1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02~beta2
> Locale ID: 0x0409
> Found USB device 'Generic Ultra HS-SD/MMC USB Device' (????:????)
> Device eliminated because it appears to contain no media
> Found USB device 'SanDisk Extreme USB Device' (????:????)
> Device eliminated because it was detected as an USB Hard Drive (score 3 > 0)
> If this device is not an USB Hard Drive, please e-mail the author of this application
> NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
> 0 devices found

When "List USB Hard Drives" checked, this append:

> Found USB device 'Generic Ultra HS-SD/MMC USB Device' (????:????)
> Device eliminated because it appears to contain no media
> Found USB device 'SanDisk Extreme USB Device' (????:????)
> 1 device found
> Disk type: Fixed, Sector Size: 512 bytes
> Cylinders: 7628, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x0084E17B
> Drive has a Windows 7 Master Boot Record
> Partition 1:
>   Type: exFAT (0x07)
>   Size: 58.4 GB (62741743616 bytes)
>   Start Sector: 2048, Boot: No, Recognized: Yes
> 

This question marks are copied as is.

Thank you for addressing this issue.

pbatard, 2015-06-25:
Thanks for the logs.

I'm going to need your help to fix this issue as I need to find out which of Rufus 2.0, Rufus 2.1 or Rufus 2.2 introduced the regression with regards to properly detecting the VID:PID of your devices (those `(0781:5580)` or `(????:????)`) on your platform.

Can I ask you to please download both Rufus 2.0 and Rufus 2.1 from [here](https://rufus.akeo.ie/downloads/), run them with your SanDisk Extreme plugged in, and let me know whether you see `(0781:5580)` or `(????:????)` for that device? This should help me narrow down the list of changes I need to look into.

For the record, I have been testing a SanDisk extreme (with the same `0781:5580` IDs) on a Windows 7 SP1 32 bit platform and the IDs were properly populated. So this means that, since I cannot replicate the problem, I'm gonna need your help with testing. I hope that's OK with you.

chrdev, 2015-06-25:
Thanks for your prompt replay.

On my machine both Rufus 2.0 and Rufus 2.1 correctly listed 'SanDisk Extreme USB Device' (0781:5580). The issue was introduced from Rufus 2.2.

And I noticed the other device on my machine, Generic Ultra HS-SD/MMC USB Device' (0424:4064), whose ID was either populated or not along with SanDisk Extreme (0781:5580). Since you cannot replicate the problem on a 32 bit system, so I guess it maybe 64-bit specific?

Glad I can help. Thanks again.

chrdev, 2015-06-25:
I tested with Patriot XT 16GB and found some new situation maybe helpful for debugging

Rufus 2.1 recognized the device ID and listed the device

> Rufus version: 2.1.649
> Windows version: Windows 7 SP1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02~beta2
> Locale ID: 0x0409
> Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3100)
> Found USB 2.0 device 'Generic Ultra HS-SD/MMC USB Device' (0424:4064)
> Device eliminated because it appears to contain no media
> 1 device found
> Disk type: Removable, Sector Size: 512 bytes
> Cylinders: 1947, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x0049C043
> Drive has an unknown Master Boot Record
> Partition 1:
>   Type: FAT32 LBA (0x0c)
>   Size: 14.9 GB (16021192704 bytes)
>   Start Sector: 2048, Boot: No, Recognized: Yes

Rufus 2.2 doesn't recognize the device ID, or at least showed only "????:????", but listed the device as well.

> Rufus version: 2.2.668
> Windows version: Windows 7 SP1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02~beta2
> Locale ID: 0x0409
> Found USB device 'Patriot Memory USB Device' (????:????)
> Found USB device 'Generic Ultra HS-SD/MMC USB Device' (????:????)
> Device eliminated because it appears to contain no media
> 1 device found
> Disk type: Removable, Sector Size: 512 bytes
> Cylinders: 1947, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x0049C043
> Drive has an unknown Master Boot Record
> Partition 1:
>   Type: FAT32 LBA (0x0c)
>   Size: 14.9 GB (16021192704 bytes)
>   Start Sector: 2048, Boot: No, Recognized: Yes

pbatard, 2015-06-25:
Thanks for the additional tests. 

It does give me a much better idea of where to look, and I think I know what change might have introduced this regression.

The funny thing is that this change was applied precisely to avoid people running into the issue you have with getting `(????:????)` instead of proper IDs. This actually the source of your listing problem as Rufus relies in part on these numbers to determine whether a drive is a flash drive, that gets listed by default, or a hard drive, that should only be listed when the 'List Hard Drives' option is selected. Note that there are more factors into play besides the IDs, which is actually why your Patriot is listed without having to do anything, even as it's ID was not detected.

Once I fix Rufus to get the ????:???? replaced back with the expected values, I'm confident your SanDisk drive will be listed by default again.

> Since you cannot replicate the problem on a 32 bit system, so I guess it maybe 64-bit specific?

That's unlikely. What's more likely is that this depends on how many USB (internal) hubs exist on your the system, whether they are using a Microsoft driver or not, as well as where your device is plugged. But I need to have a closer look at the code to try to figure out what might be happening.

In the meantime, one thing I'd be interested in is getting an idea of the USB stack up to your SanDisk. To do that, can I please ask you to open the 'Device Manager` (from Control Panel) and then select the menu _View_ &rarr; _Devices by connection_?
After that, you will need to expand the various branches, until you see your device listed as shown below:
[image]
If you can do that on your machine, and then take a screenshot and attach it to this issue (you can simply drag and drop an image into your browser - github will do the rest), I would greatly appreciate it.

chrdev, 2015-06-25:
Thank you very much. Here is my "Devices by connection". The USB3.0 eXtensible Host Controller driver is provided by Intel (instead of Microsoft).

[image]

pbatard, 2015-06-26:
OK, I think I still need a bit more insight into what your machine detects as USB devices, so I built a version of Rufus that will provide additional details. Can you please download and run https://rufus.akeo.ie/testing/rufus-2.3_TEST3.exe and post your log?

chrdev, 2015-06-26:
My 'SanDisk Extreme USB Device' (0781:5580) is at my father's hand today. So I plugged in another device, 'SanDisk ExtremePro USB Device' (0781:5588), hope that's OK.

Log for  'SanDisk ExtremePro USB Device' (0781:5588)

> Rufus version: 2.3.682 (Alpha)
> Windows version: Windows 7 SP1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02~beta2
> Locale ID: 0x0409
> Found ID[167]: USB\VID_045E&PID_0047\7&D13C22F&0&3
> Found ID[038]: USB\VID_046D&PID_C318\7&D13C22F&0&4
> Found ID[097]: USB\VID_0A5C&PID_4503\7&1C56858F&0&2
> Found ID[227]: USB\VID_0A5C&PID_4500\6&1844AE41&0&3
> IDs belonging to USBSTOR:
>   USB\VID_0424&PID_4064\T12000085372
>   USB\VID_0781&PID_5588\AA010412151651290649
> PROCESSING: 'USBSTOR\DiskGeneric_Ultra_HS-SD/MMC_1.90'
>   Matched with ID[017]: USB\VID_0424&PID_4064\T12000085372
>   Props VID:PID = 0424:4064
> Found USB 2.0 device 'Generic Ultra HS-SD/MMC USB Device' (0424:4064)
> Device eliminated because it appears to contain no media
> PROCESSING: 'USBSTOR\DiskSanDisk_ExtremePro______0001'
>   Matched with ID[203]: USB\VID_0781&PID_5588\AA010412151651290649
>   Props VID:PID = 0781:5588
> Found USB 3.0 device 'SanDisk ExtremePro USB Device' (0781:5588)
> 1 device found
> Disk type: Removable, Sector Size: 512 bytes
> Cylinders: 15566, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x00000000
> Drive has a Zeroed Master Boot Record
> Partition 1:
>   Type: FAT32 LBA (0x0c)
>   Size: 119.2 GB (128035659776 bytes)
>   Start Sector: 32, Boot: No, Recognized: Yes

For reference here is the log for 'Patriot Memory USB Device' (13FE:3100)

> Rufus version: 2.3.682 (Alpha)
> Windows version: Windows 7 SP1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02~beta2
> Locale ID: 0x0409
> Found ID[167]: USB\VID_045E&PID_0047\7&D13C22F&0&3
> Found ID[038]: USB\VID_046D&PID_C318\7&D13C22F&0&4
> Found ID[097]: USB\VID_0A5C&PID_4503\7&1C56858F&0&2
> Found ID[227]: USB\VID_0A5C&PID_4500\6&1844AE41&0&3
> IDs belonging to USBSTOR:
>   USB\VID_0424&PID_4064\T12000085372
>   USB\VID_13FE&PID_3100\07960C0890FF0638
> PROCESSING: 'USBSTOR\Disk________Patriot_Memory__PMAP'
>   Matched with ID[051]: USB\VID_13FE&PID_3100\07960C0890FF0638
>   Props VID:PID = 13FE:3100
> Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3100)
> PROCESSING: 'USBSTOR\DiskGeneric_Ultra_HS-SD/MMC_1.90'
>   Matched with ID[017]: USB\VID_0424&PID_4064\T12000085372
>   Props VID:PID = 0424:4064
> Found USB 2.0 device 'Generic Ultra HS-SD/MMC USB Device' (0424:4064)
> Device eliminated because it appears to contain no media
> 1 device found
> Disk type: Removable, Sector Size: 512 bytes
> Cylinders: 1947, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x0049C043
> Drive has an unknown Master Boot Record
> Partition 1:
>   Type: FAT32 LBA (0x0c)
>   Size: 14.9 GB (16021192704 bytes)
>   Start Sector: 2048, Boot: No, Recognized: Yes

pbatard, 2015-06-26:
Thanks.

I'm afraid that log won't do, because there's no issue with the devices you plugged above and I don't get to see what happens when the `(????:????)` occur. I have not applied any changes to fix that detection issue, so if there's no device with the questions marks, I won't get the data I wanted to see.

But that's okay - I can wait until you get your __SanDisk CZ80 64GB__ back even if it takes a few days...

chrdev, 2015-06-26:
Thank you very much for your effort. I will get the SanDisk CZ80 64GB back within two days, and post the log then.

pbatard, 2015-06-26:
I much appreciate your efforts as well! :+1:

By the way, your logs helped we figure out that I had a small issue with TEST3 (some extra detail lines were not printing), so I have now fixed thzt. When you run your next test, please make sure that you use https://rufus.akeo.ie/testing/rufus-2.3_TEST4.exe.

chrdev, 2015-06-27:
Got my **SanDisk CZ80 64GB** back. I tested using rufus-2.3_TEST4.exe, which listed the device correctly. Version 2.0, 2.1, 2.3_Test3 and 2.3_Test4 all listed the device, only 2.2 failed and gave *(????:????)*. It'll be weird if you hadn't fixed any detection code.

Log for 'SanDisk CZ80 64GB' from rufus-2.3_TEST4.exe

> Rufus version: 2.3.682 (Test4)
> Windows version: Windows 7 SP1 64-bit
> Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
> Grub versions: 0.4.6a, 2.02~beta2
> Locale ID: 0x0409
> Processing Hub 1:
>   Found ID[046]: USB\VID_0781&PID_5580\AA010224141108081036
> Processing Hub 2:
>   Found ID[125]: USB\VID_8087&PID_0024\5&1C5CF770&0&1
> Processing Hub 3:
>   Found ID[146]: USB\VID_8087&PID_0024\5&5BA7384&0&1
> Processing Hub 4:
>   Found ID[016]: USB\VID_0424&PID_2640\7&D13C22F&0&1
>   Found ID[167]: USB\VID_045E&PID_0047\7&D13C22F&0&3
>   Found ID[038]: USB\VID_046D&PID_C318\7&D13C22F&0&4
> Processing Hub 5:
>   Found ID[017]: USB\VID_0424&PID_4064\T12000085372
> Processing Hub 6:
>   Found ID[255]: USB\VID_0A5C&PID_4502\7&1C56858F&0&1
>   Found ID[097]: USB\VID_0A5C&PID_4503\7&1C56858F&0&2
> Processing Hub 7:
> Processing Hub 8:
>   Found ID[048]: USB\VID_0424&PID_2514\6&1844AE41&0&2
>   Found ID[227]: USB\VID_0A5C&PID_4500\6&1844AE41&0&3
> Processing IDs belonging to USBSTOR:
>   USB\VID_0424&PID_4064\T12000085372
>   USB\VID_0781&PID_5580\AA010224141108081036
> Processing Device: 'USBSTOR\DiskGeneric_Ultra_HS-SD/MMC_1.90'
>   Matched with ID[017]: USB\VID_0424&PID_4064\T12000085372
>   Props VID:PID = 0424:4064
> Found USB 2.0 device 'Generic Ultra HS-SD/MMC USB Device' (0424:4064)
> Device eliminated because it appears to contain no media
> Processing Device: 'USBSTOR\DiskSanDisk_Extreme_________0001'
>   Matched with ID[046]: USB\VID_0781&PID_5580\AA010224141108081036
>   Props VID:PID = 0781:5580
> Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
> 1 device found
> Disk type: Fixed, Sector Size: 512 bytes
> Cylinders: 7628, TracksPerCylinder: 255, SectorsPerTrack: 63
> Partition type: MBR, NB Partitions: 1
> Disk ID: 0x0084E17B
> Drive has a Windows 7 Master Boot Record
> Partition 1:
>   Type: exFAT (0x07)
>   Size: 58.4 GB (62741743616 bytes)
>   Start Sector: 2048, Boot: No, Recognized: Yes

pbatard, 2015-06-27:
> It'll be weird if you hadn't fixed any detection code.

Well it __is__ weird, since the only changes that were applied to the detection code are for the debug output. TEST3 and TEST4 were compiled from a different machine though, so maybe this has to do with the compiler, as I ran an upgrade on that machine right before the 2.2 release... Can you please try https://rufus.akeo.ie/testing/rufus-2.3_TEST5.exe, which was built on the same machine I used for 2.2, and let me know how if your drive is detected?
If it isn't, then can I ask you to press `Alt`-`.` simultaneously on your keyboard while Rufus is running, to enable the debug output, and see if that changes anything. You should see the message `USB Debug enabled` on the status line in Rufus if you do it correctly. If it's still not detected with TEST5 in Debug mode, I'll be very interested in seeing the log.

On the other hand, TEST5 is how the future versions of Rufus will work, so if it detects your drive in all modes, then it means that Rufus 2.3 should detect it as well.

chrdev, 2015-06-27:
rufus-2.3_TEST5 **successfully** detected the drive. Seems like it's a ghost bug of 2.2, but 2.3 goes smoothly now maybe there is no point to look through it any more. :-)

Thank you very much for your very patient debugging.

pbatard, 2015-06-27:
OK, if 2.3_TEST5 works, then I agree that it doesn't seem worth looking into it more, as TEST5 is exactly how I would build Rufus 2.3 if I were to release it today (but there's still some more feature I am planning to add, so that release is still some time away). Also, any future release of Rufus will have the USB Debug mode available, so if you or anyone encounters that kind of detection issue again, you'll be able to use `Alt`-`.` from the release version to get the details I need for analysis.

I guess I will close the issue for now then. Even if we didn't get the final word on what caused the issue, I can't thank you enough for your help with the testing. Plus your issue also convinced me to add the USB Debug mode in future releases, which I'm sure will come handy, so you definitely helped improve the application!

closed 2015-06-27 by pbatard