pbatard/rufus #533: Is it possible to support SD cards?

Hotel26, 2015-06-18:
(Really like the lightweight, unintrusive nature of your product, Pete!)

My only usable SD device for writing an image is on a Windows system and so I would like to use Rufus to write the Raspbian image to an SD card in order to get my first boot on my new Raspberry Pi systems.  Rufus has scanned devices on my Windows system and doesn't list the SD device; I am assuming Rufus is not intended to support SD cards?  If not, would you consider supporting them?

Currently, the Raspberry Pi [installation instructions](http://www.raspberrypi.org/documentation/installation/installing-images/README.md) recommend using WIN32 Disk Imager and I will fall-back to that, but it requires Admin privilege etc etc.  I would really prefer to use Rufus.

pbatard, 2015-06-18:
Rufus' prime design goal is avoiding data loss. This means, any operation where a user may format a erroneous drive by mistake should be prevented.

Because it is very difficult to allow PCI based card readers without allowing internal drives, if we were to add support for SD and other cards as you suggest, even with a cheat mode, I am positive that this would result in someone formatting an internal drive by mistake and losing their data.

Considering that Rufus gets downloaded more than 1 million times each month, and as I mentioned a few times, I'd rather get complaints from a few users each month, who can't "lose" the data that they want, than complaints from any user who lost data they never wanted to lose.

Besides, USB card readers work just fine with Rufus. As a matter of fact, I used Rufus a few times to install RPi distributions on SD cards, as I have a couple of them myself. So the annoyance is really limited to people who have an internal PCI card reader and either don't want to purchase an USB card reader or don't want to use an alternate application such as WIN32 Disk Imager.

Not gonna happen, sorry.

closed 2015-06-18 by pbatard

chatterbeak, 2016-11-05:
The problem with Win32DiskImager, which is what RaspberryPi.org recommends, is that it's on SourceForge, which has been known to inject malware into downloadable software. 

http://www.howtogeek.com/218764/warning-don%E2%80%99t-download-software-from-sourceforge-if-you-can-help-it/

I just don't have enough time to deal with bundled malware.

pbatard, 2016-11-05:
(non USB) SD card support has been integrated into Rufus not long ago. Have you actually tested it?

Please avoid digging up old issues, and always try to check with the latest version that a feature isn't present, before asking for it.

chatterbeak, 2016-11-05:
Thanks for the tip, pbatard! Sorry I missed that this issue has been closed. I'm going to try it right now.

pbatard, 2016-11-05:
Thanks.

Don't hesitate to let me know if you find issues, as SD card support is fairly new and without access to test hardware, I can't promise that all non USB SD controllers will be detected as I hope they will.