pbatard/rufus #535: Some UI elements and text do not obey Windows high contrast text color

cookieboyeli, 2015-06-21:
Text at the bottom of the window where it says ```Ready``` and ```1 device found``` is hard coded to black and does not change with windows high contrast themes like the rest of the text.
Because of this the built in Windows High Contrast themes are incompatible with Rufus as the window background is black and the text is black. (Which is terrible for contrast)! :P

Here are three pictures: one is stock, two is High contrast 1, and three is a custom theme I am using.

[image]
[image]
[image]

I would consider this an Accessibility issue.
I have Photophobia (pain induced by light, not fear) https://en.wikipedia.org/wiki/Photophobia and often get extreme light induced headaches in addition, so a white background is sometimes literally impossible to use as I cannot keep my eyes open while looking at it.

I personally know two other people though who use high contrast themes due to other vision issues such as being legally blind, color blind, or photophobic like me. They have often complained of program incompatibility with windows high contrast. Almost nobody supports high contrast correctly. Even Explorer itself has issues with it! 

Anyway, I hope that is enough for this to get added to your list.

pbatard, 2015-06-21:
That's a very valid issue. Rufus should be using the font colour from the current theme, instead of a fixed one. I'll see what I can do for 2.3.

pbatard, 2015-06-21:
This will be fixed in the next version. Thanks for reporting this.

closed 2015-06-21 by pbatard

cookieboyeli, 2015-06-21:
Excellent! Glad it was an easy fix.

Thanks a lot! :)

commit a9ee79bd (2015-06-21): [ui] fix accessibility issues for high contrast scheme

* Also make sure we use system colours everywhere, rather than our own
* Closes #535