pbatard/rufus #536: ISO Image Extraction Failure

myeclecticself, 2015-06-21:
I'm trying to make my UBS bootable with Windows 8.1. I downloaded Windows 8.1 directly from DreamSpark. For some reason, whenever I get just passed the half way point It fails every time. I am using Rufus 2.2.668.

Here is a copy of my log:

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{ccf9b3ed-17b8-11e5-ad32-001e3391b18d}\
Opened drive \\?\Volume{ccf9b3ed-17b8-11e5-ad32-001e3391b18d} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{ccf9b3ed-17b8-11e5-ad32-001e3391b18d}\ on F:\
Copying ISO files...
Extracting files...
Disc image is an UDF image
Extracting: F:\autorun.inf (128 bytes)
Extracting: F:\boot\bcd (256 KB)
Extracting: F:\boot\boot.sdi (3 MB)
Extracting: F:\boot\bootfix.bin (1 KB)
Extracting: F:\boot\bootsect.exe (106.8 KB)
Extracting: F:\boot\en-us\bootsect.exe.mui (15.5 KB)
Extracting: F:\boot\etfsboot.com (4 KB)
Extracting: F:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: F:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: F:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: F:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: F:\boot\fonts\malgun_boot.ttf (164.3 KB)
Extracting: F:\boot\fonts\malgunn_boot.ttf (161.9 KB)
Extracting: F:\boot\fonts\meiryo_boot.ttf (131.4 KB)
Extracting: F:\boot\fonts\meiryon_boot.ttf (129.8 KB)
Extracting: F:\boot\fonts\msjh_boot.ttf (151.3 KB)
Extracting: F:\boot\fonts\msjhn_boot.ttf (149.3 KB)
Extracting: F:\boot\fonts\msyh_boot.ttf (142.8 KB)
Extracting: F:\boot\fonts\msyhn_boot.ttf (138.8 KB)
Extracting: F:\boot\fonts\segmono_boot.ttf (35.2 KB)
Extracting: F:\boot\fonts\segoe_slboot.ttf (75.6 KB)
Extracting: F:\boot\fonts\segoen_slboot.ttf (75.3 KB)
Extracting: F:\boot\fonts\wgl4_boot.ttf (46.3 KB)
Extracting: F:\boot\memtest.exe (1.1 MB)
Extracting: F:\boot\resources\bootres.dll (17.8 KB)
Extracting: F:\bootmgr (389 KB)
Extracting: F:\bootmgr.efi (1.5 MB)
Extracting: F:\efi\boot\bootx64.efi (1.5 MB)
Extracting: F:\efi\microsoft\boot\bcd (256 KB)
Extracting: F:\efi\microsoft\boot\cdboot.efi (1.3 MB)
Extracting: F:\efi\microsoft\boot\cdboot_noprompt.efi (1.3 MB)
Extracting: F:\efi\microsoft\boot\efisys.bin (1.4 MB)
Extracting: F:\efi\microsoft\boot\efisys_noprompt.bin (1.4 MB)
Extracting: F:\efi\microsoft\boot\fonts\chs_boot.ttf (3.5 MB)
[... 465 lines cut ...]
Extracting: F:\sources\hwexcludePE.txt (1.4 KB)
Extracting: F:\sources\hypervcomplcheck.dll (39.5 KB)
Extracting: F:\sources\idwbinfo.txt (123 bytes)
Extracting: F:\sources\iiscomp.dll (13.5 KB)
Extracting: F:\sources\imagingprovider.dll (175.5 KB)
Extracting: F:\sources\inf\setup.cfg (4.2 KB)
Extracting: F:\sources\input.dll (253.5 KB)
Extracting: F:\sources\install.exe (4.7 MB)
Extracting: F:\sources\install.wim (2.9 GB)
libcdio: fread (): Invalid argument
  Error reading UDF file /sources/install.wim

Found USB 2.0 device 'Generic- Multi-Card USB Device' (0BDA:0158)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:A815)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 957, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00BE5D7E
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.3 GB (7875854336 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

myeclecticself, 2015-06-21:
I re-downloaded Windows 8 from Dreamspark and was able to get it to work.

pbatard, 2015-06-21:
Yes. 99.9% of `ISO Image Extraction Failure` errors in Rufus come from a corrupted ISO.
If you are working with an official Microsoft ISO, __ALWAYS__ check the SHA1. The SHA1's are [publicly made available by Microsoft on the MSDN](https://msdn.microsoft.com/en-us/subscriptions/downloads/hh442898.aspx#searchTerm=&ProductFamilyId=545&Languages=en&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0) (Just click on the _Details_ link for an image to see it) and should be checked before reporting a possible problem with Rufus..

I will close this issue now.

closed 2015-06-21 by pbatard