pbatard/rufus #537: 2.2.668 still has issues with ASUS UASP

crazycrab1, 2015-06-22:
Plugged my Transcend USB 3 stick into my system, shows up in windows explorer, Win 7 USB maker sees it, but didn't show up in Rufus 2.2.668.  Plugged in a Transcend USB 2 stick, was able to write to it ok.

Was looking at the home page and I saw that 2.2 added support for ASUS UASP turbo mode.  I decided to go into my AI Suite, and confirmed that the turbo was on for the stick.  I switched it to normal mode and now Rufus is detecting the stick.

This isn't the newest of systems, so I don't know if newer HW or SW is being more compatible.

ASUS P8Z77-V Deluxe, using version 2104 BIOS.  Not sure what version of AI Suite II this is, though I see the USB 3.0 Boost is listed as (v1.02.11)

pbatard, 2015-06-22:
Thanks for the report.

Can I please ask you to follow the steps highlighted [here](https://github.com/pbatard/rufus/issues/513#issuecomment-105062950)? I'm especially interested to find out if your older version of Turbo Mode is using a driver that is different from `ASUSstpt` as this is the name Rufus is going to try to match to enable ASUS UASP drives operating in "Turbo Mode".

crazycrab1, 2015-06-23:
Here is the pic in turbo mode
[image]

And here it is in normal mode.
[image]

I've confirmed there's newer versions out there, I may update the USB Booster part if I can do it without updating the other stuff.  It's just a case of "it works, don't break it".  I had updated the suite once when it was relatively new and hand some problems, so had reverted back to the original stuff.

I'll be forced to anyway if I go to Win10 eventually.  Oh, I should have specified, running 64-bit Windows 7 Ultimate currently.

pbatard, 2015-06-23:
Sorry, but this is not what I wanted, as I don't have the actual name of your driver.

Here is what I would like you to do:
Please go to the Windows Device Manager, then under the `Disk drives` category, select your device. Right click on it and select `Properties`, then go to the `Details` tab, and choose the `Device Stack` property. See screenshot below (where I used a Mushkin drive, as it uses UASP). 

[image]

According to the previous user who helped me troubleshoot "Turbo Mode", one of the drivers should have the name `ASUSstpt`. But if that is not the case, that would explain your issue, and your report would tell me what I need to add in Rufus so that your system is supported.

[image]

crazycrab1, 2015-06-23:
Are those shots from Windows 8?  My Win7 didn't have a "device stack" listing and there's no Events tab.  I went through the list and took some shots of what I thought you might be looking for.  The capture app I was using to whip through these only saves in BMP, so I zipped them up.

pbatard, 2015-06-23:
Windows 7 does have a Device Manager. You don't need to download anything - it comes with Windows.
Just go to Start &rarr; Control Panel &rarr; Device Manager (or see [here](http://windows.microsoft.com/en-ie/windows/open-device-manager#1TC=windows-7) if you are still unsure how to access it). Once you are in Device Manager, just follow the instructions I posted.

Also, please note that you can not reply with attachments to github e-mails. Instead you should log on to github by clicking [here](https://github.com/pbatard/rufus/issues/537) (using the account you used to create the issue) and you will be able to drag and drop pictures directly into your browser to have them display.

crazycrab1, 2015-06-25:
Here's pics of the device manager in my Windows 7, there is no listing for "device stack".
[image]
[image]

pbatard, 2015-06-25:
Shoot, it looks like Windows 7 is missing properties than Windows 8...

OK, then let's try something else: Using the [`device.exe`](http://erwan.labalec.fr/other/devices.zip) application, can you leave the _Class_ blank and then set _Enumerator_ to USB and expand your drive (it will probably be listed as Mass Storage Device, but that may depend from the ASUS, so it's possible you will get something else).
You should get something like this.
[image]
What I am interested in is the value of the _SERVICE_ field there. Note that this is different from what you did earlier on, as you were listing the _DiskDrive_ class then.

crazycrab1, 2015-06-25:
So, no listing for the flash drive under USB, I checked under all the items, no mention of the drive or anything SCSI.

So went looking at a few other enumerators, I think the 2nd pic is the one you're interested in.
[image]
[image]

crazycrab1, 2015-06-25:
I haven't updated that ASUS Booster SW yet, if you'd like me to, I can do that to see if that fixes things, and then the rufus fix will just be "make sure your SW is up to date".  I'll wait to do that until you say OK, in case you want to figure this version out just for getting rid of the issue.

pbatard, 2015-06-25:
Thanks. Actually your first picture might have the information I am after...
And yes, if you can hold off on updating your driver, I would appreciate it, as I would like the next version of Rufus to also work for people in the same situation as you.

Now, I think that the `ASUSumsc` might be what we are after. Can you please try [this test version of Rufus](https://rufus.akeo.ie/testing/rufus-2.3_TEST1.exe) and let me know if it works?

crazycrab1, 2015-06-25:
Nope, afraid it's not showing up in the test one. (shows up in explorer)

pbatard, 2015-06-25:
OK then lets try [`rufus-2.3_TEST2.exe`](https://rufus.akeo.ie/testing/rufus-2.3_TEST1.exe).

If that doesn't work, can I please ask you to copy the log from Rufus and I also would be interested in any `oem##.inf` you find in your `C:\Windows\Inf\` directory that mentions ASUS. Especially, I expect the `.inf` files around `oem50.inf` to be from ASUS. `.inf` files are text files, so you can view their content using Notepad or any text editor. The manufacturer from the `inf` should feature prominently at the top, so it should be easy to tell if they are from ASUS or not with something like:

```
;*****************************************************************************
;
; ASUSumsc.inf   -- Install the ASUS USB 3.0 Boost Storage Driver (WDM)
;
; Copyright (c) 1997-2011 MCCI Corporation
;
; MCCI is a registered trademark of MCCI Corporation.
```

As you can't attach regular files in the github issue tracker (and I wouldn't want your driver files to be public anyway), you can e-mail them to [pete@akeo.ie](mailto://pete@akeo.ie).

pbatard, 2015-06-25:
ADDITIONAL NOTE: Please make sure you download `TEST2` and not `TEST1`. There was a typo in my original link, which I have now fixed. The file you want is https://rufus.akeo.ie/testing/rufus-2.3_TEST2.exe.

crazycrab1, 2015-06-25:
Nope, not showing in test 2.  I've got over 160 inf files here, most of them starting backwards appear to be nvidia, thanks nvidia.  #50 is the ASUS one, is that one by itself good enough?

pbatard, 2015-06-25:
I expect that more than one `inf` were used by ASUS for the driver installation, so I expect `oem49.inf` or `oem51.inf` to be the ones that might have the information I need. They should have been installed around the same time, so they should be close in sequence.
If you don't want to have spend looking them up, just zip your oem files from 40 to 60, e-mail them, and I'll sort out what I need.,,

pbatard, 2015-06-25:
By the way, can you please copy your Rufus log from TEST2?

pbatard, 2015-06-26:
OK, I think I still need a bit more insight into what your machine detects as USB devices, so I built a version of Rufus that will provide additional details. Can you please download and run https://rufus.akeo.ie/testing/rufus-2.3_TEST3.exe and post your log?

pbatard, 2015-06-26:
Fixed a small issue with the extra debugging I added. Please make sure you use https://rufus.akeo.ie/testing/rufus-2.3_TEST4.exe when testing.

crazycrab1, 2015-06-26:
That seemed to do the trick, here's the log.

pbatard, 2015-06-26:
I'm afraid the log wasn't attached as you replied through e-mail. You should just copy/paste the log data in your reply instead of using an attachment.

Also, I'm very surprised if TEST4 fixes your issue as all it does, really, is add some debugging output. Unless it's a timing thing, I really don't see how this should change the outcome. Or fid you plug your drive in a different USB port by any chance?

crazycrab1, 2015-06-26:
Nope, it's plugged into the same connector, never unplugged it from yesterday.  The only thing different between today and yesterday is I had an external SATA drive turned on for a C: backup, it's off today.  That drive I had assigned as B:   I did a Windows update yesterday, looks like the system restore was done at 2pm Pacific, the system would have gotten restarted in the process.  But it looks like I ran the Test 2 file after 3pm, judging by the time I sent you a reply, so it would have been after the reboot, in case that made some difference.

Here's the output on the log.

Rufus version: 2.3.682 (Test4)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Processing Hub 1:
  Found ID[029]: USB\VID_05E3&PID_0607\5&2FA0C562&0&3
  Found ID[234]: VID_8564&PID_1000&ASUS\{98F2FC55-265C-4405-9AA3-57C78C242DAF}\1547967683
Processing Hub 2:
  Found ID[094]: USB\VID_8087&PID_0024\5&27E8C4B3&0&1
Processing Hub 3:
  Found ID[011]: USB\VID_8087&PID_0024\5&38CA7A24&0&1
Processing Hub 4:
Processing Hub 5:
Processing Hub 6:
  Found ID[095]: USB\VID_046D&PID_C228\6&2CE94079&0&1
  Found ID[194]: USB\VID_046D&PID_C229\6&2CE94079&0&2
  Found ID[077]: USB\VID_046D&PID_C06B\AF11DC160035
Processing Hub 7:
  Found ID[162]: USB\VID_046D&PID_C531\7&3A7E93F1&0&5
Processing Hub 8:
  Found ID[253]: USB\VID_2001&PID_F103\6&182A12DA&0&5
  Found ID[024]: USB\VID_045E&PID_0291\6&182A12DA&0&8
Processing Hub 9:
Processing IDs belonging to USBSTOR:
  Root\LEGACY_USBSTOR\0000
Processing IDs belonging to ASUSSTPT:
  ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\1547967683
Processing Device: 'SCSI\DiskJetFlashTranscend_32GB__1.00'
  Matched with ID[105]: ASUSCR\{d6ef3e17-93db-4bd9-9cdd-e23bd67eaa55}\1547967683
  Matched with (GP) ID[234]: VID_8564&PID_1000&ASUS\{98F2FC55-265C-4405-9AA3-57C78C242DAF}\1547967683
  Props VID:PID = 8564:1000
Found UAS (USB 3.0) device 'JetFlash Transcend 32GB SCSI Disk Device' (8564:1000) [GP]
Processing Device: 'SCSI\DiskWDC_WD1001FALS-00J7B1___05.0'
  Non USB => Eliminated
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 3843, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002FD78A
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.4 GB (31610372096 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

I just ejected, unplugged, and replugged it back in to see if it'd still show up, it did.

pbatard, 2015-06-26:
Interesting... Can you try `Alt`-`.` in Rufus and see if your drive still shows?
That's pressing the `Alt` and "dot" keys simultaneously and you see a message `USB Debug disabled` on the status bar...

crazycrab1, 2015-06-27:
Alt-. doesn't seem to do anything.  I'm just hearing a "ding", I'm seeing nothing say USB debug disabled

pbatard, 2015-06-27:
Oops, my bad. TEST4 was built before I added the shortcut. Please try https://rufus.akeo.ie/testing/rufus-2.3_TEST5.exe. Actually, this one has USB Debug disabled by default, so it should logically behave the same as Rufus 2.2. I'm very curious to find out if your device will be detected or not. If not, then please try `Alt`-`.` (i.e. enable USB Debug) and see if that works.

crazycrab1, 2015-06-27:
Drive shows up in test 5

pbatard, 2015-06-27:
Weird... I really don't have a good explanation for why the behaviour changed, as there really is zero functional change in the USB detection procedure. TEST5 is pretty much how Rufus is going to be from now on, so if that works, we'll have to leave it at that. 

One last thing I'd like to see you do if you don't mind is try to run TEST2 again (if you deleted it, you can pick it back up from [here](https://rufus.akeo.ie/testing/)) and see if your drive is still not detected there. This will tell us for sure if the change has to do with Rufus or with your environment.

crazycrab1, 2015-06-27:
Hmmm, yeah, test 2 worked now.

pbatard, 2015-06-27:
OK, then it looks like this had something to do with your machine. I guess if you ever get back to the conditions where your drive wasn't detected, we can try to get more data...

At any rate, thanks for helping with these tests!
We may not have found what your issue was, but at least this helped me add some more debugging info into Rufus, and I'm sure that will come handy in the future to help other people.

crazycrab1, 2015-06-27:
Yeah, that's weird.  I just tried the original 2.2.668 and it shows up there now.  Shrug.  Maybe something that updated yesterday magically fixed it.  I just confirmed that turbo was still turned on.  Oh well, thanks for the help. :)

pbatard, 2015-06-28:
I'm going to close this issue now. Thanks again for the help.

closed 2015-06-28 by pbatard