pbatard/rufus #540: Unattended Win10 setup settings are not preserved after RUFUS iso->to->usb

user1313, 2015-06-27:
Hi,

Unattended setup of Windows 10 which was created in NTLITE (https://www.ntlite.com). When i use the ISO directly in my VM, or if i burn it to a DVD, everything works perfect, including all the unattended steps (like automatic partitioning, skipping OOBE, skipping user creation). However, when using Rufus 2.2 (the only tool that recognizes USB sticks in my VM setup by the way), installation still works but unattended info is lost. I tried the stick both in VM and on a real computer. Can you help?

Regards,

pbatard, 2015-06-27:
I'm afraid that, unless I have a lot of time on my hands, I don't think I will go the extra mile in Rufus to support non-official Microsoft ISOs. It's just not worth the return on investment considering that Rufus is not designed to be a sysadmin tool and [as for multiboot](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiboot_support), I expect people who want to perform custom Windows installation to have the knowledge required to create a working bootable USB drive on their own.

As much as I'd like to add every feature that gets suggested, or fix every compatibility issue, my development time is simply to limited to do that, and I have to consider [the actual amount of users the feature would benefit vs. the cost in terms of development+testing+maintainance](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard). So right now, I think that, unless someone performs the analysis and tells me exactly what Rufus would need to do to be compatible with nLite's unattended ISOs (i.e. in which fashion should the ISO content be copied over to the USB or modified) and these requirements aren't found too costly to implement, I'm gonna pass on doing any extra work for nLite compatibility. It just doesn't seem to be worth the investment for a tool that is primarily aimed at being used by the general public on "attended" installations, using unmodified Microsoft ISOs.

Then again, Rufus is Open Source, so if someone else really cares about adding nLite compatibility and is willing to do the work and send a patch, I will gladly accept it.

closed 2015-06-27 by pbatard

Walkman100, 2015-06-27:
Just a note, it's N<b>T</b>Lite not nLite - delete this comment as you like.

user1313, 2015-07-04:
Turned out to be completely unrelated to Rufus, sorry.
The solution from the thread below has fixed it for me. The workaround is to introduce additional "autounattend.xml" copy into one of two images inside "boot.wim".

https://social.technet.microsoft.com/Forums/windows/en-US/82410df7-2e70-47c7-a9dc-7dba7c17fbf8/autounattendxml-will-not-load-from-usb-hdd

pbatard, 2015-07-04:
Thanks for the update. I'm sure this will help other people facing the same issue!