pbatard/rufus #541: CPU-Burn-In - UI: Check needed - HOST: Windows NT 6.3.1 NVHD - Error / Feature Request

S466531257BOSS, 2015-06-27:
Hi.

as DUMB as it sounds:
I run into an issue ( so to say ) while using Rufus 2.2.668 hosted in Windows 8.1 Industry Embedded booted via Native-VHD-Boot;

Due to the circumstance that Fedora 22 Workstation ( Version downloaded today, whatever it is ) do not want to function in any combination tested, i ( idiot me ) ran Rufus intentionally with administrative rights and while with burning eyes fumbled the settings, obviously the focus was on Drive Selection and i used the Scroll-Wheel; So though out of the Window i switched the target Drive. No problem at all, normally, because i am always watching that i have ONLY ONE FLASH-DRIVE IN USE when using Rufus.

However ...

when running with administrative rights, the System-Drive of Windows ( know i know ... ) is showing up ( Bwuahaha - here am i - plz problem sitting there staring at and abusing me, kill me ! );

You are able to follow ? Can think what has happened ?

Well, if it would have been Windows 98, i would think i could not write you anymore.

The only reason why i was recognizing, somethings wrong, was the NOISE my HDD (15k Seagate Barracuda ) was producing. Sounded something like HULK on Binaural Beats eating a caterpillar ...

FACT IS:

At least, for Bollocks like me, who use this wonderful Piece of SoftWare even in unconscious state of mind, it would be very helpful if there could possibly a:

CHECK IF THE AVAILABLE VHD IS POSSIBLY IDENTICAL WITH %SYSTEM_DRIVE% ...

I guarantee : Reproducible - Funny noise and ones awake right that moment!

I wish: You might double check for total dumbness on user-side.

I hope: i don't do it again, because Windows tried to scan for possible repair actions needed WHILE Rufus was trying to rewrite the System-VHD. It took around 30 secs until rufus could be killed and after that 3 minutes until the system reacted again with the note: No errors found. Phuh ...

I'm tired ...

So, yah? Ah! Yeah :+1 for the fun i have had right now!

Thanks and sorry for the lengthy, but hopefully insightful description.

If you find some bits of time, thanks in advance.

THANKS FOR YOUR INTEREST.

PAETH CLAUDIUSRAPHAEL
S466531257 BOSS

pbatard, 2015-06-27:
I'm very sorry to hear what happened. I have a handful of checks in Rufus, such as making sure internal (non VHD) drives are never listed, and making sure the drive Rufus is running from cannot be formatted, but I didn't go the extra mile for VHDs as my expectation was they would not be used for anything critical and it was convenient to have them listed by default. Unfortunately, it looks like you proved me wrong.

If I understand correctly:
* You are running your main Windows system from a VHD, which contains your system partition (most likely `C:`) and possibly more
* You were running Rufus from a drive that wasn't your system one (else Rufus should have removed it from the list)
* When you did that, the system drive was listed as a selectable target in Rufus, and there were no errors when trying to repartition and reformat it (I would have expected some errors from the default Windows safety mechanisms, but of course reformatting the system drive is not something I ever tried...)

> CHECK IF THE AVAILABLE VHD IS POSSIBLY IDENTICAL WITH %SYSTEM_DRIVE% ...

I think that's a very valid check. I never thought this would be needed, but now I'll make sure the System Drive can never ever be listed with the next version of Rufus.

I can only hope that you either had any valuable data backed up, or that any data you lost was not that important.

closed 2015-06-27 by pbatard

S466531257BOSS, 2015-07-04:
Thanks pbatard for helping out and yes, you almost got it right in your conclusion.

The battle Windows fought ended in no data-loss ( besides actual session ) at all, due to the fact the system ran off a child-vhd as customized differ to a master-vhd. Could be that this is the main reason, why the system-vhd was listed. I just did it, because other tools were also attempting to kill the system disk if purpose was given by dumb in front of pc :) So, anything is fine and hey:

NOW YOU HAVE THE MOST SECURE SUPER CHECK AT ALL - hopefully - because i don't want to try out yet :)

However, thank you very much for taking the time and have a thought on it.

ALL THE BEST TO YOU.

PAETH CLAUDIUSRAPHAEL
S466531257 BOSS

BTW.: if you wish to test cuddly vhd-stuff in future-versions feel free to contact me, i'd like to help with testing. And just to hotten some thinking on that:

What about:

PURPOSED WRITE TO RUNNING SYSTEM VHD IF SYSTEM-VHD IS CHILD-VHD?

Scenario i can think of is: using RUFUS to add blends/versions/editions of OS ( in batch ) besides running system. Feel free to use this idea, i am totally believing in RUFUS to be the perfect tool for such a function. You know what: I post it as a new feature request, so its apart this thread.

Thank you for your wonderful practical neat and lean tool.

Just to mention it, any idea why Fedora 22 Workstation is not squashable via RUFUS ? Well, i think i add this as a problem report.

commit 47767401 (2015-06-27): [core] make sure the system disk is never listed

* When using VHD or VMDK drives, it was possible for the system disk to be listed.
* Closes #541