pbatard/rufus #542: Mageia 5 live .iso report

measter1, 2015-06-28:
I don't know if you prefer email or a report here.  Links below.

Win7 Rufus 2.2 alert Syslinux 6.0.3 directory renamed a Mageia
Mageia link below advises do not use UNetBootin and instructs to use Usbdumper.

Attempt to use YUMI failes to boot.  Attempt to use Rufus 2.2 fails to boot.  Attempt to use Rufus to just quick format fails to satisfy usbdumper (info at link below).  Attempt to use Win7 format fails to format.  Attempt to use Rufus to format followed by Win7 succeeds in format but Usbdumper fails.  All the usbdumper fails are the same writing error/access denied as described at link below.

For failures as above, Usbdumper requires that the USB be prepared with Win diskpart + list disk then select disk + clean.  Then usbdumper succeeds.  Usbdumper created USB from .iso boots.

https://www.mageia.org/en/downloads/  (see infos, torrents, directs  at links on that page)
Mageia-5-LiveCD-KDE4-en-i586-CD.iso  English only CD 32 bit

usbdumper http://mirrors.kernel.org/mageia/people/marcom/usbdumper.exe

measter1, 2015-06-28:
2014 June Mageia help page includes Rufus in list
https://wiki.mageia.org/en/Dump_Mageia_ISO_on_a_USB_flash_drive_-_Alternative_tools

pbatard, 2015-06-28:
I prefer the issue tracker.

> Win7 Rufus 2.2 alert Syslinux 6.0.3 directory renamed a Mageia

??? What does this sentence mean?

>  Attempt to use Rufus 2.2 fails to boot.

What is the error? Were you trying in BIOS or UEFI mode? Also did you check the Rufus log for details?

>  Attempt to use Rufus to just quick format fails to satisfy usbdumper

Well, of course, since it looks like the developers of usbdumper are trying to apply a `dd`-like operation and probably failed to realize that having a handle and a lock to a physical Windows drive will still not allow you to write data over a Logical Drive (i.e. partition) [unless you also have obtained a lock to them](https://github.com/pbatard/rufus/blob/master/src/format.c#L1571). So they manually require people to remove these partitions. 

I'm assuming the Mageia developers drank the ISOHybrid kool-aid (and might have based their boot process on something similar OpenSuse, which does away with FAT32 support, __purely for distro maintainer convenience__, and therefore leaves people using UNetBootin or Rufus unhappy).

But if they just want people to use `dd` to write the ISO, they should just advise to use established Windows tools to do so, such as _Win32 Disk Imager_ or _WinImage_, as pointed in your second entry. Have you tried any of those?

Or, they can also use Rufus to write in `dd` mode. All you have to do is use the [`Alt-I` option from Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes) to force image mode, and it will not try to extract the ISO, but just write the image as is. If Mageia uses an ISOHybrid, that too should work fine - can you please try it?

I'll try to have a look at the Mageia ISO, and whether they are introduced the same kind of USB incompatibility as OpenSuse does when I get a chance...

pbatard, 2015-06-28:
Looking further, it seems Mageia is a derivative of Mandriva. In that case, I'm pretty sure #478 applies, since it applies to Mandriva...

measter1, 2015-06-28:
I don't know how to quote with this interface.

    Win7 Rufus 2.2 alert Syslinux 6.0.3 directory renamed a Mageia

??? What does this sentence mean?

Similar to PCLOS;  Rufus says: This system uses Syslinux 6.03/Mageia, but this application only includes the installation files for Syslinux 6.03/2014-10-06

That causes Rufus to create a Mageia directory in its rufus_files/ Syslinux-6.03 to contain the ldlinux .bss & .sys files to use.  Future attempts give no error unless the directory and files are eliminated.

    Attempt to use Rufus 2.2 fails to boot.

> What is the error? Were you trying in BIOS or UEFI mode? Also did you check the Rufus log for details?  

I will try to figure out a way to copy the boot error info.  BIOS mode.  I see the Rufus log which I can copy and paste all or part of here.  There doesn't seem to be anything unhappy. :-)

closed 2015-06-28 by measter1

measter1, 2015-06-28:
alt-I dd mode works.

Thanks. I try to find these things on my own but I didn't have enough insight to get me there.

measter1, 2015-06-29:
Suggestion to Mageia forum for improving the Mageia guides to USB to include more integration to the Rufus advice and to supplement the Rufus info line with alt-I dd suggestion.

https://forums.mageia.org/en/viewtopic.php?f=7&t=9952

papoteur-mga, 2015-06-29:
Hello,
I'm the maintainer of usbdumper. It was known as Mandriva-seed, and yes, it use the dd method to write the ISO.
Thanks for pointing out the portion of your code were the ISO is written. I will study it.
For the method of ISO creation, I have no knowledge on it and don't know if it would be easy to create ISO in the standard you refer to.
Papoteur

papoteur-mga, 2015-06-29:
And the actual code is here :
http://gitweb.mageia.org/software/usbdumper/tree/liveusb/creator.py#n353
It's python and Qt.

pbatard, 2015-06-29:
Hi __Papoteur__,

> It was known as Mandriva-seed, and yes, it use the dd method to write the ISO.

Yeah, I kind of figured that already. So I guess it uses the same kind of problematic ISOHybrid method as openSuse, since that's what Mandriva seems to be using.

> Thanks for pointing out the portion of your code were the ISO is written. I will study it.

Let me know if I can help. I got stung with this one too when I started developing Rufus: It doesn't make sense for write access to fail on a Physical Drive you have exclusive access to, until you realize that Windows has an extra check to find if the data you're trying to write to also belongs to an existing Logical Drive. It's a bit of a pain in the ass to work with, but it's manageable. I suspect, since you're going `dd`, you probably want to delete the whole partition table and refresh before you do anything, so the following might help you out: https://github.com/pbatard/rufus/blob/master/src/drive.c#L1253-1276. You may still have/want to unmount any existing partition from the drive first, to give a chance to users to find out if they are currently in use.

> For the method of ISO creation, I have no knowledge on it and don't know if it would be easy to create ISO in the standard you refer to.

Well, the core issue is with openSuse and is what I described [here](https://github.com/pbatard/rufus/issues/478#issuecomment-87504806). The issue is that the `searchImageISODevice` script is only set to look for an ISO9660 filesystem, which works fine when the image is written in `dd` mode, but which doesn't work so well when, as tools like Rufus will try to do by default, the content gets copied on a FAT32 partition (which might be useful in case users want to edit it, or add a WiFi driver, etc...).

As opposed to what Debian does, which is to ensure that both FAT32 and ISO9660 file systems are supported for the detection of the installation media, even though they too are using an ISOHybrid, the openSuse people limited themselves to ISO9660 only... :disappointed: 

I don't believe it should be that difficult for the openSuse maintainers to support both, and I tried to open a report with them about this, but I found that their issue tracker system was less then friendly, and no matter which browser I tried, I just couldn't seem to create an issue... The openSuse people also [didn't seem too keen about modifying their script when I mentioned the issue on reddit](https://www.reddit.com/r/linux4noobs/comments/36mxds/i_cant_seem_to_get_opensuse_to_boot_from_liveusb/cridpca).

Note however that I have some plans to try to detect if an ISO image is an ISOHybrid and prompt users in Rufus to ask them if they want to write it in `file copy` or `dd` mode (#543). This should ensure that, even if the openSuse people don't change their scripts, users will get a better chance of getting a bootable USB.

measter1, 2015-06-29:
Apparently all of these names are related:

isodumper
opensuse studio imagewriter
ubuntu usb-imagewriter
mandriva-seed
usbdumper

papoteur-mga, 2015-06-30:
Yes isodumper is a fork of Ubuntu usb-imagewriter.
And usbdumper is the suit of mandriva-seed.
Papoteur

measter1, 2015-06-30:
I'm trying to educate myself currently from entries at syslinux about isolinux and isohybrid and I am finding especially helpful numerous Ub forum threads, tutorials, messages, and links by Nio Wiklund (sudodus) who 'developed' mkusb a bash script which uses dd and a number of other tools for different types of USBs. The/His threads are current and active.

http://www.syslinux.org/wiki/index.php/ISOLINUX
http://www.syslinux.org/wiki/index.php/Isohybrid
https://help.ubuntu.com/community/mkusb
9 pages http://ubuntuforums.org/showthread.php?t=1958073 Howto make USB boot drives
2 pages http://ubuntuforums.org/showthread.php?t=2259682 One pendrive for all PC (Intel/AMD) computers