pbatard/rufus #543: Alert users if the image is an ISOHybrid

pbatard, 2015-06-28:
Should we try to detect ISOHybrid and [let users know/confuse them with] what they are dealing with?

measter1, 2015-06-29:
I think that it would be useful if there were a 'probe .iso' feature/option which could provide the 'generic' user with information which would likely help him.

In this context, (I think) the generic user should/could be considered to be someone who has:
 - experienced several different USB boot writers, including their varieties of failures
 - experienced writing dozens if not scores of linux .iso/s to USB
 - experienced different kinds of live .iso boot failures in addition to those related to/ caused by/ the .iso to USB write, ie the same kind of boot failure which would occur if the .iso had been written to optical or a USB successfully; aiding the user in distinguishing boot failures unrelated to the USB write

I don't understand enough about the problems of writing .iso/s to USB to know anything about what kind of information from a probe would be useful; but I don't think 'confusion' should deter the information gathering, because confusion can be mitigated by FAQ information.

pbatard, 2015-06-29:
I'm sorry but I disagree with your definition of generic user. Rufus gets downloaded more than 1 million times a month, and as such, I think it's fair to say that the generic users of Rufus are definitely not the kind of people you describe above but instead people who are not familiar with other tools and who are creating a bootable Linux USB for the first time. So this is how I will approach the information I am going to pass to the user.

Also, I don't ever expect the vast majority of users to ever read the FAQ when all they want is boot a Linux system. Either it works (and maybe they have to pick a choice they aren't too sure about in the process, and will remember that they can try the other choice if their first attempt fails), or it doesn't and __they will look elsewhere__.

As such, it's not going to be beneficial to try to cater to the limited group of people who _"experienced several different USB boot writers"_ or _"experienced writing scores of linux .iso/s to USB"_ because I consider that these people should actually be able to handle themselves if they encounter an issue. It's the people who just want to perform a one-time Linux install that need to be considered.

measter1, 2015-06-29:
I think almost every linux USB writing user starts with UNetBootin because of the duration - breadth and depth - of its exposure.  I also think that the vast majority of those users who find & dl Rufus do so because they are having some kind of problem.  If we agree on both of those premises, then all we have left to debate is how _much_ experience generic Rufus users already have with writing .iso/s to USB and how _much_ experience they have with USB writing failures.

Rufus probes the .iso and then acts on that information algorithmically.  Maybe the user could be given some written Rufus-guru guidance somewhere about how to use Rufus log information.

pbatard, 2015-06-29:
>  Maybe the user could be given some written Rufus-guru guidance somewhere about how to use Rufus log information.

Oh boy! The last thing I want is have to teach millions of users how to interpret the log data (even more so as there is currently __zero__ indication about whether your ISO is an Hybrid or not). The log is for the developer. Now you can try to interpret the data you see in it (some of which may be fairly straightforward), but please don't count on the developer to help you out. I barely have enough time to answer all the user requests I get and try to save enough time for development, so I'm not gonna teach anyone on how to interpret log data. Not a chance.

Also, you can try to push the debate all you want, but as far as I'm concerned, the amount of experience generic Rufus users have is plain as day. I'm not trying to be dismissive here, but the many user requests __I receive every single day__, are certainly not looking in the favour of the agenda you are trying to push, with users having some experience writing ISOs and stuff. You may also want to have a look [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard) to understand where I am coming from.

Finally, remember that any UI change in Rufus means rallying more than 30 translators, which, as much as I'm super grateful to the awesome work they do (especially as they are doing it for free!), is a bit like herding cats - it literally takes __months__ to get more than 90% of new translations back, and I also expect that some of these translators will have as much trouble trying to translate the concept of an hybrid ISO as users may have to understand it => The confusion __is__ a genuine concern, and as such, my duty as a developer is to keep the choices (and their description) simple, rather than expect users to acquire the experience they would actually need to make an informed decision. If it doesn't fit the direction you would prefer Rufus to go, so be it.

measter1, 2015-06-30:
> there is currently zero indication about whether your ISO is an Hybrid or not)

Oh.  I see.  So the current 'probe' I conceive wouldn't aid that or the subject/target of this thread.

> You may also want to have a look here to understand where I am coming from.

Yes I've read that FAQ par and respect its issues.

I don't think the generic user's best strategy is to try one USB writer after another 'willy-nilly' without some concept of how to choose one writer over another.  I think the generic user with various experiences of USB writers that don't work sometimes who wants to have some primary USB writer and some alternative strategy depending on the nuances of a particular distro wants and needs to understand better - see, appreciate somehow - what some particular failure problem is caused by.

I think the problem of 'generic' users having to seek 'direct' answers such as I did from newsgroups and web forums by asking questions (as opposed to 'simply' finding information based on the particular .iso/s nature) should be minimized.  YUMI thinks it has insight into which distros its strategies and modifications thereof are compatible.  That YUMI insight or categorization is not accurate.

Your expertise as a developer in this sphere gives you great insight.  (But) You are still learning the nuances of the various distros.  The generic user can't expect your current level of insight, but I believe the generic user can expect to know that the various writers have specific and known strengths and weaknesses and they should also be able to have some strategies to guide them in how to get some .iso written successfully to USB  more 'directive and beyond reckless try this or try that and then if that doesn't work try this other one blindly.

I don't believe that our generic user is inclined in that 'random'  way at all; but s/he needs sufficient understanding of how .iso writing to USB works differently between the writers.

measter1, 2015-06-30:
According to a discussion of isohybird:  "These isohybrid images contain in addition to the normal CD-based ISO9660 filesystem, a valid-looking DOS-style partition table"

Leaving opensuse discussions out, is that isohybrid condition described above discoverable?  Easily enough?

pbatard, 2015-06-30:
Easily, yes. But it has ramifications (because I can't just add a check for ISOHybrid without first having to redesign other stuff). I __did__ mention detecting ISOHybrid in my original comment.

My plan, if I add the feature, has always been to:
1. Detect if the currently selected image is an ISOHybrid
2. Pop a dialog that will prompt the user between the DD and file copy mode of writings, alerting them that, if one doesn't work, they should try the other.

Again, while that looks simple to do on paper, please do understand that, as with any application, there's stuff that needs to happen behind the scenes to make __seemingly__ simple features like the one above work.

commit fb09802c (2015-08-22): [ui] prompt users how they want to write ISOHybrid images

* Closes #543
* Also update localization messages and image analysis report
closed 2015-08-22 by pbatard