pbatard/rufus #544: Rufus not detecting USB!

windows10insider, 2015-06-29:
I am trying to install windows 10 preview on a flash drive with Rufus. It doesn't seem to be detecting my USB drive. Here is the log:
Rufus version: 2.2.668
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'PNY USB 2.0 FD USB Device' (????:????) [ID]
Could not open '\\?\usbstor#disk&ven_pny&prod_usb_2.0_fd&rev_1100#aa00000000000246&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
Any help would be appreciated. Thanks!

pbatard, 2015-06-30:
```
Could not open '\\?\usbstor#disk&ven_pny&prod_usb_2.0_fd&rev_1100#aa00000000000246&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
```
Well, something other than Rufus seems to be preventing access to your USB drive. You will have to figure out what it is. It could be one of the software listed [here](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) or it could be something else.

The error above is basically Windows telling Rufus _"Hey, I know you'd like to access this drive, but something else doesn't want you to..."_.

windows10insider, 2015-06-30:
Thanks for the quick reply. I will do what you suggested. Will post soon.

windows10insider, 2015-06-30:
I was wondering whether this error could be caused by the fact I am not running Rufus as an administrator on my computer.

Walkman100, 2015-06-30:
Does rufus not immediately ask for admin permissions on your machine? as soon as I run the executable, I have to approve it through UAC before going further...

windows10insider, 2015-06-30:
No, it doesn't ask for admin permissions.

pbatard, 2015-06-30:
> No, it doesn't ask for admin permissions.

Then this is your issue. Rufus [__must__ run with administrative privileges](https://github.com/pbatard/rufus/wiki/FAQ#Why_does_Rufus_need_elevated_privileges_to_run) and is set to ask for elevation on startup using Microsoft's recommended approach to do so. However, I suspect that the reason you are not seeing the prompt is being you are using XP, and UAC was handled differently here.

Since this is not something I can fix in Rufus (and I am not going to go out of my way to add an additional notification message for a platform that has been retired for more than a year now), you will have to make sure that you run Rufus from an account with sufficient privileges. Oh, and you should really consider moving away from Windows XP, especially if your machine is connecting to the internet...

I will close this issue now.

closed 2015-06-30 by pbatard

windows10insider, 2015-06-30:
Okay, thanks.