pbatard/rufus #545: Failed to create MS-DOS bootable USB drive on Windows 10

BrazilianJoe, 2015-06-30:
I am Windows10 build 10130 64 bits.

I tried to create a bootableMS-DOS USBdriveusing Rufus, but it balked with an error where it failed to find a .DLL to copy to the disk. 

inspecting the disk it seems the partitioning/formatting went OK, but only the LOCALE folder was created, and no files were copied. 

I guess Windows 10 removed or moved some files which Rufus relies on to create the bootable disks.

bjpafa, 2015-06-30:
RUFUS WITH WINDOWS 10  AFTER 9845
RELEASE.

I repeated the procedure with a HP G6 i7 notebook.  Stopped after install
new partition.
Hp support changed the HD for a new one.
Just 10130 and never get there.

>
> I am Windows10 build 10130 64 bits.
>
> I tried to create a bootableMS-DOS USBdriveusing Rufus, but it balked
with an error where it failed to find a .DLL to copy to the disk.
>
> inspecting the disk it seems the partitioning/formatting went OK, but
only the LOCALE folder was created, and no files were copied.
>
> I guess Windows 10 removed or moved some files which Rufus relies on to
create the bootable disks.

pbatard, 2015-06-30:
Care to provide a __FULL LOG__ from Rufus? Thanks.

pbatard, 2015-07-01:
Again, could you please provide the log from Rufus? You just have to click on the _Log_ button after creating the USB? Also, did you try _FreeDOS_ instead of _MS-DOS_?

pbatard, 2015-07-01:
Well, a log would have been nice, but I have now reinstalled Windows 10 preview to reproduced the issue. Microsoft appears to have removed `diskcopy.dll` which is the library that contains the MS-DOS floppy image we extract the MS-DOS files. So I'll be modifying Rufus to remove MS-DOS support for Windows 10.

Oh and before you ask, no it is not possible to embed the MS-DOS files in Rufus as they are copyrighted by Microsoft and can't be redistributed without a license, which is the whole reason why we had to fetch them from an existing library. But that's not a big loss anyway, since FreeDOS works just fine in Windows 10 and should be preferred over MS-DOS anyway.

closed 2015-07-01 by pbatard

mirh, 2016-07-19:
FreeDOS still fails at some BIOS related tasks, for the records. 
But I hope they'll be able to overcome this in the incoming 1.2 version.

pbatard, 2016-07-19:
Have you contacted the FreeDOS mailing list about this, with the specifics of the BIOS related tasks that fail.

__Unless__ you report your findings to them, it will never be fixed.

mirh, 2016-07-19:
I hate mailing lists :p
Anyway, I know it's not your business to fix it. It was just a detail.

commit 3ccae1d2 (2015-07-01): disable MS-DOS creation for Windows 10 or later

* Microsoft removed the diskcopy.dll that contained the MS-DOS floppy image
  Not a big loss, since we have FreeDOS... ;)
* Closes #545
* Also apply a long overdue refactoring for boot and target types
* Also update the README to add an icon