pbatard/rufus #546: Win10 to go on USB disk, having trouble booting with secure boot on Surface Pro 3

myly10, 2015-07-01:
I have a 2T USB disk installed Win10 To go with Rufus, but when I tried to boot it on Surface, the screen shows that this OS is unauthorized and blocked by secure boot. When turn off secure boot, everything is fine except the bright red screen during booting.

My surface is running Win8.1 Enterprise, Win 10 to go is Win10 Pro Insider, version 10130.

Thanks.

pbatard, 2015-07-01:
Please provide the __full__ log of how you created your USB in Rufus.
If you used NTFS as the target file system, this is normal and does not make your system any less secure. You can turn back secure boot after installation .

pbatard, 2015-07-07:
Closing this issue, as there doesn't seem to be anything to do.

closed 2015-07-07 by pbatard