pbatard/rufus #547: Problem : Fedora 22 Workstation won't run

S466531257BOSS, 2015-07-04:
Just to announce it:

I have problems getting Fedora 22 WorkStation to become "burned" by RUFUS:

Used Flash-Drives:
- Kingston TOO USB2 2GiB with full USB 1.0 , 1.1, 2.0 Compliance ( Anno 2004 )
- Sandisk Cruizer Extreme 64GB USB 3.0
- LogiLink Active USB-2.0 µSD-Card-Adapter ( 2GiB , 8GiB , 16GiB , 32GiB -- Samsung , SanDisk, LiteOn , Medion -- SD/SDHC Class 6 / 10 )
- direct to SDHC 4GiB SanDisk
- direct to CF II SanDisk Industrial 16GiB
- direct to CF II SanDisk Extreme III 8 GiB

I attempted to target UEFI-based ( FW < 2.3 ) 64-bit Systems so tried all options ( that came to my mind ) and combinations RUFUS spends.

The SysLinux Version RUFUS mentioned and downloaded seems correct.
In any case the system boots but hangs critically right the moment it checks for its own content.
Seems to be the typical, i forgot where i run from, please tell me where i can find me -stuff.

The only way in contradiction to RUFUS and comparable Tools i found to be working is indeed using dd for direct write of iso -- what obviously is a problem, due to the fact not all Flash-based media handle this well. If i rember correctly, there is a default for alignment for dd if not specified? Maybe thats the hickup for Fedora on EFI-Boot.

Btw.: If using RUFUS to write Fedora to HDD via USB-Adapter and then connect that HDD internally, fedora installer boots fine. ( GPT-based FAT32 )

I just want to ask if someone possibly run into this issue too and may apply to this via comment, or even one is able to spend a solution ?

THANKS FOR YOUR INTEREST .

PAETH CLAUDIUSRAPHAEL
S466531257 BOSS

pbatard, 2015-07-04:
I'll try to run a test when I have chance. So far, you're the only report I have about Fedora 22 (and I don't believe the Fedora boot detection process is the same as openSuse, which is [known to be problematic](https://github.com/pbatard/rufus/issues/478#issuecomment-87504806)).

Note however that you can force `dd` writing for ISOHybrid images in Rufus by [using <kbd>Alt</kbd>-<kbd>I</kbd>](https://github.com/pbatard/rufus/wiki/FAQ#Power_keysCheat_modes).

pbatard, 2015-08-13:
I have now reproduced the issue and will have a fix for it in the next version.

commit ab6f0ef0 (2015-08-13): [iso] fix Fedora UEFI support

* One of the ANNOYING distros that relies on an extended label to
  find the install media and is now doing that for Grub/EFI.
* Closes #547
closed 2015-08-13 by pbatard