pbatard/rufus #548: Request Feature : VHD-based Systems - Operations

S466531257BOSS, 2015-07-04:
Feature requested :

PURPOSED WRITE TO RUNNING SYSTEM VHD IF SYSTEM-VHD IS CHILD-VHD?

Scenario i can think of is :

- using RUFUS to add blends/versions/editions of OS ( in batch ) besides running system . 

TEMPLATES FOR RUFUS VHD-CREATION

Scenarios i can think of is :

- deploy windows off vhd instead of wim ( cloning ( to child-vhd ) )

- allow to write a bunch of isos to a native-boot vhd in GPT-format ( microsoft conform - bootsect/bootmgr ) to allow bundling of isos for deployment, hyper-v or USB-drives ( FiraDisk ); possibly most versatile if combined with next option ...

- allow creation of vhds with hybrid-mbr ( results in e.g.: GPT-based VHD with hybrid mbr, where P1: 
is system-reserved aligned to 1MiB to allow grub reside additionally afterwards, followed by P2 EFI- System-Partition to hold rescue/PE/Install-wim/-iso, followed by P3 Main NTFS in deployable target size with 64k Cluster-size, followed by Extended partition which render in GPT-mode to same-numbered primaries ) ( just as an idea for some possibilities )
- allow pushing wubi-installs ( e.g.: Ubuntu) to native-boot vhd's 

These are just some out of a bunch of examples that definately would enrich RUFUS, because they are not necessary for anyone, but can be added/blended into the given options without disrupting the GUI. also these examples are in general use by those that drive EFI-HYBRIDS or just VHD's, mostly developers, deployment staff, repair-your-computational-junk-facilities, etc. .

Maybe, one would like to add an idea, especially pbatard, as i am hoping for a commented thought on this.

P.s.: As i wrote about templates, that one might make Rufus even much fancier: given the option to use templates ( xml / mbr / gpt ) which can be predefined via script/cmd/xml or maybe based on given mbr/gpt off a connected disc ( e.g.: to prepare a clone to vhd )

Well, that's it for now .

THANKS FOR YOUR INTEREST .

PAETH CLAUDIUSRAPHAEL
S466531257 BOSS

S466531257BOSS, 2015-07-04:
P.s.: It's all out of lazyness ... :)

pbatard, 2015-07-04:
Feature rejected. That's __way__ too specific, and Rufus is a generic tool. I'm not going to maintain something that's only going to benefit a handful of users. Also, see [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want_I_mean_youre_a_developer__it_shouldnt_be_that_hard).

closed 2015-07-04 by pbatard