pbatard/rufus #549: Split automatic a Windows image file (.wim) for FAT32 media

gongoscho, 2015-07-04:
..to fix boot problems (GPT, NTFS and UEFI) withn windows setup´s.
Link: https://msdn.microsoft.com/en-us/library/windows/hardware/dn938329%28v=vs.85%29.aspx

pbatard, 2015-07-04:
Why would I do that when Rufus has [UEFI:NTFS support](https://github.com/pbatard/uefi-ntfs)?

There's no need to split images - Rufus __can__ boot an NTFS drive using UEFI [ever since version 2.0](https://github.com/pbatard/rufus/blob/master/ChangeLog.txt#L26).
Have you simply tried selecting NTFS in Rufus?

gongoscho, 2015-07-04:
Yes, the boot of the rufus uefi files (/efi/boot/bootx64.efi or /efi/boot/bootia32.efi) not work.

pbatard, 2015-07-04:
Can you describe the error then? I'd rather work on fixing it than add a split feature for the few people for who UEFI:NTFS doesn't seem to work. Also, please be mindful that you need to disable Secure Boot for UEFI:NTFS to work (but you can re-enable it afterwards and it won't make your installation any less "secure").

pbatard, 2015-07-04:
By the way, EFI boot for Windows 7 32-bit will NEVER work as Microsoft never designed Windows 7 32-bit to be bootable from UEFI. Only the 64-bit version will work (and even then, since Microsoft doesn't provide an actual `/efi/boot/bootx64.efi`, it needs to be worked around). Can you please try a Windows 8 ISO and see if you have the same issue?

gongoscho, 2015-07-04:
There is no Secure Boot, and i only have a Windows 7 64bit ISO.

Here is the error:[image]

With fat32 & gpt there is no problem

pbatard, 2015-07-05:
Interesting. First of all, please make sure your USB is not corrupted by running a bad blocks check in Rufus. Also, what's your motherboard model (or computer model if you didn't build it yourself)? Can you load UEFI drivers when you launch an UEFI shell? For instance, what happens if you download the [latest refind ISO](http://sourceforge.net/projects/refind/files/0.8.7/refind-cd-0.8.7.zip/download) and then use Rufus with that ISO to create a GPT/UEFI bootable USB, and run the UEFI shell from it (first option on the second line)? Can you see NTFS partitions

Please understand that, so far, you are the very first person reporting such an error, and as far as I know, the vast majority of people are booting UEFI:NTFS just fine. As such, it is __not__ worth for me investing time to add a FAT32 split feature (and then have to support it as well as educate people about it), if it's just to help one or two people who have a hardware/firmware combination that seems to prevent UEFI driver loading. I think trying to figure out why a driver cannot be loaded on your specific computer will be more helpful than just trying to sweep the problem under the carpet and use a workaround.

gongoscho, 2015-07-14:
I tested everything with the same USB 3.0 32 GB Stick and refind v0.8.7
Computer: HP ProBook 4535s

Here you see there is no problem with the default file system (UEFI-GPT, FAT32)

[image]

But, when i change the file system there is also a problem (UEFI-GPT, NTFS)
The error is not exactly the same with the win7 setup.

[image]

pbatard, 2015-11-04:
Finally managed to get `Driver did not start: [1] load error` while testing for #615.

However, this happened only after UEFI:NTFS ran once successfully, and was re-executed after exiting from the EFI bootloader from the NTFS partition. I guess the issue there is that the EFI NTFS driver will not load twice, but "fixing" this to unload the driver (in case it is already loaded), or simply ignore and continue would result in an infinite loop, so I don't see it as a proper solution.

I guess what may happen is that UEFI:NTFS works in its first instance, then the EFI bootloader on the NTFS application boots, but it then returns back to the UEFI firmware, which tries to boot again... I also checked the HP firmware to see if there wasn't an NTFS driver already available there, but there wasn't (only the FAT one), and I wouldn't expect this behaviour if there was.

Still that doesn't explain the second type or error, which is the same as with #615, and, surprise, surprise, also happens on an HP UEFI notebook... At this stage, I all seems to point that there is something very special in the HP UEFI firmwares. But short of getting my hand on such a machine, it may be very difficult to troubleshoot...

gongoscho, 2015-11-05:
Interesting, it is a HP UEFI - Problem. I think we close this issue, and you concentrate all your power to the #615 issue. Thank you for your time :)

pbatard, 2015-11-05:
Yeah, sorry it took so long to get back to that issue, but as far as I could see, it's very dependent on the machine being used. I agree that, at this stage, #615 and this issue seem to revolve around the same problem, and I'm still looking at what could be achieved for #615, so I will close this issue.

closed 2015-11-05 by pbatard

olivermarshall2, 2017-03-28:
Hi

We're seeing the same issue on a Dell PowerEdge T330. Is there a fix at all?

Olly

pbatard, 2017-03-28:
@olivermarshall2, do you have a screenshot of your output?

harksolution, 2018-08-09:
My problèmes
UEFI:NTFS (*64)*****

harksolution, 2018-08-09:
Voilà mon problèmes