pbatard/rufus #551: SD card not detected as a drive

8ctopus, 2015-07-09:
Thank you for a great tool!

I just noticed that the latest version of Rufus 2.2 does not recognize the SD card reader I have on my HP EliteBook 8570w laptop. Here's the Rufus log:

Rufus version: 2.2.668
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.

The SD card reader has ID PCI\VEN_197B&DEV_2391

I also confirm that I can fully access the drive using Windows Explorer.

pbatard, 2015-07-09:
```
The SD card reader has ID `PCI\VEN_197B&DEV_2391`
```

Your card reader is __PCI__. Rufus is an __USB__ tool (hence its name). I have no plans to add detection for PCI devices as it would enable the listing of internal disks and result in users formatting a valuable drive by mistake.

Please use an USB adapter.

pbatard, 2015-07-09:
Also see [here](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_support_internal_HDDsSSDs_or_non_USB_card_readers).

closed 2015-07-09 by pbatard

8ctopus, 2015-07-09:
Sorry my bad. I should have checked closer. I understand your concern about accidental wipes of important drives but would definitely be useful for PCI SD card readers. I'll see if I can add this myself.

Saga-Lout, 2016-07-30:
I know this is an old thread but in case it helps anyone, I needed to have Kali Pen Test on a bootable SdCard and popped the card into a USB carrier.  Rufus picked it up and it worked perfectly, as it always does.  

Thanks pbatard for a splendid utility.

pbatard, 2016-07-30:
Thanks for the kind words. :smile: 

As the Changelog indicates, Rufus 2.8 [added support for PCI card readers](https://github.com/pbatard/rufus/blob/b204065f888968e35f63c2eb4db6089b104a5bed/ChangeLog.txtt#L23), which was [further expanded in 2.10](https://github.com/pbatard/rufus/blob/b204065f888968e35f63c2eb4db6089b104a5bed/ChangeLog.txt#L3)

fallais, 2017-09-07:
You can also use `Win32 Disk Imager`

pavman42, 2019-03-24:
As of 3.4 Rufus doesn't pick up internal SDCard reader (Thinkpad).  Had to put it into a usb reader to have it see the drive.

I find it a bit odd, but would be nice if it could recognize removable drives rather than limiting it to merely usb devices.

pbatard, 2019-03-24:
> As of 3.4 Rufus doesn't pick up internal SDCard reader (Thinkpad)

Please provide your full log after pressing <kbd>Ctrl</kbd>-<kbd>.</kbd>.

pavman42, 2019-03-24:
I can't find any logs.  I've tried both editions from rufus.ie (regular and portable). I tried using the non-portable, but when it didn't detect, switched to portable.  Then used the reader (as I had originally used this on a system w/o a card reader).
I do see these logs when I click the icon:

`Rufus x86 v3.4.1430 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found`

[image]

pbatard, 2019-03-24:
> I can't find any logs.

I can assure you they are there, and you can access them by either clicking on the log button (last small button left of _START_ or by pressing <kbd>Ctrl</kbd>-<kbd>L</kbd>.

Oh and the means of accessing the logs are provided to you when you open a new issue, so you would have gotten that information if you had done just that instead of piggybacking on an old issue (which you should always avoid doing).

> Note: this system had an older version

Always use the most recent version. But the log button has been present from the beginning, so no matter your version you should have been able to access it.

> I do see these logs when I click the icon:

Yes, that is what I want. Except you should make sure you click <kbd>Alt</kbd>-<kbd>.</kbd> (sorry, earlier shortcut I mentioned said<kbd>Ctrl</kbd>-<kbd>.</kbd> but it's really <kbd>Alt</kbd>-<kbd>.</kbd> you should use).

If you do that you should get a whole lot more information in your log. Then you can just copy/paste what you have.

pavman42, 2019-03-26:
Just trying to help you improve your product, complaining because I failed to open a bug seems like a petty way to say 'I don't care about my user base, you should learn to use git's bug reporting.'

Just be glad I'm trying to help to contribute to YOUR software.  I appreciate it, but you could work on your customer service skills.  There are other tools, this just seemed the most convenient (and it is, thanks).
`Rufus x86 v3.4.1430 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0409
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&275d4bcc&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[066]: USB\VID_8087&PID_0024\5&1E58A885&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub20#4&290d2617&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[241]: USB\VID_8087&PID_0024\5&29765D24&0&1
Processing Hub 4:
  Hub[3] = '\\?\usb#vid_8087&pid_0024#5&1e58a885&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[050]: USB\VID_147E&PID_2016\6&1906F761&0&3
  Found ID[073]: USB\VID_04F2&PID_B217\6&1906F761&0&6
Processing Hub 5:
  Hub[4] = '\\?\usb#vid_8087&pid_0024#5&29765d24&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[183]: USB\VID_056A&PID_00E6\6&15F39A7A&0&5
Processing IDs belonging to 'USBSTOR':
Processing 'IDE' device:
  Disabled by policy
Processing 'IDE' device:
  Disabled by policy
Processing 'RISD' device:
  Disabled by policy
0 devices found`

FWIW, when I first tried this w/o running as administrator, it blue screened windows (I was surprised.... just letting you know.  Once I ran as administrator, it worked fine (albeit still didn't see the built-in sdcard reader)).

pbatard, 2019-03-26:
> seems like a petty way to say 'I don't care about my user base, you should learn to use git's bug reporting.'

It's just a way to say _"I wish I didn't have to waste time explaining stuff that I made sure to explain when users create a new issue"_ such as how to access the log. You are busy, I am busy. Anything that saves me time is welcome which is why I'd prefer if people didn't piggyback on 3+ year old issues and instead created a new one. This way, they'd go through the new issue checklist and it would save time for __everybody__.

Also, you get the utility for free, so I owe you absolutely nothing. And if you want to interpret that as yet another "proof" that I don't give a damn about application users... from within an issue tracker where there is ample evidence of the contrary, so be it.

As a developer, time is what I am fighting against, because I __never ever__ have enough of it to satisfy all the requests I receive on a daily basis. So, yeah, when I feel like someone could have done things differently to avoid time wastage, I have no issue stating so, even if the end result is that said user will be displeased...

Anyway, thanks for providing the log. It contains the information I was looking for: `RISD` which is the ID I should just have to add [here](https://github.com/pbatard/rufus/blob/master/src/dev.c#L343-L344) to get your card reader detected. I'll do that for the 3.5 release.

> when I first tried this w/o running as administrator, it blue screened windows 

Can you indicate exactly how you did that?

First of all, you're the very first person, in hundreds of millions of downloads, to report that Rufus could cause a BSOD, so I a __exceedingly__ interested in trying to replicate your test conditions.

Now, I know that, if you are running Windows 7, then some images (such as FreeBSD) will BSOD Windows regardless of the method you use to create the USB, or even if you created the USB on Linux using `dd` and then plug it into Windows, and I have had some reports from people believing that Rufus was the issue, whereas it's just Windows 7 and earlier containing a bug that makes it BSOD when it reads garbage data in the backup GPT table (which it will when you try to restore a GPT partitioned DD image to a drive of a different size). So obviously the first thing I'll ask is: were you also playing with a specific image or USB drive when that BSOD occurred?

If not, I'll need some details about what you mean w/o running as administrator:
Does that mean running from an account, not called `Administrator` that also doesn't have administrative rights (in which case Windows should refuse to launch Rufus since it requests elevation)? Or does it mean running from an account with administrative rights, but not called `Administrator` vs running with the actual `Administrator` account?

Once again, I'm afraid I'm going to give a more precise set of details on how you conducted the test that resulted in a BSOD exactly...

pavman42, 2019-03-28:
Ubuntu 18.04.LTS server (think this was LIVE and not the server version I had to use later).
I put it in the usb card reader, opened rufus (no admin privs), hit create and within a second or two BSOD.  I chalked it up to not being admin and when I rebooted re-ran it as admin without issues.  This was local (I have images on a samba drive on my network that worked just fine w/ rufus sans warning about not being able to boot w/ older boot loader).

I can try to find the mem dump & window logs, but not sure if I let it run or just rebooted.  Haven't had issues since, but I run it as admin privs now.  I do use comodo (which usually runs things contained initially but because I had ran 2.07 previously (but not that day), I think the exe was on the do not isolate list).

pbatard, 2019-03-28:
> opened rufus (no admin privs)

How? First, there should have been a Windows prompt asking you for elevation, and even if you managed to run it without admin privileges, all you should have gotten should be a [prompt telling you that Rufus cannot run without elevation](https://github.com/pbatard/rufus/blob/master/src/rufus.c#L3063-L3072). I'm pretty sure I tested this on Windows 7 with a regular account at the time, so what you are reporting is very puzzling...

Can you detail how exactly your account was created on Windows 7?

You will agree that it would be quite a feat for a non elevated application to produce a BSOD, so I am exceedingly curious to have more details on how you got that issue, and whether you can replicate it...

pbatard, 2019-03-28:
Also:

> no admin privs, but it's an admin user

What does that mean?

pavman42, 2019-03-28:
Right click -> run as administrator.  That's what I mean.

I'm not sure how to explain it beyond that.  I turned off UAC a long time ago and this user is in Administrators.  Occasionally I've had issues running cygwin and other cli's (powershell) as non-admin privs as well as some installs, so I end up cancelling then re-running w/ right-click -> run as administrator.

One thing that might be pertinent: last year I migrated the stock 2.5" ssd to msata and added a "D:" drive, I then moved everything to D less the C:\Windows dir (e.g. D:\Users\User\Desktop), replacing registry entries to reflect the drive.  I also gave the msata ssd the drive letter 'A:' (which may or may not cause issues with older software).

pbatard, 2019-03-28:
> Right click -> run as administrator. That's what I mean.

Yes, I got that. But you should not have to do that since Rufus requires elevation so if you have UAC turned off, it automatically start in the same manner as if you had manually done the above. My understanding is the only difference between UAC on and UAC off should be that with UAC on you are prompted on whether you accept elevation, whereas with UAC off, it is granted automatically.

Anyway, I'll see if I can test with UAC turned off on Windows 7 and see what comes out.

pavman42, 2019-03-28:
Good luck, I'll see if I can find the logs based on timestamps from when I first tried this.

pbatard, 2019-03-28:
Btw, I released Rufus 3.5 [earlier today](https://rufus.ie/), which I believe should now detect your SD card reader.

pavman42, 2019-03-28:
Cool.
I don't think it was related.  I had woken the computer from sleep and I see a different application throwing this error during the time-frame the crash happened.  Between the razr software garbage and quickbooks, I wouldn't be surprised if either of those caused it and it was just coincidental to when I started writing the image.

Faulting application name: QBW32.EXE, version: 25.0.4017.2506, time stamp: 0x5a967bd0
Faulting module name: MSVCR120.dll, version: 12.0.21005.1, time stamp: 0x524f7ce6
Exception code: 0xc0000417
Fault offset: 0x000a46bb
Faulting process id: 0x17b8
Faulting application start time: 0x01d4e59590fe2b96
Faulting application path: C:\apps\QuickBooks Pro 2015\QBW32.EXE
Faulting module path: C:\Windows\system32\MSVCR120.dll
Report Id: d5db6a5f-5188-11e9-8776-f0def1b6b01a

pavman42, 2019-04-02:
Thanks for fixing the card reader issue so quickly!