pbatard/rufus #552: Rufus do not work with MS-DOS bootable ISOs

JonasNo, 2015-07-24:
On various sites there are instructions on how to make an ms-dos bootable usb.

See here:
http://blogs.technet.com/b/csstwplatform/archive/2012/06/26/how-to-create-a-ms-dos-bootable-usb-flash-drive.aspx

and here:
http://www.sysprobs.com/how-to-create-bootable-dos-usb-disk-in-windows-8-8-1

Both use iso's from this site:
http://www.allbootdisks.com/download/iso.html

I have not tested them all. Rufus do not work with these two iso's:
http://www.allbootdisks.com/downloads/ISO/AllBootDisks_ISO_Image_Downloads25/Win98SE_bootdisk.iso
http://www.allbootdisks.com/downloads/ISO/AllBootDisks_ISO_Image_Downloads25/Win98_bootdisk.iso

Both rufus_v1.2.0 and rufus_v1.1.7 reports Unsupported ISO
"This version of Rufus supports bootable ISOs
based on 'bootmgr/WinPE' ir 'isolinux'.
This ISO image doesn't appear to use either..."

The instructions in one of the sites clearly say "Download rufus utility, Rufus v 1.1.7 without FreeDOS Support (154KB) from here".

Do not know if the site is broken or Rufus.

Either way 7-zip reports "Incorrect big-endian headers" on both ISOs.

It did work before based on the instructions.

pbatard, 2015-07-24:
Rufus only supports creating __blank__ DOS bootable USB drives. This means that you should not select an ISO if you want to create a DOS disk - just select FreeDOS or MS-DOS in the dropdown.

Then you can copy the content from your DOS ISO onto the USB created by Rufus. Note that you may have to edit the DOS boot parameters so that the USB behaves like the DOS ISO (which is why Rufus does not support DOS ISOs as this conversion is tricky to automate).

Also see https://github.com/pbatard/rufus/wiki/FAQ#Im_trying_to_work_with_an_ISO_but_Rufus_says_its_unsupported

closed 2015-07-24 by pbatard

JonasNo, 2015-07-30:
Thanks for answering. Yes, I did read the FAQ before posting the issue.
I'm not sure how I did it but I manage to get it working.

Still wonder why those guides exist. Followed the instructions to the letter and it didn't work.

Note that using FreeDOS for tools written specifically for MS-DOS such as bios updaters, etc is very risky. Many say that and I'm not a risk-taker with expensive hardware.

pbatard, 2015-07-30:
FreeDOS is very mature and, IMO, more trustworthy than MS-DOS in terms of stability. I would actually recommend using FreeDOS over MS-DOS for flash updates.