pbatard/rufus #556: Allow multiple ISO in same drive

nyroDev, 2015-07-28:
I don't know if it's possible and the work that it will require, but it will really great if we can create a boot with multiple ISO.
When booting, it will let me choose which ISO I want to load in Grub-like menu and then boot it from there.
Regarding the drive creation:
1) create a USB drive with an option "choose iso menu"
2) Add an ISO with an option "Add element to existing drive"

This will be really helpful in case of recovering computer. Only 1 USB drive to rule them all.

pbatard, 2015-07-28:
https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiboot_support

closed 2015-07-28 by pbatard

nyroDev, 2015-07-28:
Thanks for the quick reply and the link you provided.