pbatard/rufus #558: bcdboot errors during Windows 10 To Go creation from Windows 8.1

pbatard, 2015-07-30:
Trying to create a Windows 10 To Go drive (MBR/UEFI) from a relatively clean Windows 8.1 machine results in a stream (20+) of `The program can't start because api-ms-win-core-libraryloader-l1-2-1.dll is missing from your computer. Try reinstalling the program to fix this problem.` popups as well as a single `The program can't start because api-ms-win-core-version-l1-1-1.dll is missing from your computer. Try reinstalling the program to fix this problem.`.

The potential `bcdboot` error was meant to be addressed with a fallback in cc6234296dd04d6512b169f87cb18d309b3773c2, but unfortunately, a user still needs to click through all these popups, which is exceedingly annoying.

We should probably try to use a UI popup killer as we do in `CloseFormatPromptThread()`, to eliminate those windows and detect that the command fails for fallback.

Also, we should add the `/v` option to our `bcdboot` invocation to try to get more verbose output, such as this one from the failed `bcdboot` from Windows 10 above:

```
BFSVC: ServiceBootFiles MuiOnly:n Res:y Fonts:y BootMgrOvw:nBootStatOvw:n DbgTransports:y CreateSetupObj:n
BFSVC: Using source OS version a000028004000
BFSVC: Copying boot files CopyBootManager(Yes) E:\Windows\boot\PCAT -> \\?\GLOBALROOT\Device\HarddiskVolume8\Boot
BFSVC: SetNamedSecurityInfo failed! Error code = 0x2
BFSVC: Unable to open file \\?\GLOBALROOT\Device\HarddiskVolume8\Boot\bootmgr for read because the file or path does not exist
BFSVC: SetNamedSecurityInfo failed! Error code = 0x2
BFSVC: Copying font files from E:\Windows\boot\Fonts to \\?\GLOBALROOT\Device\HarddiskVolume8\Boot\Fonts...
BFSVC: Copying resource files from E:\Windows\boot\Resources to \\?\GLOBALROOT\Device\HarddiskVolume8\Boot\Resources...
BFSVC: Servicing debugger files
BFSVC: List of debugger files is empty
BFSVC: BfsInitializeBcdStore flags(0x0000021c) RetainElementData:n DelExistinObject:n
BFSVC: Opening template from \Device\HarddiskVolume8\Windows\System32\config\BCD-Template.
BFSVC: Opening store from \Device\HarddiskVolume8\boot\BCD
BFSVC: Creating General objects.
BFSVC: Creating Resume object.
BFSVC: Creating MemTest object.
BFSVC: Creating OsLoader object.
BFSVC: OsLoader identifier: {e6ee3264-365c-11e5-82f8-acfdce2112d0}
BFSVC: Removing duplicate entries.
BFSVC: Removing duplicate object {e6ee3264-365c-11e5-82f8-acfdce2112d0}
BFSVC: Create BOOTMGR object RetainBootDefault:n
BFSVC: Setting {default} to {e6ee3264-365c-11e5-82f8-acfdce2112d0}
BFSVC: Cleaning up debugger settings.
BFSVC: ServiceBootFiles MuiOnly:n Res:y Fonts:y BootMgrOvw:nBootStatOvw:n DbgTransports:y CreateSetupObj:n
BFSVC: Using source OS version a000028004000
BFSVC: Copying boot files CopyBootManager(Yes) E:\Windows\boot\EFI -> \\?\GLOBALROOT\Device\HarddiskVolume8\EFI\Microsoft\Boot
BFSVC: Creating Recovery directory.
BFSVC: Copying font files from E:\Windows\boot\Fonts to \\?\GLOBALROOT\Device\HarddiskVolume8\EFI\Microsoft\Boot\Fonts...
BFSVC: Copying resource files from E:\Windows\boot\Resources to \\?\GLOBALROOT\Device\HarddiskVolume8\EFI\Microsoft\Boot\Resources...
BFSVC: Servicing debugger files
BFSVC: Copying boot debugging files from E:\Windows\boot\..\system32 to \\?\GLOBALROOT\Device\HarddiskVolume8\EFI\Microsoft\Boot (kd_*_*.dll)
BFSVC: Copying boot debugging files from E:\Windows\boot\..\system32 to \\?\GLOBALROOT\Device\HarddiskVolume8\EFI\Microsoft\Boot (kdstub.dll)
BFSVC: Done servicing debugger files.
BFSVC: BfsInitializeBcdStore flags(0x0000021c) RetainElementData:n DelExistinObject:n
BFSVC: Opening template from \Device\HarddiskVolume8\Windows\System32\config\BCD-Template.
BFSVC: Opening store from \Device\HarddiskVolume8\EFI\Microsoft\Boot\BCD
BFSVC: Creating General objects.
BFSVC: Creating Resume object.
BFSVC: Creating MemTest object.
BFSVC: Creating OsLoader object.
BFSVC: OsLoader identifier: {f9f9602c-365c-11e5-82f8-acfdce2112d0}
BFSVC: Removing duplicate entries.
BFSVC: Removing duplicate object {f9f9602c-365c-11e5-82f8-acfdce2112d0}
BFSVC: Create BOOTMGR object RetainBootDefault:n
BFSVC: Setting {default} to {f9f9602c-365c-11e5-82f8-acfdce2112d0}
BFSVC: Cleaning up debugger settings.
BFSVC: Opening recovery store from \Device\HarddiskVolume8\EFI\Microsoft\Recovery\BCD
BFSVC: Creating General objects.
BFSVC: Create BOOTMGR object RetainBootDefault:y
Boot files successfully created.
```

Walkman100, 2015-07-30:
> Trying to create a Windows 10 To Go drive (MBR/UEFI) from a relatively clean Windows 8.1 machine

I thought you were supposed to only be able to create Windows To Go drives with a version equal to or lower than the version of Windows that you're running?

pbatard, 2015-07-31:
And this issue is precisely about trying to remove part of this limitation in future versions of Rufus.

Walkman100, 2015-07-31:
That would be really awesome if/when you get it to work.

pbatard, 2015-08-22:
Part of this issue seems to have to do with compiling with MinGW. When compiling with VS2015 or VS2013, none of these errors appear. But the exact same source compiled with MinGW will produce them.

At this stage, I am neither ready nor willing to switch to compiling Rufus using non-free tools, so a MinGW solution or workaround will have to be found.

There is a possibility that the issue has to do with Windows XP compatibility, since VS has `WINVER` set to `0x601` (Vista or later) whereas we set `WINVER` to `0x501` (XP).

pbatard, 2015-08-23:
This is even more tricky than it looks. It's only in some circumstances that the VS compiled version will not produce the errors above. But we do get them from the commandline regardless, so I don't think I want to spend time investigating what's going on, especially as it only happens after applying a full image, which is __very__ time consumming. Instead, I will force the use of `bcdboot.exe` from the host always, as this is the safest approach anyway.

commit d1eccbd1 (2015-08-23): [togo] fix bcdboot errors

* Closes #558
closed 2015-08-23 by pbatard