pbatard/rufus #561: ISO Image Extraction Error

dkramer47, 2015-08-01:
I'm trying to create a bootable Kali Linux USB on a Windows 10 system. I have searched for hours trying to figure this out, but I haven't been able to find anything on it, so I'm posting... hopefully it's something I'm just missing.

Every time I use Rufus i get the error "Error: ISO image extraction failure". I'm using Rufus 2.2.668
Here is the log:

Rufus version: 2.2.668
Windows version: Windows 10 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00060225
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16007561216 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
libcdio: from_733: broken byte order
Will use '/isolinux/isolinux.cfg' for Syslinux
Detected Isolinux version: 6.03/20141209 (from '/isolinux/isolinux.bin')
ISO label: Kali Live
  Size: 2992195584 bytes
  Has a >64 chars filename: Yes
  Has Symlinks: Yes
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (6.03)
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: kali-linux-1.1.0a-amd64.iso
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.03/20141209/' for Syslinux installation

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{0531d3c8-37c2-11e5-826a-f0761caa5445}\
Installing Syslinux 6.03...
Using existing './rufus_files/syslinux-6.03/20141209/ldlinux.sys'
Using existing './rufus_files/syslinux-6.03/20141209/ldlinux.bss'
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{0531d3c8-37c2-11e5-826a-f0761caa5445} for write access
Successfully wrote Syslinux boot record
Successfully remounted Volume{0531d3c8-37c2-11e5-826a-f0761caa5445}\ on F:\
Copying ISO files...
Extracting files...
Disc image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: F:\.disk\archive_trace (1 bytes)
  Error reading ISO9660 file /.disk/archive_trace at LSN 1345466
Created: F:\syslinux.cfg

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0019D288
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16007561216 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

P.S. I do not believe windows 10 would be the problem (at least not this problem).
    as Rufus seems to work perfectly normal other then this, but i may be wrong.

Thanks in advanced!!

deadair3210, 2015-08-01:
You likely have a bad iso, seems to be the case due to errors with the byte order and with finding files in the iso

dkramer47, 2015-08-01:
That's what i thought at first too, so i re-downloaded the ISO and tried again, same error.

deadair3210, 2015-08-02:
Possibly a bad build of the ISO, try downloading an older version or from a different website.

dkramer47, 2015-08-02:
Okay, will let you know how it works.

dkramer47, 2015-08-02:
Yes! you were right! i downloaded the torrent instead of the direct ISO and it worked! Thank you!

closed 2015-08-02 by dkramer47