pbatard/rufus #563: Check device for bad block will not show any message when it is finished

Sopor-, 2015-08-01:
I was running a bad block check and when it was finished i couldn't see if there was any errors or not. I was forced to open the log window and try to find any information about it. All log entries are in English so if the user run Rufus in any other language then English (maybe the user don't understand English?) he will have to go through all the log entries to find this in a language he don't understand:
Bad Blocks: Checking from block 0 to 124735487
Bad Blocks: Using offset 0 for fake device check
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)

pbatard, 2015-08-02:
The problem with displaying a message is that then the format operation will not start right away. Remember that the bad block check is part of the whole formatting and bootable USB creation, and may take a long time to complete. So if I were to display a message, I would have to keep the process running in the background which could confuse people (especially if they see that message after everything is complete).

I'm sorry, but I disagree with your assumption that people will not be smart enough to understand that, if they don't see any error reported during the bad blocks check phase, then it means that there are no bad blocks. So far I have not received any complaints about this.

So I am going to dismiss this enhancement request too, sorry.

closed 2015-08-02 by pbatard

Sopor-, 2015-08-04:
Noting is obvious in the computer world. If no message is showing after a 3 hours test run it seems to me that something could be wrong. Maybe Rufus have crashed or the test stopped... You can't be serious on this and assume that if no message appeared everything was okay? I only run the test and when it was finished after 3 hours it looks exactly the same as before i started the test.

pbatard, 2015-08-04:
There is a __progress__ bar giving you an idea of how far along the bad blocks checks and formatting operation is. If there is any kind of error during the bad blocks heck, the progress bar will turn orange or red, and stop progressing. If you can demonstrate that, when you started your test, the progress bar didn't reset to zero, then I might look into this request. Otherwise, sorry, but your statement that "it looks exactly the same as before" is wrong. It will NOT look as before during the test due to the very obvious progress bar getting filled.

BAM5, 2019-02-10:
Want to point out that the progress bar not turning orange or red doesn't mean anything to someone unfamiliar with the program. I too was skeptical about whether the device had any bad blocks since I got no feedback from the program about it. If I tell software to run a certain test, I would like the results of that test displayed, whether good or bad. Simply omitting the test results if there were no issues is not what I would expect. It doesn't have to halt the process until the user acknowledges the results, just a "details" collapsable text area that's not as verbose as the log would do. You have to realize that the users of the program didn't write the program and don't know the specifics of how it operates.

pbatard, 2019-02-10:
> I too was skeptical about whether the device had any bad blocks since I got no feedback from the program about it.

So basically you're saying that you're not trusting the application to tell you very explicitly when there is an error.

Just like Rufus is not a partitioning application, Rufus is not a bad blocks application. The bad blocks check is simply provided as a courtesy (and, if you look at the UI, you'll see that it is indeed __hidden__ by default and only provided after a user displayed the Advanced Options, therefore it's not something that's been designed to feature prominently). Therefore, the only result that matter is that Rufus will display an error if error during the bad blocks check there is, which it most certainly does. But because it is not designed as a bona fide bad blocks checking application, with all the bells and whistle one can expect from such a program, anything else but the reporting of errors becomes superfluous.

If I wanted to keep users happy during the bad blocks check, I'd make Rufus display a grid similar to what the old Windows Defragmenter used to do, that shows precisely which block is being checked, and where blocks change colour once they have been validated, as well as other UI niceties. However, since I have more critical things to do, with increasingly limited time to do them, and __anything__ that requires UI change automatically becomes a multi-month operation, then, when the application does fulfil its mandate of reporting bad blocks when there exist one, I will continue to consider that Rufus' job in terms of bad blocks reporting is achieved, and that anything that would be "nice to have" and that goes beyond that can be left to the wishful _"When I have nothing more important to do"_ time...

So, please let me make that absolutely clear again: While Rufus can do bad blocks checking is only provided as a courtesy and __Rufus was never designed to be a full fledged bad blocks checking application__ (Again in the same manner as, while Rufus can partition disks, Rufus is not designed as a generic disk partitioner). If you are unhappy with that, and the fact that the only notice you'll ever get from Rufus about the a bad blocks check is if it fails, then I will respectfully ask that you use a proper bad blocks checker, with all the bells and whistle you can expect from an application that was designed precisely for this job, rather than one where the feature was only added as a "by the way". There's only so much time a developer can spend on "by the way" features.

Still, if some __other__ developer has disposable time on their hand, I'll be happy to accept a nice grid feature display when the bad blocks check is running, to keep you complaining lot happy... 😄

BAM5, 2019-02-10:
> Just like Rufus is not a partitioning application, Rufus is not a bad blocks application.

> Therefore, the only result that matter is that Rufus will display an error if error during the bad blocks check there is, which it most certainly does (through an **explicit alert dialog** and not just the progress bar changing colour as you seem to assert).

This is exactly my point. I don't know what would happen if it detected bad blocks. Or what would happen if it didn't. There's no feedback, and no information on what to expect. I had to google to find out what would happen and came across this issue. So I figured I'd throw in my two cents.

It's like going to the doctor and asking for an medical test and then the doctor never contacts you about it again unless you go and pester him about it. Rufus is the doctor. Sure, I could go to a lab tech and get rna sequencing data for whatever (metaphor for using a program specifically used to check bad blocks), but all I want to know is if there is a bad block or not, and be told so when the test is complete, or be told ahead of time what sort of feedback to expect.

I don't care about feedback _during_ the block check test. What currently exists is perfectly fine for an application that is not specifically made to check for bad blocks. I just want some feedback about the test _once everything completes_, because, once again, rufus is not a specialty software for checking for bad blocks, and thus I don't know if it will just continue on and write the image anyway. And I don't want to sit around for 2 hours on a 2 pass check and watch the status bar for errors. This is the point we're both trying to make.

Easy & quick solution to improve ux: Once the bad block check checkbox is checked, pop up a message box that says "Rufus will check for bad blocks and if none are found it will continue with the write process. If bad blocks are detected the process will be halted and an **explicit alert dialog** will appear. If you want more detailed information on the test then check the log."

pbatard, 2019-02-10:
>  I just want some feedback about the test _once everything completes_, because, once again, rufus is not a specialty software for checking for bad blocks, and thus I don't know if it will just continue on and write the image anyway.

As it is written, this doesn't seem to make sense. You say you want feedback AFTER everything completes yet feedback that tells you BEFORE everything completes what it's going to do BEFORE everything is done? Thankfully, you are more or less clarifying what you want later, but please try to formulate things a bit better. 

> And I don't want to sit around for 2 hours on a 2 pass check and watch the status bar for errors.

You don't have to, and I think you are being very disingenuous if you want to posit the idea that the default expectation for the vast majority of users is that they should watch the status bar because application is not going to pop up an alert AS SOON AS it detects an error, which is __exactly__ what it does right now. Reports about actual bad blocks are done with an alert, not the progress bar, which, IMO, is what any reasonable user will expect. And likewise, a reasonable user will expect that, as long as the application does not throw an alert, then it __logically__ means that it hasn't found any bad blocks.

> Easy & quick solution to improve ux: Once the bad block check checkbox is checked, pop up a message box that (...)

I don't write user interface for the minority of users who expect hand holding about a bad blocks check which is both provided as a courtesy and as an __advanced__ feature. As a matter of fact, I don't provide much help for any advanced features, really. If we go this way, then next, you're going to ask me to pop up a message box that explains what the heck the _"BIOS ID"_ number in _"Use Rufus MBR with BIOS ID"_ does, and I'll have to add a 5 paragraph technical blurb, translated in about 40 languages, that explains what this super specific feature, that was mostly designed to help with XP installation, is meant to do.

Or how about what the heck SLC, TLC or MLC mean for the choice of patterns for the bad blocks check, which is the precise feature you are recriminating about not being clear enough? Surely, if we start providing basic hints for people who don't see reasonable software usage expectations (such as telling them that, __of course__, Rufus will throw an alert as soon as a bad blocks are detected) as fairly obvious, then we can't stop there, and we also need to explain the whole shebang about the much more obtuse details of what Rufus does. And there are quite a few more I could list here...

Well, as a developer of software that you get absolutely for free, who has to make choices about what they can and cannot spend their limited development time on, the cost/benefit ratio of catering for a minority of users is simply not high enough to add every single hand-holding feature under the sun, even more so when you also have to consider the implicit cost (and I'm not talking money here) of translating any UI messages, which never __ever__ boils down to _"just tell the translators that a new message has been added and wait for them to reply with their translation"_.

Therefore, if you or other users don't want to accept that all of the advanced features of Rufus are provided "as is" and are not designed to provide hand-holding for the minority of users who don't have what I would qualify as _reasonable expectations_ with regards it comes to their usage, I will kindly ask that you use a different software.

pbatard, 2019-02-10:
Also, speaking from experience on the length I went through to advise users that, if they have any issue writing an ISOHybrid image in ISO mode, then they should retry writing the same image in DD mode, in an alert that pops up at the precise location where you'd like me to insert your "Just so you know, this is how the bad block check is going to work" pop up, and how it seems to be completely ignored most of the time, because it appears a large number of users simply don't read the content of alert boxes (especially if multiple alerts are going to appear in sequence), I can tell you that the one thing you want to be careful about is add yet another alert that is going to be mostly ignored by the exact group it is designed to help.

I have spent enough time "berating" people on reddit and elsewhere for not paying any heed to that darn _"If it doesn't work in ISO mode, remember that you should retry in DD mode"_ to be painfully aware that inserting yet another alert, to try to guide the very users who need their hand held, is genuinely not as effective a means to help said users as people who never had to develop UI apps tend to think it is.

BAM5, 2019-02-11:
Hey, just throwing my two cents in. As a developer myself I don't take anything for granted and if I don't get specifics about what will happen or any feedback on what has happened I don't assume one thing or the other. I would hope that the software would operate as you've designed and described when I am not presented with any feedback, but since I didn't explicitly know how it works beforehand I needed to look at logs & google to be sure.

> I have [spent](https://old.reddit.com/r/linux4noobs/comments/aixvgw/live_usb_cant_load_in/eermi19/?context=0) [...]

Heh, well I usually read those as they are usually there for a purpose. I'm not the type to be tricked into installing mcafee while installing some other software. 🙂

Anyway, cheers, and thanks for the software.

pbatard, 2019-02-11:
> As a developer myself I don't take anything for granted and if I don't get specifics about what will happen or any feedback on what has happened I don't assume one thing or the other.

Then you'll have to explain to me how, if you selected bad blocks check and the application does not report any explicit error, and whereas the application indicates that it was actively checking for bad blocks with text on its status bar as per the screenshot below, then your first reaction is to __assume__ that the application may forget to also explicitly report when it finds a bad block:
[image]
then 

In other words, you are __assuming__ that the software was poorly designed.

Well, that's great and all, but frankly I have better things to do than cater of people who assume, without reason, that developers don't know what they're doing and would somehow design their application to fail to report errors, with alerts and whatnot, when errors there exist.

At this stage, I'm getting a strong feeling that a lot of people haven't even bothered to look at how the bad blocks check operate, and think that, as opposed to what I am __showing__ above, users have to guess what's going on, whereas the application always explicitly indicates what it is doing (including immediate feedback on any bad block error it found) in plain English on the progress bar, and __of course__, you do get feedback during lengthy operations.

So could you guys please maybe spend some time actually properly looking at how the application operates, rather than __INFER__ erroneous behaviour from the comments of people who, IMO, don't appear to have been looking that closely at what the application does?

Or are you going to declare once again that, even as the application is displaying how much of the bad blocks check it has completed, and how many errors it has found up to the moment it has completed, then the moment this data disappears from the UI, the only reasonable assumption one should make is that, maybe, the application might has still found some bad blocks and failed to report them, __because__ software can only be expected to be poorly designed up to the point of failing to halt and report explicit errors when it finds them?

Well, good luck rying to get a developer add a feature if your only justification for it is: _"I think it's on;y fair that everybody should assume that your software was designed poorly and will fail to report errors..."_

Or do you not see the issue here?

> Heh, well I usually read those as they are usually there for a purpose.

And the fact that __you__ read alerts disproves the point which I went the trouble of demonstrating with a lengthy string of examples precisely because I expected that a few people would retort _"Well I'm not doing that, so, clearly, other people aren't either"_ because...?

I am not designing the software for __you__ alone. I am designing the software for you... and everybody else. Therefore I will continue to have to consider behaviour that you may not exhibit as an individual user, but that other users do.

BAM5, 2019-02-11:
> In other words, you are **assuming** that the software was poorly designed.
> Well, that's great and all, but frankly I have better things to do than cater of people who assume, without reason, that [...]

My reasoning is simple. Humans are stupid, bugs happen. Now that's a large generalization and I don't put you in that category specifically, but since I don't know you, nor have I reviewed your code, I will assume that you're a mere mortal and capable of making mistakes. Plus, I've also helped many a person through some cs classes and it doesn't give me a lot of confidence.

> At this stage, ... [I] think that, as opposed to what I am **showing** above, users have to guess what's going on, whereas the application always explicitly indicates what it is doing...

Oh absolutely it does, like I thought I stated clearly earlier it's perfectly fine at keeping the user apprised of what it's doing in the moment. I kept the window on a monitor for up to ~10% with 0 errors, then I went to bed. Woke up and since I wasn't greeted with any results for the block check I had to do some investigating, because, I do in-fact assume people are not only capable of making mistakes, but a lot of them do on a regular basis. But that's my bias.

> rather than **INFER** erroneous behaviour from the comments of people who, IMO, don't appear to have been looking that closely at what the application does?

Well I mean, that's what I'm saying. I don't have anything official to go off of. I had to search google and came across this issue. This has always been my point. I don't have extensive knowledge of how the program operates because I haven't used it for a lengthy amount of time nor have I dug through the source code. I just came across this issue and thought I'd say that OP isn't alone.

> And the fact that **you** read alerts disproves the point which I went the trouble of demonstrating with a lengthy string of examples precisely because I expected that a few people would retort _"Well I'm not doing that, so, clearly, other people aren't either"_ because...?

Well I mean, that's somewhat of a statistical fallacy isn't it? You don't have exact numbers on who reads / ignores dialog boxes so how can you say they do nothing?

> I am not designing the software for **you** alone. I am designing the software for you... and everybody else. Therefore I will continue to have to consider behaviour that you may not exhibit as an individual user, but that other users do.

Oh for sure, that's your prerogative. I just came across this ... I've been saying issue, maybe I should be saying "topic?"... and thought I'd share my experience as well to help you make such decisions. Since  you only get the info that's presented to you I thought I'd present you with the experience I had.

So this is what happened and what my thought process was, do with it what you please.

I used etcher to attempt to write an octopi image to a microsd card twice. Both times I was presented with a pretty undescriptive error that I don't remember the particulars of anymore. So I figured I'd try rufus. When I opened rufus and explored the options I saw the bad block check and thought, "hey, maybe I can get some more detailed information if this fails!" So you see, my expectations for the media beginning to fail were somewhat high. I watched the bad block check for a bit then went to bed. When I woke up I saw the green status bar and a positive message within. "Strange," I thought, "this didn't seem to find any errors, or maybe it has some sort of trick like the one I read about a while ago for HDD media where it can mark bad areas of the disk so they won't be used, or maybe it just writes the image anyway" So since I didn't have any results for the bad block check displayed and I had suspicions that I may have issues with blocks I did some googling and here we are. I found out that the results are in the log and I didn't actually have any issues with the media. Hooray.

Anyway, once again, thanks for putting your time into this software. Many people do appreciate it.

pbatard, 2019-02-11:
> My reasoning is simple. Humans are stupid, bugs happen. Now that's a large generalization and I don't put you in that category specifically, but since I don't know you, nor have I reviewed your [...]

Ergo, and as opposed to what you are stating initially above, you are actually putting me in that category specifically.

> and since I wasn't greeted with any results for the block check I had to do some investigating,

Yes, as I stated, you didn't see any error, and you __immediately assumed__ that the absence of errors being reported was an application bug, instead of the __logical__ thing to assume, which is that bad blocks check completed without errors.

At this stage, I don't know what else I can state that will make you lot realize that you simply cannot come to using an application by assuming it will be behaving the way others applications might behave. Most of the confusion I seem to be seeing here comes from people who want to see Rufus as single purpose bad blocks check application, with the expectation that it will stop whatever else it still needs to do to report on the bad blocks, even when no bad blocks are found. 

Well, sorry, but you __are__ misreading the UI and the fact that, the way it is presented, the "Check for bad blocks" is meant to convey an __ALSO__ ("also do this while you are performing the rest of the operations") rather than an __EXCLUSIVE__ ("Only do this and stop with a report about that task once it's done").

And misreading the UI is the same as misreading a book's sentence. You can argue as much as you want as to what you __intended__ the author to mean, but it is still what the author actually wrote that prevails. Therefore, when there is nothing in what the author says that is even remotely designed to indicate an alternate meaning (such as an __EXCLUSIVE__ instead of an __ALSO__), then you have to come to terms to the fact that, even if it can be understood how you may interpret things wrong (_"Well, applications that have been designed for the exclusive purpose of bad blocks checking will always present a report about the bad blocks check, whether successful or not"_), you still cannot skirt around the fact that you __are__ misinterpreting things.

> Well I mean, that's somewhat of a statistical fallacy isn't it? You don't have exact numbers on who reads / ignores dialog boxes so how can you say they do nothing?

Wow, if you want to talk about fallacies, let's talk about fallacies.

First of all, you are completely distorting the statement I made:

> (...) inserting yet another alert (...) is genuinely not as effective a means to help said users as people who never had to develop UI apps tend to think it is

Into

> dialog boxes (...) do nothing

So, first of all, hyperbole much?

All I stated was that dialog boxes are clearly not as effective as some proponent tend to think they are.

Also, I made sure that people could see that my assertion was derived from __strong empirical evidence__, of which I provided a verifiable sample set, whereas your retort seems to be a simple overgeneralization (_"Well, since __I__ am not doing it this way, then I am entitled to say that your assertion is false"_), which, unfortunately, is not how you contradict fact-backed evidence. Whether you like it or not, I'll continue to declare that, due to my position as the software developer, and even if I don't start to include a survey as part of the Rufus application, asking for feedback with regards to the effectiveness of dialog boxes, I can and have accumulated enough evidence to get a good idea of whether they are as effective as people think they are. That's even more true as it genuinely took me less than a couple minutes to find all the examples I linked to to validate my point, which I pre-emptively did __precisely__ because I was expecting people like yourself to immediately start dispute that assertion with an _"How can you know?!?"...

> but I was also presented with a pretty undescriptive error that I don't remember the particulars of anymore.

Not a great start, especially for someone who stated they always pay attention to the content of alerts. But I'll admit that I am kind of pulling your leg here, because it's too good an opportunity to pass... 😄 

> When I opened rufus and explored the options I saw the bad block check and thought, "hey, maybe I can get some more detailed information if this fails!"

Yes, you completely misread the UI by thinking it was an "exclusive" operation when it is only an "also", since you were preoccupied with other matters. That does make sense to me. But I still see no justification whatsoever to alter the UI when your issue was that your mind was elsewhere and you were making a lot of assumptions whilst trying to solve an issue by using multiple similar applications, that of course are using different approaches and cannot be amalgamated, as is the risk when using multiple applications in rapid sequence, as behaving with the same expectations.

> When I woke up I saw the green status bar and a positive message within. "Strange," I thought, "this didn't seem to find any errors"

In other words, between _"Maybe the application is right and it didn't find any bad blocks"_ and _"Maybe the application is buggy/does not properly report stuff/developer must be incompetent"_ you immediately went for the second one.

> I found out that the results are in the log

[... cut ...] l minority of people who interpret things wrong is not a good basis for software development.

Or maybe I should add a big alert when Rufus starts that says _"Hey, user, please pay a little bit of attention to the actual UI, rather than assume that the application is designed in the way you might erroneously imagine it to be. Especially don't assume that because some other application that was designed for a different purpose is doing it one way then Rufus is going to do it the same way, okay?"_