pbatard/rufus #565: Damm small Linux distro Rufus USB boots on both computers but Windows Vista x86 only in one.

mametoc, 2015-08-05:
(lastest Rufus 2.2. used)

It seems older BIOS (from a old Asus barebone Pundit-S Pentium 4 @2.6ghz) cant boot whit NTFS (Windows Vista) but yes GNULinux (FAT32).

Both USBs works in another PC (AMD Athlon XP 3000+).

Any help?.

Thx.

pbatard, 2015-08-05:
Sorry, but this is a hardware or BIOS issue not a Rufus issue. You will have to contact Asus or find Asus user forums.

closed 2015-08-05 by pbatard

mametoc, 2015-08-05:
Sorry to insist but Windows Vista x86 whit YUMI-2.0.1.8 (wich seems discontinued since 2014 i asked them but not response at the moment) at least appears a windows boot manager screen whit error BCD whit Rufus any windows think boot but: "disk error: press alt+ctrl+supr".

pbatard, 2015-08-05:
Again, you will have to contact Asus or people who have the same motherboard model as you have.
It's not because I created a software that can help computer boot from USB that I know anything about the specific of ASUS hardware or firmware, sorry.