pbatard/rufus #568: Windows 10 Home Windows to Go Install

dexwolfram, 2015-08-09:
Hi,

I am trying to install Windows 10 Home edition as windows to go (WTG).  When I downloaded windows 10 iso from Microsoft directly (without using windows download tool), I used the following link (since changed / no longer available from microsoft, so I provided the link at google cache):

http://webcache.googleusercontent.com/search?q=cache:yFWPDQgZOI8J:https://www.microsoft.com/en-us/software-download/windows10ISO+&cd=1&hl=en&ct=clnk&gl=us

The only option here was to download windows 10 "valid for both Windows 10 Home and Windows 10 Pro."  No separate options for downloading an iso exclusively for Home edition or Pro.  When I used Rufus to install WTG from this image, it installed Windows 10 Pro to my USB drive without prompting me. 

The other alternative I know of, is to download Windows 10 iso using the Windows 10 Download tool:

https://www.microsoft.com/en-us/software-download/windows10

Using this method, I do get an option to download a dedicated iso for Windows Home, but the iso from Microsoft is apparently corrupted for purposes of use with Rufus, producing the error  "Error:  ISO image extraction failure."  I think this is a known issue with Rufus being very particular about only using valid .iso files and Microsoft frequently producing .iso files that are a bit off.

I also tried creating a Windows 10 Home installation DVD directly using Windows Download Tools, then ripping the installation DVD to iso using cdrtfe, infrarecorder, and ImgBurn.  None of these were able to produce an iso from the DVD that Rufus was able to read.

Are there any workarounds at this point?  Is there a way to direct Rufus to install windows home rather than pro from the combined home/pro iso from Microsoft, which is the only thing I have been able to get to work so far?

Thanks.

pbatard, 2015-08-12:
The license key decides the version that will be installed. You don't need a specific "Home" or "Pro" ISO.

Also, you can get retail Windows 10 ISOs from the official [Microsoft Tech Bench](https://www.microsoft.com/en-us/software-download/techbench). If you are creating a WTG drive, these are the ISOs you should use as the ones created using the Microsoft Download tool contain an `install.wim` that isn't compatible with Microsoft's WIM extraction API (and Rufus).

I will close this issue now as all your questions should have been answered.

closed 2015-08-12 by pbatard

duozmo, 2015-10-16:
Empirical results from my recent experience:

- The downloads from the [general download page](https://www.microsoft.com/en-us/software-download/windows10ISO) and [Tech Bench](https://www.microsoft.com/en-us/software-download/techbench) were identical, to the byte.
- Using Rufus to create a Windows To Go disk forced the install to be Windows 10 Pro. No product key was required (activation happened on its own as soon as I connected to the Internet), and in fact my attempt to change to my retail Win 10 Home key was rejected.

dexwolfram, 2015-10-19:
That's similar to the behaviour I experienced.  On booting the first time after using Rufus, it would not accept my Win 10 Home activation key; gave error along the lines of "that activation key is not valid for this version of windows."  Chose to boot without entering key (demo mode) found I was in Wind 10 Pro.  Would not allow me to change back to Win 10 Home.

dexwolfram, 2015-10-19:
That's similar to the behaviour I experienced. On booting the first time after using Rufus, it would not accept my Win 10 Home activation key; gave error along the lines of "that activation key is not valid for this version of windows." Chose to boot without entering key (demo mode) found I was in Wind 10 Pro. Would not allow me to change back to Win 10 Home with my Win 10 Home Key.  Haven't yet had a chance to try this again...

dexwolfram, 2016-01-01:
Hi duozmo, pbatard,

I finally got my license key activated for Windows 10 Home.  It was not intuitive for me.  Here's what I did:

(1)  The following link says that "A Windows 10 Home product key cannot activate or change Windows 10 Pro to Home."  http://answers.microsoft.com/en-us/windows/forum/windows_10-windows_install/cant-activate-windows-10-0xc004f014/4d05e6fd-b972-4412-ba0f-05121ba222eb

(2)  Counter-intuitively, one solution is to download the "Windows 10 SL (Single Language)" version .iso from windows techbench  (https://www.microsoft.com/en-us/software-download/techbench) which the respondent at the above link says installs Windows 10 Home rather than Windows 10 Professional, and I can confirm that after bypassing the license key entry at startup, that is the case.  Why Microsoft would not clearly label "Single Langauge" as Windows 10 Home, I don't know.

(3)  Using the "Single Language" download, I was able to log into windows home version after bypassing the license key activation step in setup.

The other problem I had, is that the windows install does not have drivers pre-installed for my Wifi card.  I had to hook up a USB to Ethernet adapter to get internet connectivity before entering my Home license number into Setings -> System -> About -> Change Product Key.  Still having some troubles getting WiFi drivers up & working.

One thing I'm not really sure of, is whether I checked whether the "Windows 10" (standard) iso install could have accepted a Windows 10 Home key at startup if the ethernet had been connected.  I thought I tried that permutation and it hadn't worked, but I'm not 100% sure.  Maybe if you have one set up, you could try.

FYI, I had to use a USB connected drive of "fixed" rather than "removable" type in order to get rufus to install "Windows 10 Single Language" iso, though I'm a bit confused because previously I thought I had installed "Windows 10" iso to the same removable drive without too much trouble.

FYI, what led me to the link recommending the "Windows 10 Single Language" version was the error code 0xc004f014, which I was getting while trying to downgrade the "Windows 10" (Professional) install to Windows 10 Home using my license key.

Maybe if you or I can verify that "Windows 10" iso does not accept Windows 10 Home license key at setup, even with internet active and running (which I think it does not), we could let pbatard know as I would think this would be a common stumbling block for windows to go installations.

duozmo, 2016-01-05:
Good info, @dexwolfram. I feel my Win10 install is so precarious now that I'm fearful to touch it. If I ever am forced to reinstall, I'll try to remember to report back.

realtebo, 2016-09-22:
Please, could you link / tech a way to download an ISO compatible with windows-to-go.

All the links to techbench in this thread are broekn

pbatard, 2016-09-22:
https://www.microsoft.com/en-us/software-download/windows10ISO/

realtebo, 2016-09-22:
I redownloaded ISO but it's a normal ISO and it's not compatible with rufus.... What am I doing wrong?

pbatard, 2016-09-22:
> but it's a normal ISO and it's not compatible with rufus....

It is

> What am I doing wrong?

Post your log.

NSbuilder, 2016-09-22:
Why is the ISO created by Media Creation Tool is incompatible with Microsoft's WIM extraction API?
Had Microsoft done it on purpose ?

I'm using the media creation tool because it include x64 and x86 installation in 1 ISO.

@realtebo keep in mind that if you are using a removable usb flash drive - windows is most likely to stuck in the boot screen.

pbatard, 2016-09-22:
> Why is the ISO created by Media Creation Tool is incompatible with Microsoft's WIM extraction API?
Had Microsoft done it on purpose ?

Your guess is as good as mine. All I know is, if you try to use the native Windows API to open the MCT `install.wim` it won't work. Oh, and if you're using the MCT to create a dual image, it's even worse, cos then you get 2 separate `install.esd` files, which are even more painful to handle...

dexwolfram, 2016-09-23:
I used the "Single Language Edition" of Windows 10, but that was while
ago (available at link below using the pull down menu).  Using the
regular installation iso worked for me initially, but it somehow ended
up loading in Windows 10 Pro by default, for which I had no license
key.  Gave me some error in trying to backgrade the install back to
Windows 10 home.  I think someone from Microsoft once told me that
Windows 10 pro is not built to backgrade to Windows 10 Home, so once the
install goes that way, you can bring it back to Windows 10 Home.  Not
really sure about that.

Per my understanding, Single Language Edition by default installs
Windows 10 home, for which I did have a key.

Also: 

(1) Be sure you are running Rufus while booted into Windows 8 or later
(it's in the instructions).

(2) I had not much luck installing onto USB pen drives. I believe this
is because USB pen drives are not set up by default to have their flag
set to FIXED (as opposed to REMOVABLE).  There may be some pen drives
out there that are set to FIXED, or that have manufacturer utilities
available to change the flag (Lexar?), but I did not bother.  In the
end, I installed to a small USB connected external SSD with the drive
flag set to FIXED.  There may be other ways.  Not sure what happens if
you buy one of the Windows To Go certified drives -- these are supposed
to be for official installs of Windows To Go.  Anybody had any
experience using Rufus to install a "homemade" version to one of these? 
Doe sit really run faster on Certified Windows 2 Go pen drive, even if
it is not an official Windows 2 Go installation?

Thanks, let me know if you have other questions.  I'm no computer
expert, those are just the sticking points I had before I could get it
to work.

dexwolfram, 2016-09-23:
I used the "Single Language Edition" of Windows 10, but that was while ago (available at link below using the pull down menu) -- see above description of my situation.

Eventually, I did get drivers to work in Windows 10 on my macbook -- that involved getting windows drivers for macbook from Apple boot camp, then running those installs while booted in Win 10.

Also: 

(1) Be sure you are running Rufus while booted into Windows 8 or later (it's in the instructions).

(2) I had not much luck installing onto USB pen drives. I believe this is because USB pen drives are not set up by default to have their flag set to FIXED (as opposed to REMOVABLE).  There may be some pen drives out there that are set to FIXED, or that have manufacturer utilities available to change the flag (Lexar?), but I did not bother.  In the end, I installed to a small USB connected external SSD with the drive flag set to FIXED.  There may be other ways.  Not sure what happens if you buy one of the Windows To Go certified drives -- these are supposed to be for official installs of Windows To Go.  Anybody had any experience using Rufus to install a "homemade" version to one of these?  Doe sit really run faster on Certified Windows 2 Go pen drive, even if it is not an official Windows 2 Go installation?

NSbuilder, 2016-09-23:
(1). It is normal that non-FIXED drives get stuck during windows to go boot.

(2) I used a windows software called WinToUSB which can let you install a vhd/vhdx based windows to go ( I think windows is running different in vhd/vhdx because it worked for me unlike Rufus which create normal windows to go drives  ).
However this is unofficial method and I'm sure that it is the best choice to just use a Certified Windows 2 Go pen drive.

(3) Certified windows to go are USB 3 drives that are very fast and are supported by Microsoft to run the windows system.
BTW: using windows to go on macbooks is unsupported but maybe it can work.

(4) Even if you  can change the removable flag - it doesn't mean your drive will be faster - It just mean that windows will think your drive is FIXED.

you can learn more here: https://technet.microsoft.com/en-us/library/jj592680