pbatard/rufus #570: "0 Devices Found" with ADATA UV128/16GB in USB 3.0 Port

wstrohm, 2015-08-12:
I have the subject USB stick formatted using Apple Disk Utility, partitioned as an "MBR" "FAT 32" device. I copied the files in the "FreeDOS" image to the drive. Opening Rufus under "Crossover" results in Rufus opening, but "0 devices found." All text fields are empty.

Is my problem that Disk Utility is running under Mac OS Yosemite (10.10.4)? I thought this computer was both Windows- and Mac OS X- compatible.

Motherboard: Gigabyte GA-H87N-WIFI
CPU: Intel Core i7 4770K
Storage: two Samsung 840 EVO SSDs
RAM: Kingston 16 GB DDR3 kit
USB drivers: Apple "USB Mass Storage"

Any comments?

Thanks!

----    Bill

pbatard, 2015-08-12:
Please try the latest from https://rufus.akeo.ie/testing/ and if that doesn't work, try <kbd>Alt</kbd>-<kbd>.</kbd> in that version to enable USB Debug (you should see a message telling you that USB debug is enabled on the status bar) and post the __FULL__ log from Rufus.

wstrohm, 2015-08-13:
Tried latest ("Test 6") and same result; "0 devices found." 
Copy of log:

Rufus version: 2.3.696 (Test6)
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
0 devices found
0 devices found
 
Please note that the USB icon shows up on my desktop, and the "About this Mac/System Report/USB" area of the system report shows this:

USB 3.0 SuperSpeed Bus:

  Host Controller Location:	Built-in USB
  Host Controller Driver:	GenericUSBXHCI
  PCI Device ID:	0x8c31 
  PCI Revision ID:	0x0005 
  PCI Vendor ID:	0x8086 
  Bus Number:	0x0a 

ADATA USB Flash Drive:

  Product ID:	0xdb8a
  Vendor ID:	0x125f  (ADATA Technology Co., Ltd.)
  Version:	1.00
  Serial Number:	13C0713521560020
  Speed:	Up to 5 Gb/sec
  Manufacturer:	ADATA
  Location ID:	0x15200000 / 6
  Current Available (mA):	900
  Current Required (mA):	480
  Capacity:	15.82 GB (15,822,487,552 bytes)
  Removable Media:	Yes
  Detachable Drive:	Yes
  BSD Name:	disk2
  Partition Map Type:	MBR (Master Boot Record)
  S.M.A.R.T. status:	Not Supported
  Volumes:
MBR USB:
  Capacity:	15.82 GB (15,822,486,528 bytes)
  Available:	15.78 GB (15,780,749,312 bytes)
  Writable:	Yes
  File System:	MS-DOS FAT32
  BSD Name:	disk2s1
  Mount Point:	/Volumes/MBR USB
  Content:	DOS_FAT_32
  Volume UUID:	BF789339-8A3B-3C00-8558-B7692497AA69

pbatard, 2015-08-13:
Please provide a log with USB Debug turned on.

With Rufus open, you need to press the <kbd>Alt</kbd> and <kbd>.</kbd> (dot) key on your keyboard simultaneously until you see `USB debug enabled` at the bottom of Rufus.

Then you should plug your USB and send me the log from Rufus.

At the very least you should see something like:
```
Processing Hub 1:
Processing Hub 2:
Processing Hub 3:
(...)
```

pbatard, 2015-08-13:
Now, reading on what _Crossover_ really is (an emulation layer that runs within OS-X, rather than an actual Virtual Machine), I would say that there is a good chance that what you are trying to achieve will never work.

Rufus needs access to Windows low level APIs that are only found when an actual Windows Operating System is running (which _Crossover_ does __NOT__ provide). _Crossover_ seems to be mostly designed to try to make regular user mode applications run, that don't need to access the hardware or anything. Rufus is not such an application.

So my advice will be the same as for people trying to run Rufus using _Wine_ in Linux: if it works, great! If it doesn't, you're on your own, because Rufus was only ever designed to run in an actual Windows OS.

wstrohm, 2015-08-13:
Thanks for your responses. I suspected my issue was something like this; however, please note the second bullet on your main page:

Rufus is a utility that helps format and create bootable USB flash drives, such as USB keys/pendrives, memory sticks, etc.

It can be especially useful for cases where:

you need to create USB installation media from bootable ISOs (Windows, Linux, UEFI, etc.)
you need to work on a system that doesn't have an OS installed
you need to flash a BIOS or other firmware from DOS
you want to run a low-level utility

So it does NOT really work on a system with no OS installed, if you require a Windows OS to be running. Do you have a clue as to what procedure might work to create a minimal DOS (such as FreeDOS) that will boot on a USB stick? I have also tried UNetbootin, which cannot see the USB stick, same problem as FreeDOS.

pbatard, 2015-08-13:
Except for boot loaders, every application requires an OS to run, and Rufus is no exception. A DOS BIOS flashing utility is an application that runs under the DOS Operating System (which is actually a redundancy since 'OS' in DOS stands for 'Operating System', a Windows installer is an application that runs under the minimal Windows execution environment that the boot loader loads on Windows installation media, and UNetbootin is an application that (depending on your download) runs under the Linux OS or the Windows OS.

You are confusing an application such as Rufus that can install a bootloader (plus a minimal OS and the set of applications that go with it, depending on the options you select), and that needs to run on a specific Operating System, such as Windows, with something that can simply be copied over to a flash drive, no matter the OS you use, and will magically boot a blank system. That last part does exists, but is limited to UEFI boot applications and will only work if you boot that drive on an UEFI system... which is incompatible with DOS.

Please don't ignore the _"Rufus is a utility that helps format and create bootable USB flash drives"_ part to try to fit your narrative. Utility means application means requirement for the underlying OS that the application was designed to run on (in this case Windows). Oh, and if you don't think this is obvious enough, you should read a bit further down where it __clearly__ states:

> System Requirements:
> Windows XP or later

If there's no mention of OS X or Linux, then it means that it's a Windows only application. Therefore, if you can't meet that requirement, there's not much I can do to help you out.

closed 2015-08-13 by pbatard

wstrohm, 2015-08-14:
Mr. Batard,

I wasn't confused. I thought Rufus performed that "last part," based on the quote from your website that I posted above. "It can be especially useful for those cases where... you need to work on a system that doesn't have an OS installed." That according to your responses to my questions, is simply not true. You are telling me (and I do appreciate it) that there is no such thing as an app that can install a minimal OS (and by that I mean a minimal stand-alone OS of any kind) that can reside on a USB stick, unless another OS is already present somewhere else (like on a hard disk). I think I see a market.

My "Hackintosh" is a UEFI system, but I am blocked out by the incompatibility with DOS... bummer.

Thanks again for clearing that up and spending your time & effort educating me!

pbatard, 2015-08-14:
> That according to your responses to my questions, is simply not true.

It __is__ true. You are just misinterpreting the statement.

When you run it on Windows, Rufus has no trouble creating an USB with DOS, or with a Windows or Linux installer or live drive, __that will allow you to work on a system that doesn't have an OS installed__.

wstrohm, 2015-08-15:
"When you run it on Windows, Rufus has no trouble creating an USB with DOS, or with a Windows or Linux installer or live system, that will allow you to work on a system that doesn't have an OS installed."

That statement is very hard for me to understand, so now I am definitely confused. I guess I interpret that to mean something like the following:

" Once Rufus (running under Windows) creates a USB, the resulting USB can create Windows on a system that has no OS." So Rufus creates a Windows installer on a USB? Isn't Rufus doing what an original Microsoft Windows on purchased media does? And why wouldn't an accessible full copy of Windows also be required to install Windows using that USB? I ask this because:

I have a USB which will install Mac OS X. I made it per the instructions on tonymacx86.com using their software ("Unibeast"). But for it to work, I had to download the OS X full installer from Apple's "App Store" onto my hard disk. Without the full installer (5+ GB) present on my hard disk, the USB does nothing. So using the USB alone, Mac OS X cannot be installed from it.

So what does "work on a system that doesn't have an OS" mean, exactly?

I would really like to understand this. Thanks for putting up with me.

pbatard, 2015-08-15:
Assuming that you are running Rufus on Windows (since running it anywhere else is unsupported):
- If you select DOS or FreeDOS, Rufus will create a DOS-bootable USB drive, on which you can copy DOS applications, such as a DOS BIOS flashing utility or even a set of DOS/FreeDOS utlities. You will then be able to boot a system that doesn't have anything installed (it may not even have a hard drive) and work on that system regardless, using the DOS application you copied to your DOS USB
- If you select ISO mode in Rufus and then open a Windows ISO installation media, Rufus will create an USB which can be used to install Windows on a system that doesn't have any OS.
- If you select ISO mode in Rufus and then open a GNU/Linux Live ISO media, Rufus will create a bootable Linux USB that usually includes a whole set of GNU/Linux utilities allowing you to boot and work a system that doesn't have any OS installed, by using the utilities and applications offered by the Live media. Most of the time, you will pretty much have a full OS experience (including the ability to surf the web, or edit documents), even if you don't actually have any OS installed on any hard drive of your system (or even if you don't have any hard drive)
- If you select ISO mode in Rufus, and point to one of the __many__ utilities that come in ISO form (such as partition editors, disk rescue apps, antiviruses apps, mini Windows OS, such as BartPE/HirenCD, etc.), then you will be able to boot and run these utilities from your USB drive, even if you don't have an OS installed on the system you are using.

That's what _"work on a system that doesn't have an OS"_ means.

wstrohm, 2015-08-15:
OK, thank you very much. I will digest this for a while, and hopefully eventually grasp it entirely. Sorry to take so much of your time. Over & out.