pbatard/rufus #572: error 0x0000045D

zeitnoot, 2015-08-14:
Sorry for my bad English huh
During my work with ISO i has problem. USB-disk was formatted but while rufus started record ISO on it in some minutes log show me  Unable to create file: [0x0000045D]  (something wrong with input/output)
Can you help me? I can't decide do I need stop this process (but program write this may result in malfunction of my device) or do smth more? 
Thank you!

pbatard, 2015-08-14:
???

I assume that your problem was resolved. If not, then you're not providing enough data for me to help.

closed 2015-08-14 by pbatard

zeitnoot, 2015-08-14:
Sorry...
I resolved this problem by own but when I wanted to delete this message, I didn't find this button :D 
But...If you can answer me why error 0x0000045D occured when I tried to write ISO on the USB I will be grateful...

pbatard, 2015-08-14:
`0x0000045D` is `ERROR_IO_DEVICE`.

This looks like a hardware error (bad cable, bad USB, bad flash?) and is unlikely to have anything to do with Rufus. You should try with a different USB key and make sure that you don't use any USB extension cables.

zeitnoot, 2015-08-14:
Thank you for answer and sorry to disturb you.