pbatard/rufus #575: Rufus does not see mounted shared folders under VirtualBox

jelabarre59, 2015-08-17:
I have a MS Windows 7 virtualboc VM, with 5 "shared folders" set to auto mount (about 1/2 writable, rest read-only).  The mounted folders are visible as mounted "network" drives in the file manager, but when browsing for the ISO image in Rufus, the file browser in Rufus will only see the VM's hard disk and the virtual DVD drive.  No network mounts are seen, therefore cannot select ISO file on host OS.

pbatard, 2015-08-17:
https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cannot_see_my_network_shares

closed 2015-08-17 by pbatard

jelabarre59, 2015-08-17:
To **READ** a file???   Not much use to me then.

jelabarre59, 2015-08-17:
I don't understand.  It's **reading** a file.  It should be able to work.  So will this ever get ported to Linux. since that's what the HOST operating system is on my systems, and apparently that is the only way it could be made to work.

pbatard, 2015-08-17:
> I don't understand.  It's **reading** a file.  It should be able to work.

Network access is determined by your credentials (username and password). Because it's a formatting tool and needs to run as Administrator, Rufus uses the Administrator credentials, not the credentials of the regular user.

This limitation comes from Windows, __NOT__ Rufus. You can work around it by mounting your shares under the Administrator account too.

>  So will this ever get ported to Linux. 

https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxOS_XSome_other_OS

jelabarre59, 2015-08-17:
Mounting under Admin, now there's an idea I hadn't thought of.  Will have to try that one. Thanks.