pbatard/rufus #576: Failure, 'fixed' with ImageWriter

measter1, 2015-08-18:
Attempting to write a 200 meg .iso to a 1G USB, Lexar FireFly.   http://distro.ibiblio.org/quirky/precise-5.7.1/precise-5.7.1-retro.iso

Sequence:  Rufus 2.2 fail.  Reformat non-quick, fail.  Format with HP format tool. Rufus fail.  Check for bad blocks x1 negative. Rufus fail.  UNetBootin write appears successful; boot fail. Rufus continues to fail with the same report.  Windows ImageWriter write appears successful, boot fail.  After failed ImageWriter write, Rufus write successful.

Log of the last unsuccessful Rufus:

Rufus version: 2.2.668
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0039E21D
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
Will use '/isolinux.cfg' for Syslinux
Detected Isolinux version: 3.73/2009-01-25 (from '/isolinux.bin')
ISO label: CDROM
  Size: 209840128 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (3.73)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: precise-5.7.1-retro.iso
Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0039E21D
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{dac6b86c-0943-11dc-8e75-0018f3f1b544}\
Installing Syslinux 3.73...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{dac6b86c-0943-11dc-8e75-0018f3f1b544} for write access
Could not write Syslinux boot record: [0x00000005] Access is denied.
\\?\Volume{dac6b86c-0943-11dc-8e75-0018f3f1b544}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x032AEC8E
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

measter1, 2015-08-18:
> Reformat non-quick, fail.

I meant the reformat succeeded, but the Rufus write failed in the same way each time it failed.

pbatard, 2015-08-18:
> `Could not write Syslinux boot record: [0x00000005] Access is denied.`

This usually means that another application has the drive open (it can be a poorly designed antivirus application, or some other software such as the ones mentioned [here](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)). Unlike other formatting tools, Rufus will not attempt to format a drive if it can't get exclusive access (because this is very bad practice, and may result in unwanted data loss).

Also:

> `Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]`

Can you try with the latest TEST version from [here](https://rufus.akeo.ie/testing/)?
I'd like to see if it does a better job at getting the VID:PID of your device.

measter1, 2015-08-18:
I'm currently unable to retry to solve the problem because the problem has 'vanished'.  That is, it seemed as if something about the drive was wrong/bad which I could not fix with Rufus, HP, or UNB.  During the failed attempts to fix it, I moved the drive more than twice by employing Win XP and Win7 remove device functions; so I can't believe the problem was because of some application having the drive open.

At the present time, the drive works fine; Rufus wrote the .iso to it and the .iso boots and runs fine.  It seemed to have been 'cured' by attempting to write the .iso with ImageWriter which failed to boot, followed by a standard/default Rufus write.

I don't know how to reproduce at this stage.

pbatard, 2015-08-18:
> so I can't believe the problem was because of some application having the drive open.

You'd be surprised how many applications try to keep temporary exclusive access on USB devices. That's why I have the list at the end of the FAQ.

> I don't know how to reproduce at this stage.

That's fine. But can I still ask you to please try the latest TEST version with your device connected and post the log? I'd __really__ like to see if the recent changes I applied fixed the `(????:????)` issue (as I am hoping they have).

measter1, 2015-08-18:
> But can I still ask you to please try the latest TEST version with your device connected and post the log? I'd really like to see if the recent changes I applied fixed the (????:????) issue (as I am hoping they have).

If I'm to understand, you want me to 'rewrite' the same .iso to the same USB with the test Rufus and post the log, even tho' the .iso presently writes to the USB just fine with the current Rufus.

Currently writing.  Back with log soon.

measter1, 2015-08-18:
Ha.  It failed again.

Rufus version: 2.3.696 (Test6)
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00037A9B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00037A9B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
Disc image is an ISO9660 image
Will use '/isolinux.cfg' for Syslinux
Detected Isolinux version: 3.73/2009-01-25 (from '/isolinux.bin')
ISO label: CDROM
  Size: 209840128 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (3.73)
    With an old menu.c32: No
    With an old vesamenu.c32: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: precise-5.7.1-retro.iso
Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00037A9B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume GUID \\?\Volume{dac6b86c-0943-11dc-8e75-0018f3f1b544}\
Installing Syslinux 3.73...
Successfully wrote 'ldlinux.sys'
Opened drive \\?\Volume{dac6b86c-0943-11dc-8e75-0018f3f1b544} for write access
Could not write Syslinux boot record: [0x00000005] Access is denied.
\\?\Volume{dac6b86c-0943-11dc-8e75-0018f3f1b544}\ was already mounted as G:\
Re-mounted volume as 'G:' after error

Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03D4EE0A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

measter1, 2015-08-18:
Oooh.  I didn't realize that my Lexar FireFly was the same as the Lexar® JumpDrive® FireFly USB Flash Drive.  Your faq lists Lexar JumpDrive as incompatible.  There are actually a 'bunch' of different JumpDrives and Lexar has advice about messing with the partition condition.

I'm going to do some Lexar reading and see if I can 'neutralize' its special condition.

measter1, 2015-08-18:
JumpDrive Secure FAQs http://www.lexar.com/support/jumpdrive-secure-faqs

bunch http://www.lexar.com/support/product-support?field_typeref_nid=207

pbatard, 2015-08-18:
> If I'm to understand, you want me to 'rewrite' the same .iso to the same USB with the test Rufus and post the log

Ah sorry, I just needed the log with the USB plugged in - there was no need to rewrite.

It looks to me like the JumpDrive FireFly is a standard USB flash drives, but I've had a few reports about Lexar JumpDrives, and I know some of these drives have extra "security" features that make it troublesome for software like Rufus to work with them.

Especially, the fact that it's not possible to access the VID:PID of the drive (that `????:????` should have numbers in it), seems to indicate that there is something very special about these drives.

If you feel like it, can I ask you for one more tests (you __don't__ need to reformat your drive). With your Lexar drive plugged in an using the TEST version, can you please press the <kbd>Alt</kbd> and <kbd>.</kbd> while Rufus is running (that the Alt and 'dot' keys). You should see a message at the bottom of Rufus that says _USB Debug enabled_. Then if you can simply post the log from Rufus (again, no need to start a format operation) I'd appreciate that, as it may contain more details about why that drive seems to be so problematic.

measter1, 2015-08-18:
http://forums.anandtech.com/showthread.php?t=1912667   How do I turn a 512mb Lexar Jumprive Secure into a regular jumpdrive?

measter1, 2015-08-18:
alt-dot

Rufus version: 2.3.696 (Test6)
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03D4EE0A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Processing Hub 1:
Processing Hub 2:
  Found ID[075]: USB\VID_05DC&PID_A575\106A6603052448281106
Processing IDs belonging to USBSTOR:
  Root\LEGACY_USBSTOR\0000
  USB\Vid_05dc&Pid_a575\106A6603052448281106
Processing Device: 'USBSTOR\DiskLEXAR___JD_FIREFLY______1100'
  Matched with ID[147]: USB\Vid_05dc&Pid_a575\106A6603052448281106
  Matched with (GP) ID[114]: USB\ROOT_HUB20\4&1E70E850&0
  Props VID:PID = 0000:0000
Found USB device 'LEXAR JD FIREFLY USB Device' (????:????) [ID]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x03D4EE0A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-08-18:
Thanks a lot for this. While I don't think there's much I can do with the formatting issue as it seems to have to do with the extra security on these devices, at least now I have a good idea how I should be able to fix that `????:????` display and populate the actual numbers, which should be `05DC:A575` in this case.

measter1, 2015-08-18:
It seems to be unpredictable whether the syslinux gets written or not.  If Rufus makes it past that step due to some combination of factors like WinImage or UNB or HP formatter, then the write succeeds.

Rufus is distinguished from the others by knowing if it fails.  The others may seem to write successfully but then the USB doesn't boot.

At the present time, I've gotten/ I'm getting/ that .iso to write (slowly) with Rufus again, but it took a lot of messing around; it wasn't as simple as just writing with WinImage first.

There used to be a 'tool' available for download at Lexar to 'configure' the mystery partition (which is not visible with GPartEd), but it is no longer accessible there.

measter1, 2015-08-23:
This is what the log looks like when the .iso is successfully written to the Lexar Firefly with Rufus

Found USB 2.0 device 'LEXAR JD FIREFLY USB Device' (05DC:A575)
Using autorun.inf label for drive F: 'CDROM'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0007FE6A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'LEXAR JD FIREFLY USB Device' (05DC:A575)
Using autorun.inf label for drive F: 'CDROM'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0007FE6A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-08-23:
Thanks. Unfortunately, I suspect that the access denied issue is not something I can fix in Rufus, as I suspect they are due to your hardware/software. I think Rufus is actually behaving as it should, by reporting that "something" is preventing access, and it is up to you to figure what that "something" might be.

I'll add that I spent some time trying to figure out if I could make Rufus report the apps that are preventing access (#312), but it looks very difficult... While Windows has some APIs in place to do something like that for DLLs, it doesn't have anything for whole drives. This means that, short of adding a background process that monitors all disk access (and hope that whatever is locking disk access started __after__ that background process), it's next to impossible for Rufus to tell you what exactly is locking your drive...

As such, I think I will close this issue.

Now, since I was keeping it open as a reminder to address the `????:????` issue, and I have just published a BETA version that should fix that problem, if you still feel like running some more tests (again, no need to reformat your drive), I'd appreciate if you could just download the [latest BETA](https://rufus.akeo.ie/downloads/) and confirm that the `????:????` have properly been replaced with `05DC:A575` when you look at the log with your USB plugged in.

closed 2015-08-23 by pbatard

measter1, 2015-08-23:
beta: there are no ????

Rufus version: 2.3.703 (Beta)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'LEXAR JD FIREFLY USB Device' (05DC:A575) [GP]
Using autorun.inf label for drive F: 'CDROM'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0007FE6A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Processing Hub 1:
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
Processing Hub 6:
Processing Hub 7:
  Hub[1] = '\\?\usb#root_hub20#4&3a913089&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[075]: USB\VID_05DC&PID_A575\106A6603052448281106
Processing IDs belonging to USBSTOR:
  USB\VID_05DC&PID_A575\106A6603052448281106
Processing Device: 'SCSI\DiskST3750640NS_____________3CNR'
  Non USB => Eliminated
Processing Device: 'USBSTOR\DiskLEXAR___JD_FIREFLY______1100'
  Matched with ID[075]: USB\VID_05DC&PID_A575\106A6603052448281106
  Matched with (GP) ID[047]: USB\ROOT_HUB20\4&3A913089&0
Found USB device 'LEXAR JD FIREFLY USB Device' (05DC:A575) [GP]
Using autorun.inf label for drive F: 'CDROM'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 123, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0007FE6A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 966.5 MB (1013448704 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

pbatard, 2015-08-23:
Great! Thanks a bunch for reporting on this! :+1: 

While this issue is closed, I'm leaving #312 open, as I'm hoping that I can still come up with a bright idea on how to detect which apps are locking access... or how to kick them out to prevent these kind of errors.

measter1, 2015-08-23:
I believe that some Lexar USB has JumpDrive Secure which involves a partition^1.  I also believe that some Lexar users have Lexar software written to their Windows^2  C:\WINDOWS\SYSTEM32\LxrJD31s.exe.  I don't believe this is the case with the Firefly, but I don't recall its original condition when new.  Currently, advanced google search on the lexar.com site doesn't find any such lxrjd31s.exe^3 or jdsecure31.exe (mentioned elsewhere).  However, Lexar does provide free software for download at the site^4 called "EncryptStick Lite software now available for all Lexar® JumpDrive USB flash drives" 

^1 http://www.lexar.com/support/jumpdrive-secure-faqs  JumpDrive Secure FAQs

^2 http://www.bleepingcomputer.com/startups/LxrJD31s.exe-21825.html Home > Startup Programs Database > LxrJD31s.exe Information 

^3 http://www.neuber.com/taskmanager/process/lxrjd31s.exe.html lxrjd31s.exe is part of the software for Lexar's JumpDrive USB thumb drives.

^4 http://www.lexar.com/flash-drives/lexar-jumpdrive-s50-usb-flash-drive?category=207  EncryptStick Lite

commit 7ff5b3ca (2015-08-23): [core] fix the retrieval of VID:PID for grandfathered devices

* Closes #576