pbatard/rufus #579: FAT instead of FAT32 fixes "failed to load ldlinux.c32" error

maphew, 2015-08-22:
I made a 1gb boot usb stick using Rufus 2.2 build 668 using a Lubuntu amd 64 iso image (670mb) on Win7x64. I answered yes to the prompt to download new Syslinux files and accepted all the defaults for cluster size etc.

When attempting to boot I would get the error `failed to load ldlinux.c32`, `please change disks error`. I rebuilt the stick several times, including testing for bad sectors (only 1 pass though), with no change in behaviour.

Following a suggestion in [ [SOLVED] Boot failed: Failed to load ldlinux.c32][1] I finally got a bootable device by changing the file system to FAT instead of FAT32

[1]: https://lime-technology.com/forum/index.php?topic=38160.msg353473#msg353473

pbatard, 2015-08-22:
This looks like a pure Syslinux issue to me. There's no reason a Syslinux module should be able to access FAT16 content and not FAT32, so there really is nothing for me to fix in Rufus.

Thanks for reporting the workaround though.

closed 2015-08-22 by pbatard

jasonblum, 2016-07-31:
Just wanted to share FAT instead of FAT32 also worked for me too.

JaneSmith, 2016-10-23:
I'm encountering this issue when trying to setup an Ubuntu 16.10 boot USB stick. Unfortunately, using FAT does **not** fix the issue. :( Any ideas?

AhmedSayed77, 2017-04-03:
I'm also encountering the same issue with ubuntu 16.04. help

lbdBG, 2017-04-11:
@AhmedSayed77 @JaneSmith try to download an older version of ubuntu and try it, im not sure but it might fix it.

AhmedSayed77, 2017-04-12:
@lbdBG actually that worked. downloaded 14.04, installed it and then updated to 16.04. more steps but works like a charm @JaneSmith

marslan390, 2017-04-29:
Please download latest version of software to make USB bootable from https://rufus.akeo.ie/ and launch. If your ISO image will latest then it will ask you to download some necessary files to make this ISO bootable. Download that and then make USB bootable. I hope this will solve your problem

rs99cool, 2017-05-01:
My experience has been with Linux Mint. Tried version 18.1 Serena with Fat 32 and then just with Fat and neither one worked. Thanks to this thread, I tried the earlier version 17.1 Rebecca and was delighted to find that it worked just fine! You guys were a great help. Thanks very much!

gsim, 2017-08-12:
The latest version of Rufus (2.16) fixed this for me.

superbag2010, 2017-10-03:
I tried to install ubuntu 16.04.2 with rufus 1.3.4 and had the same problem. However as marslan390 said, it worked with a latest version of rufus. Thank you!

iqbaldp, 2018-01-01:
i dont know

markuskramerIgitt, 2018-01-07:
Got error "failed to load ldlinux.c32" with rufus-2.18.1213 and ubuntu-16.04.3.
There is no format FAT instead of FAT32.
Chosing exFAT instead of FAT32 removes option "Create a bootable disk"

pbatard, 2018-01-07:
@markuskramerIgitt, thanks for the update!

Just for the record, you can also verify the SHA-256 checksum of your ISO in Rufus by pressing the `#` symbol at the bottom of Rufus.

bozeyman9000, 2018-11-08:
How were you able to use FAT rather than FAT32? Im using rufus 3.3 updated 9/17/2018

ibuprophen1, 2018-11-09:
@bozeyman9000, It's been a while since I needed to use a FAT/FAT16 partition myself but, maybe the option isn't there for you since it's not an option available for your current partition size. 

So what you need to do is create a separate partition for FAT/FAT16. The following information, I believe, is still correct for Maximum Volume Sizes of the Partition. 

FAT/FAT16 = Max volume size 4GB & Max file size 2GB.

FAT32 = Max volume size 2TB & Max file size 4GB.

exFAT = Max volume size 128PB & Max file size 16EB.

And so on... 

Basically, if your USB Drive has more than a 4GB Partition, you will need to resize/create a Partition no more than 4GB.

I hope that I had explained this okay via text... :-) 

Good Luck! 

~Ibuprophen