pbatard/rufus #582: access to nas hard disk

ironmanspcasa, 2015-08-28:
Hi: 
I have a nass dlink 320LW unit with a 3TB hdd with letter Z: in the system windows 8
Your soft is not able to get acces to this letter, the rest of units appears ok but this one no.
can you check it?
I hope the problem is with no local unit hdd

pbatard, 2015-08-28:
Rufus is designed to eliminate hard drives by default (to prevent the accidental deletion of important backup data for instance).
Please see [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected) to enable listing of USB HDDs. Note that internal (SATA, etc.) HDDs will still not be listed.

closed 2015-08-28 by pbatard

pbatard, 2015-08-28:
Also, if this is a __network__ share, please see [this](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cannot_see_my_network_shares).