pbatard/rufus #584: USB Recognized on 2.2 but not 2.3

Sunchaser6, 2015-08-31:
I have an Corsair Flash Voyager GTX 256GB USB 3.0 drive that was successfully recognized and worked on version 2.2.668

[image]

When I try and use this exact same drive on version 2.3.709 the drive is not detected at all
[image]

Operating system is windows 10 pro x64.

If i open up version 2.2 i can see it, then i close version 2.2 and without removing the USB, open version 2.3 it is not detected at all. 

Initially I thought this may be a broken USB so I completely formatted it and still had the same issue.

Sunchaser6, 2015-08-31:
This actually is a USB, but is not detected as such:

Rufus version: 2.3.709
Windows version: Windows 10 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found UAS (USB 2.0) device 'Corsair Voyager GTX SCSI Disk Device' (8087:0024)
Device eliminated because it was detected as an USB Hard Drive (score 3 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found

This can be found here: http://www.scorptec.com.au/product/USB_Sticks_&_Hubs/USB_Sticks/57889-CMFVYGTX3B-256GB

pbatard, 2015-08-31:
Rufus tries to differentiate between USB HDDs (which may be used as backup drives, and which you probably want to avoid formatting by mistake) and regular USB Flash Drives ('UFD'). To do that it uses in part the `8087:0024` number you see in the log above. But this detection process is mostly empirical, which means I need to adjust it according to new devices, etc.

Now, before 2.3, there was an issue with retrieving that `8087:0024` number (you would see something like `????:????` instead, which of course isn't correct), so I fixed that. But of course, now that these numbers are properly filled, the HDD vs UFD detection process might require some adjustments.

I will make sure your Corsair device is properly listed by default in the next version.

In the meantime, __as indicated in the log you pasted__, you can simply click on the white triangle, and then check the _'List USB Hard Drives'_ checkbox. Or you can also see [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected).

commit 102e1769 (2015-08-31): [core] fix detection of some Corsair UFDs

* Closes #584
closed 2015-08-31 by pbatard