pbatard/rufus #585: UEFI Windows 10 Education is not bootable.

rcarlos243, 2015-09-01:
I am trying to create a UEFI (GPT FAT32) Windows 10 Education USB installer but it won't get detected in boot. Other OS like Windows 8.1 or Windows 10 pro has no issue creating a UEFI USB installer under Rufus

pbatard, 2015-09-01:
Please provide the SHA-1 of your ISO (just select your ISO in Rufus and click the `#` button at the bottom). This will help me try to locate a Windows 10 Education ISO, as I don't have one, and therefore I cannot test.

andyhuanguccu, 2015-09-01:
I meet the issue too
use windows 10pro zh_tw download form "Media creation tool to download Windows 10"
SHA-1:b55c602a10de0ab0bdaa4cd560ac30f877eadc33

pbatard, 2015-09-01:
I just tested `en_windows_10_education_x64_dvd_6848120.iso` (SHA1: `02d3174b7853aa3465828b0ae1d3d6a0c22afd57`) and saw no issue booting from an Intel NUC in UEFI mode.

Were you trying `Windows To Go` mode or the regular installation. As always, a __FULL__ log would help...

pbatard, 2015-09-01:
@andyhuanguccu,

Can you please try with the __retail__ ISO that you can download from [Microsoft Tech Bench](https://www.microsoft.com/en-us/software-download/techbench)?
The Media Creation tool ISOs are created on the fly on your computer, and therefore, it's hard to guarantee that they didn't get corrupted for one reason or another...

andyhuanguccu, 2015-09-01:
@pbatard  
Success  
many thanks

rcarlos243, 2015-09-01:
@pbatard 

The one I have is
MD5: 1cdb78579776fc2a43eca49d0a9426cf
SHA1: d40e855e3f509e997c84d18ed1c93071bebd72ec

pbatard, 2015-09-01:
> SHA1: `d40e855e3f509e997c84d18ed1c93071bebd72ec`

Then this is not an official ISO from Microsoft. You can find the SHA1 for all the official Windows 10 Education ISOs [here](https://msdn.microsoft.com/en-us/subscriptions/downloads/hh442898.aspx#searchTerm=&ProductFamilyId=606&Languages=en&Architectures=x64&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0) (by selecting an ISO and then clicking Details). The only reference I see to any ISOs with your SHA1 is in [these comments](https://disqus.com/home/discussion/getintopcofficial/windows_10_education_iso_32_64_bit_free_download/oldest/) where someone indicates that their SHA1 for what they believe is an official `en_windows_10_education_n_x64_dvd_6847236.iso` doesn't match the one from Microsoft.

As you will understand, if an ISO is modified in any way from the Microsoft ones, I can not guarantee that Rufus will work with it (and I will not be spending time doing so), especially if I checked that Rufus worked with the official version and found no issues.

Please try to get your hand on an official ISO, and confirm that its SHA1 matches the one from Microsoft before using it with Rufus.

pbatard, 2015-09-02:
As I haven't found any problem with the official ISOs, I will close this issue.

closed 2015-09-02 by pbatard