pbatard/rufus #589: Creating Windows 10 UEFI fat32 USB Stick from NTFS Windows 10 ISO not possible

baumaeschi, 2015-09-03:
Today I wanted to create a bootable FAT32 USB Stick (UEFI) from a Windows 10 ISO. Rufus cannot create a FAT32 USB stick from a NTFS ISO. But - my UFEI boot Laptop needs a FAT32 USB stick (NTFS formatted USB sticks are not recogniced). Would be very nice if a future version of Rufus converts NTFS to FAT32 "on the fly". Your thoughts?

pbatard, 2015-09-03:
> NTFS formatted USB sticks are not recogniced

Are you sure? Rufus has an [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs) feature that enables computer to boot from UEFI, regardless of their firmwares, which is precisely the reason why NTFS is enforced. Have you really tried booting the USB created by Rufus? If you tried, what was the error you got?

cookieboyeli, 2015-09-04:
I'm having a very similar, if not the same issue. I was just about to create a ticket.

I have a laptop that will only detect UEFI bootable drives. The bios is gimped and there is no option for CSM either. I'm trying to install Windows 10 64 bit but I can't run it directly because the laptop has 8.1 32 bit. It has a 64 bit CPU though. Very irritating.

For some reason UEFI GPT FAT32 fails to boot, restarting after an instant. 
GPT UEFI NTFS detects and tries to boot but quickly says:

```
*** UEFI:NTFS ***

Loading NTFS Driver... DONE
Loading the first NTFS partition on this device... DONE
Looking for the NTFS EFI loader... DONE
Launching NTFS EFI loader 'efi\boot\bootia32.efi'...

    Load failure: [14] Not Found

Press any key to exit.
```

pbatard, 2015-09-04:
@cookieboyeli,
This is a completely different issue.

If your system tries to launch `efi\boot\bootia32.efi` it means that your UEFI system is 32 bit, __NOT__ 64 bit. The only way `efi\boot\bootia32.efi` gets invoked is on 32 bit UEFI systems.

This means you can forget about running Windows 10 64 bit on that system - it will never work. Get the 32 bit image instead.

To save a few bucks, it is actually common for cheap system to have a 32 bit UEFI firmware (and therefore force 32 bit Windows) even if the CPU is 64 bit capable. Please see [here](http://blogs.intel.com/evangelists/2015/07/22/why-cheap-systems-run-32-bit-uefi-on-x64-systems/).

baumaeschi, 2015-09-04:
Thank you replay. But, sorry, you are not right: my Laptop runs Windows 10 Home 64 bit perfectly!
You might be right about the 32 bit UEFI firmware...I dont't know.
I created a FAT32 USB Stick from the Windows 10 Home 64 bit ISO and everything worked fine.
The Stick was recogniced and bootable and then I installed the 64 bit OS without any problems.
I used this instructions for creating the Stick: https://www.thomas-krenn.com/de/wiki/Windows_UEFI_Boot-Stick_unter_Windows_erstellen.
So again my question: Would it be a big deal to create a bootable a FAT32 USB Stick from a 64 bit ISO with Rufus?

baumaeschi, 2015-09-04:
Uhps,sorry, I read only you Anwer to cookieboyeli.
Yes, I tried booting the USB created with Rufus 2.3
It is a 2 year old HP Laptop I'm working with. At boot time I press F9, and then, with the  Rufus created USB Stick I get the following options:
Boot Option Menu
  OS Boot Manager
  Boot From EFI File
When I create the Stick with the instructions of the link of my previous answer, I get the following options after F9:
Boot Option Menu
  OS Boot Manager
  USB Hard Drive (UEFI) - KingstonDataTraveler 3.0
  Boot From EFI File
Question: Could it be that my HP Laptop BIOS needs a USB 2.0 Stick to boot UEFI:NTFS?

pbatard, 2015-09-04:
> my Laptop runs Windows 10 Home 64 bit perfectly!

Yes, that's why I said @cookieboyeli's issue was not the same. Unlike @cookieboyeli, I don't believe you have a 32 bit UEFI firmware.

> Yes, I tried booting the USB created with Rufus 2.3

Thanks.

Are you sure you have Secure Boot disabled then?

Rufus' UEFI:NTFS will only work if you have Secure Boot disabled, because the UEFI:NTFS bootloader is not signed by Microsoft, and your symptoms could easily be explained with the HP "BIOS" having Secure Boot enabled, and only allowing boot from USB if the UEFI boot loader it sees is signed. If you just copy the files onto a FAT32 partition as per the guide you pointed out, this will be the case, but it will not be the case when using UEFI:NTFS from Rufus. So you have to temporarily disable Secure Boot to use UEFI:NTFS (but you can re-enable it once Windows is installed).

Otherwise, the other possibility I see is that your UEFI firmware is not following the UEFI specifications and expects the EFI partition to be the first one, whereas the specs indicate that it can be anywhere, and Rufus has to place it at the end so that the content of the drive remains accessible from Windows.

> Question: Could it be that my HP Laptop BIOS needs a USB 2.0 Stick to boot UEFI:NTFS?

I don't believe so. Most likely you need to locate the Secure Boot option, and disable it. On many UEFI firmwares, you will have to select `Other OS` somewhere under `Boot settings` in the Secure Boot category.

pbatard, 2015-09-04:
Also to answer your question:

> So again my question: Would it be a big deal to create a bootable a FAT32 USB Stick from a 64 bit ISO with Rufus?

Yes, it actually would, because then you would lose the ability to boot Windows ISO that contain a file larger than 4 GB (some official Microsoft ISOs are like that, and some custom ISOs that can be created using Microsoft tools may also have large files). Also, enforcing NTFS helps ensure that by default, Windows will only be installed either in BIOS more or UEFI mode, which avoids the risk of someone installing in BIOS mode on an UEFI system by mistake (see [here](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_both_BIOS_and_UEFI_mode).
Finally, the use of NTFS in Rufus greatly simplifies the creation of _Windows To Go_ drives.

If I were to allow the creation of FAT32 Windows drives, I would have to spend a lot of time and add many checks to ensure that Rufus users aren't shooting themselves in the foot, which I'd rather avoid.

cookieboyeli, 2015-09-04:
Thanks for that info pbatard, that was a very helpful article. I would never have figured that out on my own.

So basically, there is no possible way to boot any 64 bit OS on a machine with an exclusively 32 bit UEFI, due to UEFI the specifications. Well at least I can say I tried! :)

baumaeschi, 2015-09-05:
Thanks a lot also for all that interesting information. 
I tried booting with secure boot disabled. It did not help. The stick was not recogniced.
So, I don't know what I could try next. If the UEFI firmware is not following the UEFI specifications I don't have a chance, do I?

pbatard, 2015-09-05:
Yes, at this stage I don't know what I could do in Rufus or UEFI:NTFS to help you out.

Rufus does follow the UEFI specs with regards to how it creates an UEFI bootable drive for UEFI:NTFS. While there's always a possibility that there might be a problem in the bootloader itself, at the very least, if your system conforms to the UEFI specs, it should detect the bootloader and allow you to boot from the USB, and you should see some output from that bootloader.
If it isn't listed even when Secure Boot disabled, it looks like a clear indication to me that the HP "BIOS" isn't looking at all the UEFI bootable partitions as it should. Either that or it has an issue reading FAT16 content, but that would be very surprising as the FAT16/FAT32 UEFI driver is fairly standard and is pretty much the same for every firmware out there as it is provided by the UEFI developers in the base firmware.

Unfortunately, it is not possible to move the UEFI:NTFS partition at the beginning of the USB drive, as, due to Windows limitations, this would make the rest of the USB content inaccessible for people using USB flash drives with the removable flag set (at least 95% of all flash drives out there).

I'm afraid that the only thing I see is to try to contact HP and see if they have a BIOS update...

Alegom, 2015-09-05:
I was literally just searching now about whether I should get WinSetupFromUSB or Rufus, based on their differences. And I found this thread checking one of the most significant ones at first sight, the support for sources larger than 4 GB and FAT32 for EFI compatibility.

@baumaeschi, I think this can be a solution to your problem.

@pbatard, I'm sorry that this factor has led me to choose the other software over yours. I briefly read your concerns above about adding this function, but I believe it can be more important for some people than your nite interface and outstanding performance, in terms of time, regarding other softwares.

pbatard, 2015-09-05:
@Alegom, have you tried and encountered an actual issue with the UEFI:NTFS mode of Rufus?

Again, unless the UEFI firmware from your manufacturer is __broken__, Rufus should work just fine even with an NTFS formatted drive. Unlike what many people believe, there's nothing in the UEFI specs that mandates FAT32 for boot - all you need, really, is an UEFI driver capable of reading the file system you want to boot from, be it NTFS, FAT32, or something else...

Alegom, 2015-09-05:
@pbatard I can't say I have, no. As I mentioned, I was doing a recon on the available options to find the best suited one. Maybe in this case additional advertisement / further explanation is required on the features list. As you yourself said, not many people know about that detail, myself included even after having speed read the thread. It wasn't until I read your latest reply I thought it could actually work. I'll give Rufus a shot when I need it somewhere in the future. I would recommend you to clarify this more publicly, it could make a difference.

pbatard, 2015-09-05:
The thing is, there's nothing to clarify. Unless a manufacturer's UEFI firmware is broken, an USB drive created with Rufus will seamlessly boot from NTFS, and the vast majority of people won't even realize that they aren't booting from FAT32 (because, as I pointed out, it really doesn't make a difference as far as UEFI is concerned).

The only potential issue has to do with Secure Boot (a.k.a. [Restricted Boot](https://www.fsf.org/campaigns/secure-boot-vs-restricted-boot)), but considering that main reason a Rufus created UEFI:NTFS drive cannot be used with Secure Boot is because Microsoft, which controls the Secure Boot signing process, __anti-competitively__ declared that they would not sign anything that's GPLv3 (See [Paragraph 4](http://blogs.msdn.com/b/windows_hardware_certification/archive/2013/12/03/microsoft-uefi-ca-signing-policy-updates.aspx), where they __deliberately__ chose to misinterpret provisions of the GPLv3 to pretend that their purely arbitrary ban is justified), it's more of a political issue than anything else. UEFI:NTFS is GPLv3 (because the NTFS driver is GPLv3), and [I would get it signed in a heartbeat](https://github.com/pbatard/rufus/issues/507#issuecomment-102365845), if Microsoft allowed it.

Then again, considering that, unless you are using a computer where the user is no longer in control (of what they can and cannot boot), you should be able to temporarily disable Secure Boot when installing Windows. And since the Rufus executable (which contains the UEFI:NTFS bootloader) is itself signed, this doesn't make the boot process any less trustworthy.

pbatard, 2015-09-19:
As the issue seems to stem from broken UEFI firmware, I'm going to close this issue now.

closed 2015-09-19 by pbatard

aardelean, 2015-12-05:
Hi,
I am having a very weird ugly problem. Also got an HP laptop spectre x360 and replaced drive with a SM951 AHCI. The rufus usb stick cannot load unless legacy mode is enabled, and then the windows 10 installer detects the drive but cannot do anything on it, cannot format, cannot install, just hangs on any operation on the drive for 10-20 minutes and then gives an error number. 

Do you know what can I do to by pass this ugly issue?

pmkdatabase, 2015-12-27:
I have a fairly modern MB - Gigabyte Sniper A88X. Bios is the latest revision - I think less than a year old. The system is 64 bit. I've been running W7 64 bit for a long time, and successfully installed Windows 10 64 bit from DVD (but hate it and have gone back to 7). 

Anyway, when I tried to install the same Windows 10 ISO (and others) with Rufus 2.6 (also 2,2), formatted NTFS, all settings default, it failed with error: efi\boot\bootia64.efi not found. Not surprising, because indeed it was not on the USB drive - only the 32 bit version. Should it be there?

I'd like to understand what is going on. Does Rufus think my system is 32 bit? Does it care what system is creating the image? If so why, it might not be the same as the target.  So what is going on,  and what can I do  about it in case I want to use it on this box in the future?

pbatard, 2015-12-27:
> The system is 64 bit

I'll just start with an aside that having a 64 bit CPU does not necessarily mean that you can install a 64 bit OS, as some manufacturers may put a 32 bit UEFI firmware in there ([in order to claim more free space](http://blogs.intel.com/evangelists/2015/07/22/why-cheap-systems-run-32-bit-uefi-on-x64-systems/)). From what you explain above, I would tend to  trust that you did install a 64 bit OS, and therefore that your UEFI firmware is 64 bit, but I just want to point out that a recent system does not always guarantee that an OS can be installed in 64 bit.

> it failed with error: `efi\boot\bootia64.efi not found`

Okay, if this message is accurate, then it is exceedingly surprising and I'll tell you why: (Outside of Arm/Mips) There are 3 possible types of system that can be supported by UEFI:
1. 32 bit x86, which will boot in EFI mode using `/efi/boot/bootia32.efi` ("ia" stands for "Intel Architecture")
2. 64 bit x86, which is actually a 64 bit AMD (rather than Intel) extension, using `/efi/boot/bootx64.efi`
3. 64 bit Intel Itanium (colloquially called "Itanic", because it was a major disaster for Intel), which is a __non x86 compatible__ CPU type, and which __almost nobody uses, especially on a recent system__, using `/efi/boot/bootia64.efi`. 

I would be extremely surprised if you are actually using an Itanium CPU, all the more so as the error you report appears to come from UEFI:NTFS, which is NOT Itanic compatible (due to nobody in consumer-land using Itanium, really).

So I have to assume that the error you actually got was `/efi/boot/bootx64.efi not found` and that you also confirmed that this file was missing on the USB created by Rufus... which logically means that you must have used a 32 bit version of Windows (containing only `/efi/boot/bootia32.efi`), hence your issue, since, unlike what is the case in Windows, 32 bit UEFI code can not be launched from a 64 bit UEFI firmware, and vice versa.

What I would strongly suggest you to do then is:
1. Please confirm the error you got, or the type of UEFI firmware you got. If you see any mention of `/efi/boot/bootx64.efi` then you are running a 64 bit UEFI firmware and you __MUST__ use a 64 bit image of Windows.
2. Once you have confirmed whether you need to use a 32 bit or 64 bit version of Windows, head to [Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench) to download the relevant __retail__ image.

pmkdatabase, 2015-12-28:
Pete, Thanks for the extensive reply.You are probably right about the details of the error message as I was going by memory. 110% sure I am using Windows  7 64 bit as I type this.. 

I simply extracted all the files from the ISO image to a USB stick and that booted fine. i don't know what the implications of that are as to why this is happening with Rufus.

unlockboot, 2016-02-28:
Make sure you use the latest Rufus portable version http://www.downloadrufus.com/

unlockboot, 2016-02-28:
Use the latest windows 10 ISO file downloaded directly from Microsoft & try again. If can't boot, then install via Legacy.

cookieboyeli, 2016-02-28:
Congrats Pete, looks like you've got your first adsense revenue leaching spammer.

pbatard, 2016-02-28:
Yes, that guy (stephanxxx of unlockboot) registered the site 3 days ago, and has been trying to boost its visibility, for the sole purpose of hijacking ad revenue, by also creating a subreddit and posting these devious links all over the place (oh, and he also shadowbanned me from his subreddit when I reported that the site was unofficial, without realizing that by doing so, he is of course digging a larger hole for himself __when__ I report his subreddit.).

Of course, what @unlockboot does not realize is that there are many ways I can, and will, take action against his obnoxious scheme, and that, while there is still a chance to resolve this peacefully (I don't really have an issue against sites who provide download links to Rufus, as long as they are not deliberately trying to pass themselves for official, as his website does), the longer he keeps at it, the faster time is going to run out...

unlockboot, 2016-02-28:
1. I never stated that this site is official.
2. I put the official links - as all other download sites do.
3. I will post more tutorials on how to use the tool.

What is the problem here?

fedyvenom, 2016-06-25:
@pbatard Just FYI, I got same issue today. I want to reinstall my windows 8.1 x64 today but this is the first time i use Rufus to create bootable USB so i create it targeting GPT but also got same FAIL Message 'could not locate /efi/boot/bootx64.efi'. I have same brand with @baumaeschi , so is it possible that there's something wrong with HP UEFI system?

pbatard, 2016-06-25:
Do you actually see a `/efi/boot/bootx64.efi` file on the USB?

fedyvenom, 2016-06-25:
@pbatard Yes, i did.

pbatard, 2016-06-25:
Can I please have the __FULL LOG__ of how you created the USB with Rufus, as well as the __FULL OUTPUT__ from UEFI:NTFS (including the messages before the error)?

fedyvenom, 2016-06-25:
@pbatard let me try with Manually Extract it with FAT32 first, because usually i just did that for Windows 8.1 and it extracting now. But after that i will provide FULL LOG and OUTPUT for you.

fedyvenom, 2016-06-26:
@pbatard 
Log: [rufus.log.zip](https://github.com/pbatard/rufus/files/333372/rufus.log.zip)
Output: 
[image]
EFI Files:
[image]
BIOS Boot Conf:
[image]

pbatard, 2016-06-26:
This is not a Rufus issue. The problem is with your ISO, which is 32-bit

1. If you see a `bootarm.efi`, then you're NOT looking at the EFI boot files on the NTFS partition, but the ones on the additional FAT32 partition that is created by Rufus. In other words, please be mindful that there will be 2 partitions on your USB, and that when I asked you if there existed a `/efi/boot/bootx64.efi`, I meant on the NTFS partition. I probably should have been clearer, but unless your USB media is of type fixed, when you mount it in Windows, the only partition you will see is the NTFS one, so I thought that you'd just plug the drive in Windows to check. The `bootx64.efi` you see (as well as `bootarm.efi`) come from Rufus, not the ISO, and they are always there (else you would not see any error from UEFI:NTFS).
2. Your `rufus.log` __clearly__ indicates that no /efi/boot/bootx64.efi` is created on the (NTFS partition of the) USB, only `bootia32.efi`, which is useless with an x64 UEFI firmware like the one you have:
```
Extracting: E:\bootmgr (394.8 KB)
Extracting: E:\bootmgr.efi (1.3 MB)
Extracting: E:\efi\boot\bootia32.efi (1.3 MB)
Extracting: E:\efi\microsoft\boot\bcd (256 KB)
```

So UEFI:NTFS is exactly right - the ISO you used does NOT contain a `bootx64.efi` that can be used for boot.

Please make sure you use a 64-bit ISO, because your alleged 'windows 8.1 x64' is anything but x64.

fedyvenom, 2016-06-26:
Thanks for your reply, after i read this i found that the original source of problem is the ISO file. Though It have 64-bit version and because the size more than 4GB it cannot be partitioned as FAT32.

pbatard, 2016-06-26:
Which is where UEFI:NTFS could be useful, as it will let you use an ISO that contains a file that is larger than 4GB, such as your `/sources/install.esd` (4.6 GB).

But of course, that will only work if your ISO is really for 64-bit, since your target UEFI system is 64-bit. However, your ISO is for 32-bit only...

fedyvenom, 2016-06-26:
@pbatard Yeah, i already use other 8.1 ISO which has 64-bit it works so well. Rufus also automatically choose it as GPT.

pbatard, 2016-06-29:
I don't support AOI ISOs, and I strongly suspect that your "X86/X64" ISO is __NOT__ actually compatible with x64 EFI boot because it's probably missing a `/efi/boot/bootx64.efi` to extract. Rufus can not create this file from scratch &mdash; it must be on the ISO.

Please try with an official Microsoft ISO - they are the only ISOs I can support.

bnathz, 2016-06-30:
@pbatard Thanks for replying. I wanna know how a lot other people were able to install windows using the same ISO.
Also should i try burning the ISO to DVD and try?

NSbuilder, 2016-11-12:
Why mess with x86 and x64 ISOs?
I used Rufus to create dual x86 + x64 UEFI:NTFS Media Creation Tool windows 10 installation and it can be installed on uefi and bios without problem.

kakabest1111, 2016-12-17:
Deleted all the Disks -> Create new Disks -> Next  . Muahaha .

mrrezki, 2017-11-05:
I've all of this and just wanted to add a small contribution to put this to bed;
1- I had a problem with my Surface and needed to re-install windows
2- I used Rufus to burn a Windows 10 64bit iso image into a USB stick
3- Rufus insisted that I use NTFS format
4- Surface wouldn't start from the stick 
5- Used Diskpart on the stick
6- Extracted the iso into it manually
7- Windows 10 64 bit installed absolutely fine onto Surface

Without boring anyone with the detail as it's already been all mentioned above, there are certainly cases where Rufus wouldn't do the job and perhaps you should look into it ....
I can give any level of detail here if needed ...

Cheers

pbatard, 2017-11-05:
> I can give any level of detail here if needed ...

Well a __full__ log from Rufus after you selected your ISO and it told you that you must use NTFS would be nice.

Considering what you report, my guess is that you had selected "MBR partition scheme for BIOS or UEFI-CSM", under the impression that it would be compatible with UEFI boot, whereas what "UEFI-CSM" means is "BIOS mode", which of course is __NOT__ compatible with hardware like Surface. Instead, if you had selected "GPT partition scheme for UEFI", you would have found that Rufus allowed you to select FAT32 as the file system if you wanted, and that your USB would have been bootable on your Surface.

Or you could also have used the <kbd>Alt</kbd>-<kbd>E</kbd> cheat mode (though that assumes that you have read the FAQ, and especially [this entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode).

All in all, you need to be aware of one important thing: You need to know whether the target system you plan to boot is BIOS or UEFI, and use the right partition scheme option in Rufus if you want your USB to boot. If you don't do that, then, yes, chances are that you will end up creating a drive that doesn't boot.

Anime4000, 2017-11-13:
@pbatard 
`ALT+E` key very useful for Boot BIOS or Native UEFI, no CSM.
I run a repair shop, what I observe, most UEFI Firmware did not support to read NTFS partition for booting. obtain NTFS support will cost OEM's, unless Microsoft go easy with OEM.

Not sure if UEFI can read USB Drive with UDF partition, just like DVD

pbatard, 2017-11-13:
> what I observe, most UEFI Firmware did not support to read NTFS partition for booting. obtain NTFS support will cost OEM's, unless Microsoft go easy with OEM.

This is precisely why I created [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs), and the feedback I get is that, if you know what you are doing (i.e. make sure you are actually booting in pure UEFI mode, that you are using a Windows image that matches the UEFI firmware you use in terms of 32 vs 64 bit, and  temporarily disable Secure Boot if needed), this works just fine on 99.9% of platforms out there. My experience and the feedback I get does demonstrate that UEFI:NTFS does work when used properly, and that, unless you are using a limited subset of HP machines, you will have no issues booting an NTFS partition using UEFI:NTFS in pure UEFI mode.

Anime4000, 2017-11-22:
one more... when I load Windows XP, by default Rufus use NTFS,
for Acer TravelMate 5500Z, when USB Drive formatted as NTFS, BIOS simply remove HDD and USB from boot list,

<kbd>Alt</kbd>-<kbd>E</kbd> didn't work, can't burn Win XP using FAT32

pbatard, 2017-11-22:
As per the FAQ, Windows XP is [no longer supported](https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_NO_LONGER_SUPPORTED). As to your BIOS forcibly removing drives that are formatted as NTFS from boot (which I am very doubtful of, as they would have to pay developers to add __extra__ code to do just that, which no company would do), even if that happens to be true, I'm afraid have no intention of adding workarounds for a clearly broken BIOS. Instead I would encourage you to take this to Acer support, as, if the are effectively removing NTFS drives from boot (again, unlikely, which is why I'd like you to contact their support so that they can confirm that this is not the case, and that you probably did something that simply makes you believe that), they are the ones who should fix it.

jacquubee, 2017-11-23:
[image]
 there is ntfs_x64.efi in the folder but keep showing not found

pbatard, 2017-11-23:
It's relaunching UEFI:NTFS which can only mean that you __manually__ copied the `bootx64.efi` from the FAT partition to the NTFS partition. Don't do that.