pbatard/rufus #592: "Checking for Rufus updates..." window does not conform to Windows High Contrast theme standards

cookieboyeli, 2015-09-05:
Custom "high contrast" theme:
[image]

Official white high contrast theme:
[image]

This direct from clipboard image attachment system blows my mind.

pbatard, 2015-09-05:
Then that's no updates for you! :smiling_imp: 

More seriously, I'll try to have a look at this when I have a chance, but this will be fairly low priority...

cookieboyeli, 2015-09-05:
No problem, I'm just letting you know it's there. :)

pbatard, 2015-09-20:
I'll just comment, since the commit may make it look like _"an easy fix"_ as with #535, that it took me the best part of 3 hours to figure out the 4 lines that actually needed to be modified...

I sure hope this is the last of the high contrast complaints, because both #535 and this one were way more time consuming to fix than is apparent to the eye... ;)

closed 2015-09-20 by pbatard

cookieboyeli, 2015-09-21:
Good lord I will not be assuming anything is easy ever again. haha Thanks for putting up with Windows High Contrast. I am very appreciative of the time you dedicate to making such a stellar project.

Pbatard, you should reconsider your stance on donations! I have just reconsidered my stance on 100% total ad blocking, and will be making your page my only exception. :)

pbatard, 2015-09-21:
> This puts your open issues number below 20 for the first time in a quite a while

Yeah, I'm quite happy about this though (even as there were a couple of minor issues that I brushed under the carpet to bring that number down).

>  ever think of reconsidering your stance on donations?

I don't think so.

While the main reason I am not accepting donations is political (because I really don't think the donation system works, as long as most, and especially the companies who benefit a great deal from it, seem to think that it's perfectly okay to have mostly the little people foot the bill), I also [point out](http://rufus.akeo.ie/#donate) that I am fortunate enough right now to have enough income to cover my needs. While this may change in the near future (in which case I'll probably just add an extra ad on the page or something), it only sounds fair to me that, if people feel generous enough to want to donate money, they should donate it to causes that need it a lot more than I do especially when there never is a shortage of those... 

Or, you can always [donate to the FSF](https://my.fsf.org/donate), who have been fighting hard for the last 30 years, to make the development of software like Rufus possible.

commit 7618913c (2015-09-20): [ui] more high contrast fixes

* Closes #592