pbatard/rufus #593: Can't see my USB flash drive in Rufus 2.3

LuckyOne777, 2015-09-06:
Hi. I can not find my SanDisk Extreme 64Gb in Rufus 2.3.
Log:
Rufus version: 2.3.709
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 3.0 device 'SanDisk Extreme USB Device' (174C:3074)
Device eliminated because it was detected as an USB Hard Drive (score 8 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found

LuckyOne777, 2015-09-06:
[image]

pbatard, 2015-09-06:
Thanks for the report. This will be fixed in the next version.

In the meantime, and __as pointed out in the log you pasted__, you can still get your device listed by clicking on the white triangle and enabling "List USB Hard Drives" (or by following the steps documented [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected))

commit 7ad71a34 (2015-09-06): [core] fix detection of some SanDisk Extreme UFDs

* Also add a debug mode for troubleshooting user reports of HDDs vs UFDs
* Closes #593
closed 2015-09-06 by pbatard