pbatard/rufus #597: Windows 10 "requesting disk access" operation fail

NoNameProvided, 2015-09-15:
I am using Windows 10 (I am on the insider ring) and when I try to use rufus it cant get exclusive access for the selected device.

```
Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Could not get exclusive access to device \\.\PHYSICALDRIVE1: [0x00000005] Access is denied.

Found USB device 'Kingston DataTraveler 3.0 USB Device' (0951:1666) [GP]
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 956, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {11405999-96F6-491C-923C-018564E35E26}
Max parts: 128, Start Offset: 17408, Usable = 7866546688 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {1C928F66-D011-4ADD-9E27-24200456AA18}
  Size: 7.3 GB (7865515520 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2015-09-15:
Many people (including myself) are using Rufus on Windows 10 without encountering this problem, and some people have been reporting getting this issue on previous versions of Windows, so I can tell you that this has nothing to do with using Windows 10.

In most cases, this is due to installed software, such as Antivirus or other security solutions trying to keep exclusive access to a drive when they really shouldn't. Rufus considers that, if any application is holding access to a drive, it will not format it, hence the check for exclusive access.

Are you using any of the applications listed [here](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) by any chance?

NoNameProvided, 2015-09-15:
No, I don't use any of those programs but this issue should be closed, because I noticed I cant even remove any pendrive from my system. As soon as I plug one in, something start to hold in until I not plug it out.

pbatard, 2015-09-15:
This is indicative of an external application taking hold of the drive indeed. Also, the list I have is __not__ exhaustive. I will close the issue, but if you ever find the app that is keeping access to your drive, don't hesitate to update it.

closed 2015-09-15 by pbatard

divmgl, 2015-10-31:
This is happening to me as well on Windows 10. Have closed almost all apps and have only system processes running.

Tusticles, 2015-10-31:
Same issue here

divmgl, 2015-10-31:
Can confirm this is happening on build 10565. @Tusticles any solutions so far?

Tusticles, 2015-10-31:
LOL, a restart solved it!

divmgl, 2015-10-31:
@Tusticles damn, I already restarted. Let me try one more time.

pbatard, 2015-10-31:
Issues with build 10565 are duplicates of #605.
Please remember that Windows Insider previews are __BETA__ versions of Windows, and they __DO__ contain bugs that can impact applications like Rufus. It is for Microsoft to fix these bugs.

divmgl, 2015-10-31:
@pbatard thanks, I figured it was a bug with Windows. thanks!

pbatard, 2015-10-31:
Note that you can try <kbd>Alt</kbd>-<kbd>,</kbd> in Rufus, to try to work around the problem. But I don't consider disabling exclusive locking as a "proper solution", as it could have very detrimental effects if another application is trying to access the USB.

imverylame, 2015-11-07:
[alt] + [,] also worked for me on Win 10 Pro 1565

pbatard, 2015-11-07:
@imverylame, can you please upgrade to 10586 (the latest Insider build) and report if that also fixes the issue (without having to using <kbd>Alt</kbd>-<kbd>,</kbd>, which is there to __work around__ what is a pure Microsoft __bug__)?

NoNameProvided, 2015-11-18:
I can confirm. On build 10586 everything working well. USB devices are lockable by Rufus and removable from the system via "safely eject USB drive".

pbatard, 2015-11-18:
Yeah. Windows 1511 (a.k.a. 10586) has been out for a few days now, and I have yet to receive a complaint about Rufus having issue with exclusive locking on that version, which tells me that the Windows bug was definitely fixed there.

The one residual report I've seen on an access issue with 10586 appears to be due to one of the many applications that seem to try to get a lock on a drive, without ever relinquishing it, and is unlikely to have anything to do with the Microsoft bug.

At any rate, thanks for reporting!

Tuumke, 2016-07-08:
Running on 10586, still having the issue :(
Even on rufus 2.9.934

fgolemo, 2016-09-10:
I was having this bug in Rufus 2.11.995 on Windows 1607, build 14393.105. But the Alt-, trick worked.

zitzutest, 2016-10-21:
Trying to use Rufus but having this issue. I had defraggler opened on my PC (it wasn't doing anything, just sitting there). As soon as I closed Defragglef, Rufus happily burned ISO file in to  my USB.

Just my 2 cents.
F.

pbatard, 2016-10-21:
FYI, Defraggler is listed in the FAQ as [software Rufus is incompatible with](https://github.com/pbatard/rufus/wiki/FAQ#Software) for precisely that reason.

BNolet, 2016-11-01:
I had this same issue and reading through the comments I discovered that it was Paragon software that wasn't allowing me access to the USB device. The Paragon Mounter service to be exact. Stopped the service, then removed and replugged the USB and everything worked fine and dandy.

pbatard, 2016-11-01:
Thanks for the report. I'd like to add the Paragon software you mention to the [list of incompatible software](https://github.com/pbatard/rufus/wiki/FAQ#Software) but I don't know the actual name of the software the mounter service is part of.  Is it Paragon Hard Disk Manager? Drive Copy? Partition Manager? HFS+ for Windows? Something else??

BNolet, 2016-11-02:
The software I was using was "Paragon ExtFS for Windows" found [here](https://www.paragon-software.com/home/extfs-windows/). It's possible that HFS+ for Windows will have the mounter service as well considering the nature of the software.

pbatard, 2016-11-02:
Thanks. I agree that HFS+ probably has the same behaviour, so I am now listing both products in the incompatible software.

BNolet, 2016-11-03:
Wonderful :) Glad I could help!

NeAmico, 2016-11-22:
You still can have Paragon EXTFS software installed while using Rufus.
The trick is to stop the ParagonMounter Service and kill corresponding processes.
While in Services management go to process details and kill the damn thing (kill process tree) and then stop the Paragon Mounter.

Kynolin, 2017-02-26:
I just installed Paragon ExtFS so I could write an ISO from an EXT4 USB to Windows HDD to a USB drive via Rufus without having to copy it through a VM first.  Same issue. Stopping ParagonMounter (service) and unplug/replug the USB I was writing to did the trick. Rufus was then able to write no problem.

To help the Googlers:
"requesting disk access" "rufus" "windows 10" "paragon extfs"

pbatard, 2017-02-27:
Paragon is mentioned in the FAQ, in the [list of incompatible Software](https://github.com/pbatard/rufus/wiki/FAQ#Software). Remember, the FAQ may be lengthy, but it's really there to help you!

safwankdb, 2017-07-09:
quick formatting the usb drive solved it for me.

Macromeda, 2018-03-30:
the most famous reason for this is the readyboost, if it's enabled, disable it.
check this list [https://github.com/pbatard/rufus/wiki/FAQ#incompatible-software](incompatibles)

pbatard, 2018-03-30:
Thanks. I have added ReadyBoost to the list. However, I should point out that it should be fairly obvious that if you use ReadyBoost on the drive you are trying to format in Rufus, you will get a disk access conflict.

drydenmike, 2018-07-26:
I have the solution for this issue.
Just delete de usb partition(don't create any), after that eject it and put it back in. Now rufus will be able to do it's thing.
I used minitool partition wizard, but i'm sure others work too.

ibuprophen1, 2018-07-26:
Also, in some cases, the manufacturers provide software to accompany their drivers that does tend to take control of a device that can interfere with others like Rufus to do its thing.

BNolet, 2018-08-01:
Late I know, but @NeAmico that's exactly what I said I did