pbatard/rufus #598: "Windows to go" not offered under Windows 8.1

spacepluk, 2015-09-21:
Hi,
I'm trying to create a "Windows To Go" disk but the option doesn't appear.  I'm running Rufus 2.3 on Windows 8.1 in a Virtualbox and I created the iso installer using Microsoft's media creation tool.  What am I missing?

Thanks.

pbatard, 2015-09-21:
Care to provide a __full log__?

I added some provisions in latest release to prevent Windows To Go from being offered when the generated WIM image is incompatible with Microsoft's own APIs, which is what Rufus uses for Windows To Go (and yes, Microsoft seems to have managed to create WIMs that can't be processed even using the latest Windows 10 WIM APIs).
The official ISOs are not affected, and will provide the To Go option.
 
But the ISOs created with the Microsoft tool, are likely to be.

__HOWEVER__, there should be a line in the log that states:
```
Note: This WIM version is NOT compatible with Windows To Go
```

If you see that message, you will need to use the official Microsoft ISOs.

spacepluk, 2015-09-21:
Hmm, I see I'll try to find the official ISO somewhere and try with that (this computer came without disks :/).  Here is the log:

http://sprunge.us/WXgC

pbatard, 2015-09-21:
Hmmm, does your image have an `install.wim` file in the `sources\` directory?
If not, or if it has an `install.esd`, then it it not compatible with the Windows To Go mode of Rufus, since I don't know of any official Microsoft published ISOs that have an ESD. In that case this issue is a duplicate of feature request #586, which I'm not planning to add at this stage.

spacepluk, 2015-09-21:
I just checked, and there's no `install.wim` but there is an `install.esd`.  I created the iso files using Microsoft's media creation tool, I guess they don't work for this.

Now I tried downloading the official iso from the link in the other issue but I don't have permissions (subscription) to get it...  When I got the laptop I wiped the disk and installed linux right away, so I think I will have to restore my dd backup and activate windows there before I can do this :(

Thanks for your help.

closed 2015-09-21 by spacepluk

pbatard, 2015-09-21:
Yeah, I wish there was as easy a way to download Windows 8 official ISOs as there is for Windows 10 (through [Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench)) but short of having a subscription to the MSDN, I'm not aware of one.

Your best bet is probably to go for a torrent, and confirm that the SHA1 of the ISO matches with the one you can [publicly check from the MSDN](https://msdn.microsoft.com/en-us/subscriptions/downloads/#searchTerm=&ProductFamilyId=545&Languages=en&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0) (even if you don't have an account - just click the _Details_ link of any ISO to get the official SHA1).

spacepluk, 2015-09-21:
Thanks, I'll try that.