pbatard/rufus #603: External Hard Driver Problem

Narroto, 2015-09-25:
Hello 
i have External Hard Driver and want to burn window 10 , the problem is that i  split it into 2 partition (as in photo)
[image]
now i want to burn the windows to the J  but Rufus will format the whole  External Hard ^^ 
so can i burn windows to J and keep the K without format it ?
[image]
Thank you

blackcrack, 2015-09-25:
well, want be maybe not bad if be able use single partitions on usb-devices, 
to have the possible for have a "save"-data-partition on an usb-stick too :)
f= partition who need to format and dd(copy,move or how ever) 
the iso=800/4G/8G and the other space on the stick or 
external usb-device as data/backup space as 2nd partition ..

managing it maybe with mini-partiontool or easus partioning ..
but the drive or single partition format able and managable with rufus..
would maybe not a bad idea /suggestion ..

best regards
Blacky

Narroto, 2015-09-25:
you mean i can format the partition i want to burn windows in with mini-partiontool so that the Rufus can read it ?

if i understood right can you tell me how :)
is it hard to explain i think i will wait until Rufus  can read the External hard as single drivers maybe next update ^^ 

Thank you so much :D

pbatard, 2015-09-25:
> the problem is that i split it into 2 partition

This is __NOT__ supported. Rufus is designed to format the whole drive and remove all partitions, as it would be a major headache for reliability to do otherwise.

Please see [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiple_partition_support).

closed 2015-09-25 by pbatard