pbatard/rufus #604: Rufus 2.3 missing Win2Go (2.2 works, 2.4 beta no)

escape75, 2015-09-25:
Windows to go option is missing (Same PC, same ISO, same USB stick)
in Rufus 2.3 however 2.2 works fine. Also tried 2.4 beta and it doesn't work.

[image]

pbatard, 2015-09-25:
Yes, this is a [__feature__](cc108213611febd63c58d7ac77c1f7a1d686bca5) that was added with Rufus 2.3 to not provide the Windows To Go option for ISOs that aren't actually compatible with it. If you try to create a Windows To Got drive with 2.2, you should get an error, and if you look at the log from 2.3 or later, you're likely to see something like:
```
  Uses Install.wim: Yes (version 0.14.0)
  Note: This WIM version is NOT compatible with Windows To Go
```

If this isn't the case, feel free to reopen this call

pbatard, 2015-09-25:
Oh, and I'll just add that if you want __actual__ Windows 10 To Go compatible ISOs, you should download them from [Tech Bench](https://www.microsoft.com/en-us/software-download/techbench), as it's only the ones you create with the Windows 10 download tool that seem to be incompatible.

closed 2015-09-25 by pbatard

escape75, 2015-09-25:
Yes, you are correct - I got an error. Thanks!