pbatard/rufus #605: Could not open media Exception seems to have returned in - 2.3 Version

CreativelyMe, 2015-09-27:
I am attempting to create a Bootable USB with a new Windows 10 iso.(This is a different [new] copy of Windows 10 than the upgraded version on the workstation's Windows 10).  

The process seems to be stuck at requesting disk access...  

Here is the error message.

Error:  Could not open media.  It may be in use by another process.  Please re-plug the media and try again.

I am using Windows 10 Pro.  I tried to run Rufus as a Administrator.  I also tried using the new Beta version in case this had been addressed in the beta.

Here is the log:  I hope this helps.

Found USB 2.0 device 'PHD 3.0 Silicon-Power USB Device' (174C:55AA)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as an USB Hard Drive (score 16 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'PNY USB 3.0 FD USB Device' (154B:FA05)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 15545, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 119.1 GB (127848153088 bytes)
  Start Sector: 32768, Boot: No, Recognized: Yes
Found USB 2.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0749)
NOTE: This device is an USB 3.0 device operating at lower speed...
Found USB 2.0 device 'PHD 3.0 Silicon-Power USB Device' (174C:55AA)
NOTE: This device is an USB 3.0 device operating at lower speed...
Device eliminated because it was detected as an USB Hard Drive (score 16 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB 3.0 device 'PNY USB 3.0 FD USB Device' (154B:FA05)
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 15545, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 119.1 GB (127848153088 bytes)
  Start Sector: 32768, Boot: No, Recognized: Yes

Thanks

Joe (Creatively.Me)

pbatard, 2015-09-27:
Thanks for the report.

But for something to have returned, then you must be able to demonstrate that you don't get the same issue with older versions. So, have you tested older versions of Rufus on that same computer (if so which ones) and demonstrated that they weren't affected? Or is it just that you haven't been using Rufus in some time, and are now seeing this error, with a new version of Rufus, on a new version of Windows?

From all the other reports I have seen, the `may be in use by another process` error is exactly that, and has nothing to do with Rufus. As opposed to other applications, Rufus will request exclusive access to a drive, so if any other application is using it, you will get this kind of error, and you will need to find out which __external__ application is the culprit. I could for instance be one of [those](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) (and this list is by no means exclusive).

For the record, I've been trying to see if there was any possibility to report which application is having hold of the drive back to the user (#312), but Windows makes that very difficult to do for whole drives, so I'm afraid you may have to figure it out on your own.

mawelsh, 2015-09-29:
To add to this, I'm seeing media access errors across all 3 of my machines that I've updated to the latest Windows 10 10547 (Threshold 2 RTM) builds.  It still works on my Win 10 non insider machines, and Win 8 machines.  On at least 1 of the 3 machines updated, I know I had successfully used 2.3 hours before updating with no changes in applications and the other 2 had been used previously but not quite so immediately.  I'm going to run an update/rollback on of my Win 10 box's that's not currently an insider preview participant to see if the behavior persists.

pbatard, 2015-09-29:
Okay.

Be mindful that these releases are tagged _insider_ for a reason: they may very well inadvertently break existing capabilities, which, once reported, Microsoft will fix back. Heck, for a long time I thought Windows 10 had completely broken the Windows To Go feature of Rufus... but then Microsoft fixed the To Go feature shortly before RTM...

Oh, and as always, I'm not adverse to see a __FULL__ log from Rufus of when you encounter the issue. I'd also be interested to know if you tried with more than one USB drive...

pbatard, 2015-09-30:
I've been getting additional reports that Microsoft appears to have broken their disk partitioning services with Windows 10 insider build 10547.

For instance, if you use `diskpart` and simply try to issue a `clean` command on your USB drive, you will get the same error. This seems to confirm that the issue has nothing to do with Rufus.

I'm trying to get one of my platforms on the insider builds, but no luck so far... At any rate, if this is a global issue with disk partitioning, I'm pretty sure Microsoft will address this in a future release.

ThiloL, 2015-09-30:
We will see, if Microsoft changes something to make it work again.

pbatard, 2015-10-01:
Finally got build 10547 and was able to confirm the issue.
The strange thing is, on the platform I tested, I had no issue issuing a `clean` for that drive using `diskpart`...

By the looks of it, starting with build 10547, Microsoft broke the [`FSCTL_LOCK_VOLUME` API](https://msdn.microsoft.com/en-us/library/windows/desktop/aa364575.aspx) (at least for USB drives), as our inability to get that lock appears to be the source of the error code.

For the record, Rufus requests a lock to the drive to be sure that no other application is using it, as well as an additional safety measure to ensure that, even if something goes wrong, a system drive will not be reformatted, as such a drive would already be locked. Of course, one workaround would be to __not__ lock the physical drive before repartitioning, but before I go there, I'd rather make sure the current locking situation is not due to a bug from Microsoft, where they updated their USB handling and forgot to ensure that their lock/unlock code was properly balanced. I'll see if I can report a bug for this when I get a chance...

ThiloL, 2015-10-01:
We have to wait for the fix from microsoft. That rufus locks the drive is the right way.

—Thilo.

pbatard, 2015-10-01:
Definitely looks like a widespread bug for the latest insider builds.

If you look at the Feedback reports for "Hardware, Devices and Drivers" &rarr; "USB devices and connectivity", you see a flurry of reports from people stating that they can no longer use "Safely eject USB" due to always getting "media is in use". This is consistent with the issue observed with Rufus and seems to have started with build 10532.

As there are more and more people complaining about this, I'm pretty sure Microsoft is aware that they have a problem and will hopefully fix it soon.

Oh and for the record, I am also consistently getting the `diskpart` + `clean` issue on my 10547 platform, even after full reboot.

CreativelyMe, 2015-10-02:
I was able to use Rufus 2.4 to create a bootable USB drive with Windows 10.

I was able to do this after an update to Windows 10.  The same update seems to have fixed an ongoing issue where it was never save to remove a USB from the machine even after closing all apps that may have been accessing the drive.

I wanted to take a moment to thank you for being so responsive to the issue.

Thanks

Joe (Creatively.me<http://Creatively.me>)

pbatard, 2015-10-02:
I got the latest Defender update as well, and while it seemed to have "fixed" the issue when I first tested after applying that update, things looked good, but then the problem came back again, and disappeared again after a while, so I don't think it's entirely been fixed.

Considering that the latest Defender updates seem to have had some sort of effect (though officially these were only virus definitions), I'm wondering if Defender might not be the application keeping a lock. I haven't tried disabling Defender altogether yet, so that's just a wild guess.

Still, it looks like we'll have to wait for Microsoft to properly fix this...

pbatard, 2015-10-13:
I just got insider build 10565 pushed on one of my test machines, and while it __seems__ to improve things a little (as well as come with a new icon for USB safe ejection - Microsoft are __definitely__ fiddling with something here), it does not resolve the problem altogether.

What I observed was the following. First run of Rufus from cold boot is usually fine. But then if you eject the USB and plug it back again, the old Access Denied from Rufus/Cannot safely eject from Windows comes back again. Oh, and you can also trigger this kind of behaviour without using Rufus altogether, if you play with `diskpart`.

So, yeah, some small progress, but you'll have to try a bit harder, Microsoft!

Maybe the next build...

WynonaRaines, 2015-10-19:
I tried Rufus 2.5 Beta2 and it didn't work, PBatard. :( I still get the Requesting Disk Access and then I'm told it can't do it because something else must be using it. 

Should I give Microsoft feedback on this?

WynonaRaines, 2015-10-19:
Oh, phooey, I forgot to press Alt+, . . . Sigh . . .

WynonaRaines, 2015-10-19:
Yay! Thank you, Sir! 

Using the Beta, I first plugged in the Jump Drive, set Rufus up, pressed and held Alt plus the comma, let 'em go and then clicked on Start! And away she flew! Bootable Jump Drive with Windows 10 Build 10565 on it.

pbatard, 2015-10-20:
Glad to hear it worked for you. And yes, you need to press <kbd>Alt</kbd>-<kbd>,</kbd> simultaneously. Then you will get a message at the bottom saying _"Exclusive USB drive locking disabled"_.

I think Microsoft need to get feedback about this, if they haven't already, but preferably when trying to use `diskpart` + `clean` on the USB or some other native Microsoft tool, as they may be more receptive to properly fix this then... :wink:

WynonaRaines, 2015-10-20:
I will definitely give feedback on the Jump Drive issue and will draft a couple of friends to check out the diskpart + clean issue so that when we all send feedback, I'll at least be able to give an intelligent report. :)

Thanks again for all your hard work to get things going for us in spite of Microsoft. :)

tairun, 2015-10-26:
Beta3 solved the problem for me, no reboot required (on ver. 10.0 Build 10240)

pbatard, 2015-10-26:
> on ver. 10.0 Build 10240

I don't understand. Are you using build 10240 (which is unaffected). If not, what is your build number?

tairun, 2015-10-26:
Yes I'm using Build 10240, but I was affected. Did a reboot, no luck. I used task manager to kill as many user processes as possible. Still no luck. Then I just downloaded 2.5_BETA3 and it worked.

pbatard, 2015-10-26:
> but I was affected.

Then this was a __different__ problem. There was absolutely no fix for exclusive locking issues between 2.4 and any of the BETAs.

tairun, 2015-10-26:
Ok sorry, should i delete the comments?

pbatard, 2015-10-26:
Nah that's okay. Just be mindful that if an issue mentions something like _"with Windows 10 Insider Build 10547"_, if you are seeing a similar problem, but aren't running that version of Windows, it's probably something else.

If you do a search, you'll see that there are many other reports of drive locking with Rufus, that generate the kind of error you saw. However, this issue is specific to the Insider builds, as Microsoft broke locking altogether there.

tairun, 2015-10-26:
Will do, just thought there might be a additional Windows version with the same problem that is yet untracked.

pbatard, 2015-10-26:
Not really. 10240 is the retail build of Windows 10, and it was pretty explicit from the get go that this specific issue would not affect the retail build.

pbatard, 2015-11-06:
This seems to have been fixed in build 10586 at long last! I tested on a 10586 upgraded machine, and the issue no longer occurs.

I will therefore close this issue. If you are still seeing an issue after upgrading to 10586 or later, please let me know.

closed 2015-11-06 by pbatard

WynonaRaines, 2015-11-06:
Pete, I’m sorry to say it doesn’t seem to be completely fixed yet although it seems to be fixed for my installation.

 

Halasz is pretty knowledgeable and will probably cooperate; that’s why I snipped as much as I did.

pbatard, 2015-11-06:
Where can I find Halasz report?

Also, please be mindful that, regardless of the issue on Windows, some software and hardware, such as the one listed at the end of the [Rufus FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with) (__non exhaustive__ list) can produce the same error. In fact, even before the issue with Windows 10547 was reported, I used to get reports of similar issue about once or twice a month. Therefore, my expectation at this stage is that Halasz is simply one of the many users of an application that keeps an exclusive lock on USB media (which it shouldn't do), and he or she will have to identify what it is.

Of course, I'm also waiting to see if more 10586 users do report that the issue is still present for them, as this would be confirmation that Halasz findings are related to the previous bug rather than an external application.

WynonaRaines, 2015-11-06:
Page 10 here: http://www.tenforums.com/windows-insider/26141-device-currently-use-usb-jump-drive-10.html

 

I’ll go to the FAQ and see what’s there but I’m not a coder. L

pbatard, 2015-11-06:
Okay. Let's see if other people running 10586 report the same thing.

Be mindful that getting an error on safe ejection is very much something that can happen outside even in a non buggy release of Windows. For instance, you may get that in 10240 if you have a file open on the USB, or a disk utility application looking at the drive, or any other application that, for any reason, may have decided it needs access to the USB.

The whole point of safe ejection is precisely to detect if __any__ application is still using the drive, and produce an error if one does. I guess the one test to tell us if it's the 10547 Microsoft bug or something else would be to downgrade to 10240 and see if the issue still happens... Or we can simply wait to see how many people indicate that they still have an issue with 10586.

WynonaRaines, 2015-11-07:
We do have one member of TenForums who cannot get safe removal of her USB drive in RTM 10240. But she’s the only one I know of at this point.

lakshmandev, 2016-01-02:
We can use Sysinternal's <a href="https://technet.microsoft.com/en-us/sysinternals/processmonitor">Process Monitor</a> to find the list of process that are currently using the device. Killing those process would remove the handle on the USB drive and Rufus would have no problems in requesting for its requests a lock to the drive

pbatard, 2016-01-04:
Unfortunately, Process Monitor (and the current Windows APIs as a whole) are a bit too limited to be of much help... ProcMon may work if the issue is with an application having a __file__ open on the USB device, but isn't of much help if the application has requested access to the the __whole__ device.
You can try the following if you want to be convinced:
1. Download [HxD](http://mh-nexus.de/en/hxd/)
2. Launch it (as admin) and then go _Extra_ &rarr; _Open disk_
3. Select the USB device as follows:
[image]
4. Even though the disk was opened as read-only, Rufus will not be able to format it until you close HxD... and ProcMon will be of no help to tell you that HxD is the culprit for that.

In #312 I've been trying to see if it was possible to identify and report HxD-like processes that may have taken hold of the whole device on access error, because this is the fashion in which most of the applications that create an exclusivity issue will proceed, but I'm afraid that, even in Windows 10, there is __NO__ monitoring API in Windows that can tell you: _"Hey, applications Y, Z have requested access to disk X, and they have not relinquished that access yet"_.

All Windows will ever tells you is: _"Hey, something else is using that disk"_, but it can't tell you what, which really sucks...

ghendoaus, 2016-02-29:
I have exited explorer, stopped avg antivirus until reboot, and unplug re-plug the device - from elevated command prompt format the device - thus closing all open file handles to the device and then run rufus 2.7. I still get the dreaded cannot open device, I have tried a number of the alt-* keys including z, comma, r to try to clear anything that is accessing the drive all to no avail. 
Log:
`Rufus version: 2.7.855
Windows version: Windows 10 64-bit (Build 10240)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00357618
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8007974912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.00-13ubuntu3
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CDROM'
  Size: 276473856 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: No
  Uses Grub 2: Yes
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: EasyRE for Windows 8.iso
Downloading http://rufus.akeo.ie/files/grub-2.00-13ubuntu3/core.img
Downloading 'core.img' from http://rufus.akeo.ie/files/grub-2.00-13ubuntu3/core.img
Unable to access file: 404
http://rufus.akeo.ie/files/grub-2.00-13ubuntu3/core.img was not found - will use embedded version

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Volume is in use
Format error: The device is in use by another process. Please close any other process that may be accessing the device.
Re-mounted volume as 'G:' after error

Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0042B4B8
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8007974912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Found USB 2.0 device 'Lexar USB Flash Drive USB Device' (05DC:C75C)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 973, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0042B4B8
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.5 GB (8007974912 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
`

pbatard, 2016-02-29:
Unfortunately the log doesn't say much as to what might be holding the drive "hostage"...
Just in case, you also check if you are using any of the software [listed at the end of the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#List_of_software_and_hardware_Rufus_isnt_compatible_with)?

ghendoaus, 2016-02-29:
Im not using any of the software and I have tried with several brands of usb drive... It has only just started occuring so could be related to an AV change I will uninstall and test.,

pbatard, 2016-03-01:
> It has only just started occuring so could be related to an AV change

Interesting. I think you will indeed need to identify what has changed recently, to find the culprit. And if you do find the software that is preventing access, please make sure to report here.

HellVein, 2016-03-28:
I just made the program work (had the same problem) i just reconnected the USB drive a couple of time got Errors that the USB is Malfunctioning (has a problem) then pluged it to a DIRECT USB port to the Motherboard and the isue was GONE .
This is part of the log with the error and then working

Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 949, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x24E55E09
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.3 GB (7812939776 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GSP1RMCULXFRER_EN_DVD'
  Size: 3320903680 bytes
  Uses: EFI (win7_x64)
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_7_ultimate_with_sp1_x64_dvd_u_677332.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Could not get exclusive access to device \\.\PHYSICALDRIVE1: [0x00000005] Access is denied.

Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 949, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x24E55E09
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.3 GB (7812939776 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 949, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x24E55E09
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.3 GB (7812939776 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
0 devices found
0 devices found
0 devices found
Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 949, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x24E55E09
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 7.3 GB (7812939776 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'GSP1RMCULXFRER_EN_DVD'
  Size: 3320903680 bytes
  Uses: EFI (win7_x64)
  Uses: Bootmgr
  Uses: Install.wim (version 0.13.1)
Using image: en_windows_7_ultimate_with_sp1_x64_dvd_u_677332.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE1 for write access
Will use 'G:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Volume has an unknown Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing master boot record...
Drive has a Zeroed Master Boot Record
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume GUID \\?\Volume{fd07439c-2a89-11e5-824f-806e6f6e6963}\
Opened drive \\?\Volume{fd07439c-2a89-11e5-824f-806e6f6e6963} for write access
Writing partition boot record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted Volume{fd07439c-2a89-11e5-824f-806e6f6e6963}\ on G:\
Copying ISO files...
Extracting files...
Image is an UDF image

illogicalextend, 2016-10-08:
I had a lot of pain with this. Not sure what other program is locking my device.

Workaround is to download latest version v2.11, press alt + "," to disable locking and then run.

hiphopsmurf, 2017-02-21:
Formatting the drive to FAT32 first then disabling the volume locking was the only way I was able to get around this.

Kryptonit3, 2017-09-22:
open command prompt as admin, enter diskpart, then list disk, make note of USB disk number. then "select disk #" where "#" is the number from "list disk" command. then "clean". once done, "exit". then rufus works fine.

mad2math, 2018-01-01:
Hi guys .. I had the same problem but i tried another program called ISO to USB you van find in isotousb.com and its absolutly working!