pbatard/rufus #610: USB HDD's detected by default

passat1, 2015-10-03:
In Rufus version 2.4 behavior has changed and external USB HDD's are detected and displayed by default which could lead to formatting HDD by mistake!!!

pbatard, 2015-10-03:
That must be __specific__ to the type of HDD you use. I had to make adjustments due to Flash Drives not being detected, and isolating UFDs from HDDs is very tricky.

__PLEASE__ provide a __FULL__ log from Rufus. Without it, there's nothing I can do!

passat1, 2015-10-03:
Here are both logs, from ver. 2.3 and ver. 2.4
Both connected drives are HDD's (one is detected as such in v. 2.4 but there are some errors reported not seen in v. 2.3)
--------------------------
Rufus version: 2.3.709
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'ASMT 2105 USB Device' (174C:5136) [GP]
Device eliminated because it was detected as an USB Hard Drive (score 11 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
Found USB device 'ADATA CH11 USB Device' (125F:A11A) [GP]
Device eliminated because it was detected as an USB Hard Drive (score 8 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
0 devices found
-------------------------
Rufus version: 2.4.757
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Could not get node connection information (V2) for device 'USB\VID_174C&PID_5136\00000000000000000000': [0x00000001] Incorrect function.
Found USB 3.0 device 'ASMT 2105 USB Device' (174C:5136)
Device eliminated because it was detected as an USB Hard Drive (score 6 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
Could not get node connection information (V2) for device 'USB\VID_125F&PID_A11A\12058368091626000005': [0x00000001] Incorrect function.
Found USB 3.0 device 'ADATA CH11 USB Device' (125F:A11A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 121601, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB46AEDA6
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 931.5 GB (1000200994816 bytes)
  Start Sector: 2048, Boot: No, Recognized: Yes
---------------------------
Hope this will help

pbatard, 2015-10-03:
So ADATA are making USB HDD enclosures now?

I guess that means that the [offset I introduced with version 2.4](https://github.com/pbatard/rufus/commit/7ad71a34f85410f66c734300180853695ec14477) to force anything identified as `ADATA` to be detected as a Flash drive went a bit too far...

This is the reason why your HDD is being listed by default. Especially, the `Could not get node connection information` warnings from 2.4 have nothing to do with it, though I am interested in finding a bit more about your setup. Could you please provide an additional log from Rufus 2.4 with USB Debug enabled? To active USB Debug, just press <kbd>Alt</kbd>-<kbd>.</kbd> on your keyboard (you should see the message `USB Debug enabled` at the bottom of the Rufus screen and the log should have a little bit more data).

Anyway, I guess I'll remove `ADATA` from the list of manufacturers that provide only flash drives, which should fix your issue.

passat1, 2015-10-03:
I have this drive for almost 2 years. I've got it for very good price, it is USB 3.0 and performances were pretty impressive for that money at that time.
Here is log with USB Debug:

Rufus version: 2.4.757
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Could not get node connection information (V2) for device 'USB\VID_174C&PID_5136\00000000000000000000': [0x00000001] Incorrect function.
Found USB 3.0 device 'ASMT 2105 USB Device' (174C:5136)
Device eliminated because it was detected as an USB Hard Drive (score 6 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
Could not get node connection information (V2) for device 'USB\VID_125F&PID_A11A\12058368091626000005': [0x00000001] Incorrect function.
Found USB 3.0 device 'ADATA CH11 USB Device' (125F:A11A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 121601, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB46AEDA6
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 931.5 GB (1000200994816 bytes)
  Start Sector: 2048, Boot: No, Recognized: Yes
Processing Hub 1:
  Hub[1] = '\\?\renesas_usb3#root_hub30#5&39e9532&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[008]: USB\VID_174C&PID_5136\00000000000000000000
  Found ID[047]: USB\VID_125F&PID_A11A\12058368091626000005
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
  Hub[2] = '\\?\usb#root_hub#4&15dafd0c&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[075]: USB\VID_046D&PID_C312\5&3A5B01F9&0&1
  Found ID[226]: USB\VID_046D&PID_C046\5&3A5B01F9&0&2
Processing Hub 6:
Processing Hub 7:
Processing Hub 8:
Processing Hub 9:
Processing IDs belonging to USBSTOR:
  USB\VID_174C&PID_5136\00000000000000000000
  USB\VID_125F&PID_A11A\12058368091626000005
Processing Device: 'SCSI\DiskWDC_____WD3200AAKS-75L9A02.0'
  Non USB => Eliminated
Processing Device: 'USBSTOR\DiskASMT____2105____________0___'
  Matched with ID[008]: USB\VID_174C&PID_5136\00000000000000000000
  Matched with Hub[1]: '\\?\renesas_usb3#root_hub30#5&39e9532&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Could not get node connection information (V2) for device 'USB\VID_174C&PID_5136\00000000000000000000': [0x00000001] Incorrect function.
Found USB 3.0 device 'ASMT 2105 USB Device' (174C:5136)
Device eliminated because it was detected as an USB Hard Drive (score 6 > 0)
If this device is not an USB Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of USB Hard Drives in 'Advanced Options' (after clicking the white triangle)
Processing Device: 'SCSI\DiskWDC_____WD5000AAKS-07A7B01.0'
  Non USB => Eliminated
Processing Device: 'USBSTOR\DiskADATA___CH11____________AX00'
  Matched with ID[047]: USB\VID_125F&PID_A11A\12058368091626000005
  Matched with Hub[1]: '\\?\renesas_usb3#root_hub30#5&39e9532&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Could not get node connection information (V2) for device 'USB\VID_125F&PID_A11A\12058368091626000005': [0x00000001] Incorrect function.
Found USB 3.0 device 'ADATA CH11 USB Device' (125F:A11A)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 121601, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB46AEDA6
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 931.5 GB (1000200994816 bytes)
  Start Sector: 2048, Boot: No, Recognized: Yes

Thanks for quick repy

pbatard, 2015-10-03:
Thanks for the new log.
```
\\?\renesas_usb3#root_hub30
```
By the looks of it, you've installed the Renesas USB 3.0 drivers, instead of using the native USB 3.0 driver from Windows 8 (which would produce the output `\\?\usb#root_hub30`). Just so you know, while installing the manufacturer's USB 3.0 drivers was needed for Windows 7, it is no longer required for Windows 8 and later as Microsoft provides its own version.

I guess that explains the `Could not get node connection information (V2)` warning, as Renesas are somewhat known from not implementing all of the required features in their drivers. As I said, this is a benign warning, but now I have a better idea why you got it. Oh, and the reason you didn't get that warning in Rufus 2.3 is because part of the USB detection code was broken, and would query the wrong hub. This was fixed in 2.4.

I'll remove the extra skewing that was added for ADATA in the next version, which should restore the expected default listing in Rufus. Thanks for reporting this issue!

closed 2015-10-03 by pbatard

passat1, 2015-10-03:
And thanks to You for warning me about Renesas drivers. I'll go back to Microsoft drivers and check differences in behavior.
Thanks again

commit e1d0ab73 (2015-10-03): [usb] fix default listing for ADATA USB HDDs

* Also improve debug print statement in USB enumeration
* Closes #610