pbatard/rufus #613: "Logical drive was not found"?

rcdailey, 2015-10-08:
I'm trying to write Ubuntu 15 to my 16GB flash drive, and it hangs during the "Waiting for logical drive to reappear"... Logs below.

```
Rufus version: 2.4.757
Windows version: Windows 8.1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0409
Found USB device 'Google File-CD Gadget USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Google File-CD Gadget USB Device' (18D1:4E22)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Corsair Voyager 3.0 USB Device' (1B1C:1A03)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1949, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x285E4107
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16037969920 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
No drive letter was assigned...
Will use 'H:' as volume mountpoint
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (FAT32)...
Could not read volume name
Format error: Undetermined error while formatting.

Found USB device 'Google File-CD Gadget USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Google File-CD Gadget USB Device' (18D1:4E22)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Corsair Voyager 3.0 USB Device' (1B1C:1A03)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1949, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x28607378
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16037969920 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE2 for write access
No drive letter was assigned...
Will use 'H:' as volume mountpoint
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (MBR)...
Waiting for logical drive to reappear...
Timeout while waiting for logical drive
Logical drive was not found!
Formatting (FAT32)...
Could not read volume name
Format error: Undetermined error while formatting.

Found USB device 'Google File-CD Gadget USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Google File-CD Gadget USB Device' (18D1:4E22)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Corsair Voyager 3.0 USB Device' (1B1C:1A03)
NOTE: This device is an USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1949, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x28618303
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16037969920 bytes)
  Start Sector: 2048, Boot: Yes, Recognized: Yes
```

Note that when I open Disk Management tool in Windows 8, it shows the volume as RAW and won't let me delete the volume or modify it in any way. It seems my flash drive is in a weird state, so I'm not sure.

pbatard, 2015-10-08:
It looks to me like your flash drive decided to fail while Rufus was formatting it (or most likely before, but Rufus showed that it was failing).

In many cases, when the flash controller residing on the drive detects that there is an issue with the flash memory, it will set the drive to read-only (so that you can still attempt to read the data if possible), but of course then whatever state the drive was in will remain...

There's nothing that can damage your drive in the operation Rufus performs (yes, even reformatting or repartitioning a drive is no different for the flash as simply copying data to it). See [here](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive) if you want more details. But of course, if the flash fails during one of these operations, then the drive will be stuck in a weird state, like the one you are observing now.

If disk management cannot do anything with the drive, it looks pretty conclusive that you are faced with a hardware failure. If your drive is still under warranty, you might try to get it exchanged. Otherwise, you will have to recycle it, and purchase a new one...

pbatard, 2015-10-14:
Since there is not much more that can be done, I will close this issue.

closed 2015-10-14 by pbatard

rcdailey, 2015-10-19:
Sorry for the late response. Thank you for the details, I have tried with a different thumb drive and it is working perfectly. Something else must have damaged the drive I was using, so I'm not sure. Thanks again.