pbatard/rufus #614: Iso fails that uses a 2.88 floppy image

mpheyse, 2015-10-12:
Functional bootable ISO,  that boots from a 2.88 floppy image (Dos 6.22) fails to load in Rufus.

"This image is either non-bootable, or uses a it uses a boot or compression method that is not supported by Rufus..."

2.88 Floppy images as bootable cds are not that uncommon.

pbatard, 2015-10-12:
That's called an El Torito ISO, and these are __tricky__ to support properly, because you need to analyse the `AUTOEXEC.BAT` and `CONFIG.SYS` from the image and figure out how to merge the content so that it actually boots from USB. For instance, you'll find that a lot of these images expect `A:` as the system drive, whereas `C:` will be used when booting from USB (because the drive is seen as a HDD by the BIOS and not a floppy - unfortunately, this is not something that can be changed programmatically once DOS is booting).

Oh, and for the record, what you are requesting is a duplicate of #63. I'm planning to look into it, but don't expect anything fast...

pbatard, 2015-10-12:
Oh, and I'll add that, for the time being, you might be interested in [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Im_trying_to_work_with_an_ISO_but_Rufus_says_its_unsupported), which details how you can handle El Torito content manually (which is a good way to ensure the DOS application you are trying to use will actually work).

closed 2015-10-12 by pbatard