pbatard/rufus #615: Error: Could not find volume: [3] Unsupported

mnbf9rca, 2015-10-13:
I'm trying to create a UEFI bootable USB for a WinPE image to install Windows 7 Enterprise. I created an ISO from an NTFS-formatted bootable USB stick provided by the image maintainer (our supplier). The USB is recognised in UEFI boot, and the NTFS driver seems to be loading, then i get the following:
** UEFI:NTFS **
Loading NTFS Driver... DONE
Locating the first NTFS partition on this device... DONE
Looking for NTFS EFI loader...
  ERROR: Could not find volume: [3] Unsupported

Press any key to exit.

How can i debug this? I can't even see how it creates logs :)

pbatard, 2015-10-13:
This is with the [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs) module, so there is no logging (because logging wouldn't bring more information than what you see on screen).

It looks like this could be an UEFI firmware issue on your platform as the error indicates that [the call to `HandleProtocol()` from your computer's UEFI Boot Services failed](https://github.com/pbatard/uefi-ntfs/blob/master/boot.c#L329-L334). From the error, your UEFI firmware seems to be reporting that it doesn't support File System related protocols...

Can you provide more information about your target computer? What make and model is uit? Have you checked whether there was an UEFI firmware update that you could apply? How old is the latest UEFI firmware?

mnbf9rca, 2015-10-13:
It's an HP 9740m with the F.60 bios version (http://h20564.www2.hp.com/hpsc/swd/public/detail?sp4ts.oid=5271102&swItemId=ob_143542_1&swEnvOid=4060#tab4) - i can see a newer bios version but the release notes don't indicate anything relevant.

pbatard, 2015-10-13:
Okay. This is a fairly recent BIOS, so it's a bit puzzling that it would fail like this.

The one thing I'd also like you to clarify is this:

> I created an ISO from an NTFS-formatted bootable USB stick provided by the image maintainer (our supplier)

So you originally had a bootable USB, using NTFS as the file system.
You "somehow" converted that to an ISO (I guess you created a straight data disk from the USB content, right?).
And then you used that __custom non official ISO__ to create and USB again, using Rufus.

At this stage, I think I'd like to know what happens if you use an __official__ Windows 7 ISO from Microsoft (you don't need to install the OS from it, just see if it goes past the error you are getting right now).
Oh, and I'd be interested in seeing the __complete__ log from Rufus from how you create the USB from the ISO.

As you can understand, the only ISOs I can support are the official ones published by Microsoft, because there's just too much that can be introduced when creating custom ISOs.

So I hope you can get an official Windows 7 retail ISO. I'd encourage you to look [here](https://msdn.microsoft.com/en-us/subscriptions/downloads/default.aspx#searchTerm=&ProductFamilyId=350&Languages=en&FileExtensions=.iso&PageSize=10&PageIndex=0&FileId=0) first to locate the image you are interested in, then click `Details` to find the SHA1, and then lookup the SHA1 on google to see if you can locate a download, and then, after you downloaded the ISO, check that the SHA1 is correct in Rufus by clicking the `#` (hash) symbol at the bottom of the interface. This will ensure that you are using an officially supported ISO, and will tell us for sure if it is an UEFI firmware issue, or a custom ISO issue.

Or, if you don't want to look around, can I please ask you to download an official Windows 10 straight from Microsoft, through [Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench), and then see if it goes past your current error. Again, you don't have to install Windows 10 - this is just to see if it can boot properly on your system.

Once I have that info, I should be in a better position to investigate your issue.

mnbf9rca, 2015-10-13:
Yes, i know it's not an official official supported official build :)

i have a windows 7 ISO. I created a USB from this but this doesn't boot either. I can't seem to attach the file here so i've stuck it here: http://pastebin.com/TdaPYVSG

mnbf9rca, 2015-10-13:
to be clear though - this one doesn't even show up in the boot options.

pbatard, 2015-10-13:
> to be clear though - this one doesn't even show up in the boot options.

That's because you used MBR for your partitioning:

```
Partitioning (MBR)...
```

Can you try again, using the `GPT for UEFI` option?

mnbf9rca, 2015-10-13:
Doh! Am at work for the rest of the week will do it at the weekend.

pbatard, 2015-10-13:
No problem - take your time... :smiley:

pbatard, 2015-10-24:
Any update on this?

mnbf9rca, 2015-10-25:
Sorry, been swamped at work but i've got a few days off now. Main points:
1 I used tthe en_windows_7_ultimate_with_sp1_x64_dvd_u_677332.iso which I downloaded from MSDN.
2. I first created the USB as GPT/FAT32 and tried that. Log here: http://pastebin.com/B90LWtAR - result: I got the attached error on boot on my 9470m ("An error occurred while attempting to read the boot configuration data"). My HP 455 G2 booted it ok although windows setup never completely loaded (2 attempts).
3. Then i created the USB as GPT/NTFS and tried that. Log here: http://pastebin.com/P1VNAxd8 - result: I got the same error message as before (ERROR: Could not find volume: [3] Unsupported) on both the 9470m and the 455 G2
[image]

mnbf9rca, 2015-10-25:
BTW for the sake of completeness i repeated the test with MBR for UEFI and both FAT and NTFS and got the same results.

pbatard, 2015-10-27:
Thanks for the update.

> I first created the USB as GPT/FAT32 and tried that

If you used Rufus, then that's not possible (be mindful that Rufus will change your options __after__ you select an ISO). Rufus always formats to NTFS when using Windows ISOs, and indeed, your first log reports:
```
Formatting (NTFS)...
```
Oh, and Rufus also partitioned as MBR:
```
Partitioning (MBR)...
```

So, what you created in your "GPT/FAT32" step is an MBR/NTFS bootable USB, which booted using BIOS compatibility rather than UEFI. Not sure why you got the error you got, but it looks like Windows may not like your BIOS/BIOS compatibility module. In this mode, UEFI:NTFS is not used (but of course, then it's not USB bootable.

Note that you can use Rufus to create an empty GPT/FAT32 drive (make sure you unselect _"Create bootable drive using..."), and then __manually__ copy the ISO content (using 7-zip for instance) if you want to try UEFI boot from FAT32. But for various reasons, Rufus forces the use of NTFS for Windows installation media...

> Then i created the USB as GPT/NTFS

Confirmed by the log (which also confirms the use of UEFI:NTFS):
```
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
```

>  got the same error message as before

If you get the same with the official retail ISOs, and it works fine an other computers (so far you are the only report I got with this error from UEFI:NTFS), I suspect that this is an issue with the UEFI firmware from the 9740m.

Given that the UEFI:NTFS code is fairly straightforward (which of course doesn't meant that it's bug free! But it helps...) and the error you get (`unsupported`, which is something that comes from the firmware, not UEFI:NTFS), it really seems to indicate that the HP firmware is missing some features...

cookieboyeli, 2015-11-02:
I ran into a similar error a long time ago, would it be possible to create a cheat code to enable UEFI:FAT32 booting? Or are the "various reason" to much for it to even be implemented as a hidden option?

mnbf9rca, 2015-11-02:
"So, what you created in your "GPT/FAT32" step is an MBR/NTFS bootable USB"

The selections in Rufus were absolutely: GPT, FAT32. I have already noticed the upside down user interface which changes options further up after you select things down below :) - but why would it let me select it if it doesn't do it? - indeed, it did format as GPT/FAT32 - the actual log starts at line 538, sorry. I should have cleared first.

Is there any way to capture any more information in UEFI:NTFS? a debug build or something?

pbatard, 2015-11-02:
I guess I'll answer both of you at once... ;)

@mnbf9rca, you're right (but your log was a bit confusing). About 2 days ago, I found that I introduced a bug that enabled GPT/FAT32 for Windows starting with Rufus 2.4 (in commit 49e54904c23472732aa3fece5ee6135bc23398a8). So, unlike what was the case with Rufus 2.3 and earlier, it is indeed possible to create GPT/FAT32 with Rufus 2.4 and 2.5.

At this stage, I am waiting to see how much confusion this is likely to generate with regards to Windows installation. Especially, MBR+FAT32, which is now possible, will definitely result in people installing Windows in BIOS mode when they wanted UEFI, as well as the opposite, which I don't think is ideal because it might be way too late when they realize it. So I think I will disable "MBR+FAT32 for BIOS or UEFI-CSM" for Windows in the next version. But I will probably leave "MBR+FAT32 for UEFI" (which doesn't install a BIOS bootloader, so can only be booted in pure UEFI mode) and "GPT+FAT32" if I don't receive major complaints about these modes by them. I'm still half expecting that, by inadvertently enabling FAT32 for Windows, I'm going to receive notices that something gets broken during Windows installs, but the fact that nobody has alerted me to any of that ever since 2.4 was released is encouraging...

> I have already noticed the upside down user interface which changes options further up after you select things down below :)

Yup. I've been trying to find the time to work on this for quite a while (#117), but, as may not be apparent, it needs to be part of a much larger effort than "simply" moving some controls around. This means that getting a more logical order in the interface cannot happen before Rufus 3.0.

> Is there any way to capture any more information in UEFI:NTFS? a debug build or something?

The error code report is the most information you can get, as this is being provided by the UEFI firmware, and it's not possible to get more data unless you get a debug version of your UEFI firmware to explain why it complains. I don't have additional debug builds of UEFI:NTFS because it's already as close as it gets to debug. Because there really isn't much to it, a non "debug" version of UEFI:NTFS would probably only have one line that says `Booting...` with maybe an error code when it fails. The information displayed is about as much information you can get from the system at this stage...

I may however have a solution/explanation for the `An error occurred while attempting to read the boot configuration data`. From what I understand, as well as what I tested, this seems to happen if you are trying to boot Windows 7 __from USB 3.0__, because, unlike Windows 8 and later, Windows 7 does not know how to handle USB 3.0 by default. So what happens is, after the basic boot (which is done all through UEFI, and doesn't require anything special), Windows 7 switches to its internal bootloader, and attempts to read more data from USB... which it can't, because it's unable to talk to pure USB 3.0 devices by default. Oh, and even if UEFI:NTFS worked on your system, you would run into the same issue as with GPT/FAT32.

* __Solution 1:__ Boot from an USB 2.0 port... provided that you have one on your device (If all your ports are USB 3.0, this won't work. And using an USB 2.0 extension cable, or an USB 2.0 device, to try to force USB 2.0 mode won't work, since the problem is at the USB host controller level, which Windows 7 doesn't know how to initialize).

* __Solution 2:__ Find the Windows 7 USB 3.0 you need to install for Windows 7 boot, and insert them on your image. This is very dependent of the manufacturer of the USB 3.0 controller you use, and I can't really help you on how to do that (but if you look around, you may be able to find help from users who run into the same issue). Note that some manufacturers, like Intel, have started realizing that there was an issue for Windows 7 installation on recent USB 3.0 only machines, and are now providing [Windows 7 media update utilities](https://downloadcenter.intel.com/download/25476/Windows-7-USB-3-0-Creator-Utility). I believe that Intel tool is compatible with Rufus created media, but I haven't tested. And of course, if your controller is not an Intel one, you may have to do the whole drive insertion process manually.

* __Solution 3:__ Ditch Windows 7, and use Windows 8 or later. :smile: If you test GPT/FAT32 with the Windows 10 image I gave you a download link to, you should find that they probably boot on your system...

mnbf9rca, 2015-11-02:
yeah, i realised i'd configured it wrong and restared, and forgot to clear the logs, sorry!

i'll try with a Windows 10 DVD too.

Another option is to allow the user to specify the size of the FAT32 partition - i guess i can resize it manually later and add a large NTFS partition to put my 12GB WIM file on but it might be an option.

pbatard, 2015-11-02:
> Another option is to allow the user to specify the size of the FAT32 partition

Are you saying you have a requirement for >4GB `install.wim` as well? If so then I missed that from your earlier comments, and indeed, not being able to simply use UEFI:NTFS will be a pain in the ass...

If not, there's really no requirement for using NTFS for a Windows __installation__ media, as the filesystem used by the USB has absolutely no incidence on the filesystem Windows will be installed on. Also, Rufus is smart enough to detect >4GB `install.wim` and will force the use of NTFS then.

Outside of Windows To Go, which Rufus should support just fine by creating a FAT32 partition as required (provided the media is not REMOVABLE &mdash; see below) or UEFI:NTFS (which doesn't work for you, and can't be used for much but call on the regular Windows UEFI bootloader anyway), I'm not aware of a Windows installation media that has FAT32+NTFS partitions. Not that it may be impossible to create one, but one thing to consider is that Microsoft decided that Windows could never mount 2 partitions at the same time from an USB media, if that media had the REMOVABLE attribute, which is the case for 95% of USB flash drives out there. So this can make it quite tricky to produce drives with both FAT and NTFS on them, unless of course you manage to get your hands on a flash drive that has the FIXED attribute...

mnbf9rca, 2015-11-03:
yes, i have a 12GB install.wim. It contains drivers for about 6 different laptop models that we use.

Microsoft supports provides this example: http://blogs.technet.com/b/askcore/archive/2013/04/02/microsoft-deployment-toolkit-2012-update1-media-deployment-usb-drive-for-uefi-computer.aspx

pbatard, 2016-01-29:
Despite the extended silence, I'm still not giving up on trying to fix this issue.

Looking around at other UEFI applications seem to do (e.g. rEFInd), it looks like there may be cases where one needs to wait between the time a file system driver gets loaded, and the time where the corresponding file system can actually be accessed.

This got me thinking that this could very well be the problem with these HP systems: If they are slow to start the NTFS driver and identify the partitions that can be serviced by it, for one reason or another, and we try to open the NTFS volume right away, it could explain why we get an `UNSUPPORTED` error.

I have therefore added a 1 second delay in the latest UEFI:NTFS in the hope that this will help with this issue (coz if a system really needs more than 1 second to start being able to service a file system, you really need to get a faster CPU or get whoever designed its UEFI firmware too fix that).

If you feel like trying it, I have the new UEFI:NTFS `bootia32.efi` and `bootx64.efi` binaries available [here](https://rufus.akeo.ie/testing/uefi-ntfs.7z). You will however need to mount the UEFI:NTFS FAT partition to replace them in `/efi/boot/`...

nikon22, 2016-02-04:
Hi, i need to install windows 10 in my computer in ntfs because of the image size... I try but gives the error. I install with a «4GB windows 10 image and it work, but in ntfs no...

I have Asus x99a board. Nice program!

pbatard, 2016-02-04:
Can you try the [latest TEST version](https://rufus.akeo.ie/testing/rufus-2.7_TEST7.exe) of Rufus 2.7?
It contains the updated UEFI:NTFS and I'm very curious to see what happens if you use it.

If it doesn't work, please make sure you copy the full output from UEFI:NTFS and post it here. There are steps happening before trying to open the volume that may change depending on your setup, and I'd like to get the full output if you can.

nikon22, 2016-02-08:
Hi, i try on other computer with 2.7 test7 and ntfs with an image bigger than 4GB but dont work, it give the error that you can see on the attach file. 
When i burn with an image smaller then 4GB and choose fat32 it work.
[image]

pbatard, 2016-02-08:
@nikon22,

I'm gonna need to see the complete log (please do not truncate the log) from Rufus of how you created the USB drive with the >4GB image. Can you please provide that? Just click on the 'Log' button after you created the drive and copy/paste the full content here.
Also, please have a look at the USB and let us know if you see a `\efi\boot\bootx64.efi` file. Unless this file is present, your USB will never boot in UEFI mode.

nikon22, 2016-02-12:
Hi, i dont have conditions to do that. You can try, just burn a windows image bigger than 4GB in a pendrive and put computer boot from there ... If it work you arrive at language settings screen. If not you will see an error... Give news in this post if you can. Thanks

pbatard, 2016-02-12:
That's my problem - I do not see an issue on the computers I tried with.
be mindful not to assume that, because you see an issue, everyone will see the same.

The reason I asked for your Rufus USB creation log is precisely because I can't reproduce what you see, so unless you can recreate the USB as you did, and provide a log, there's not much I can do to try to address your issue, especially as the most logical explanation is that you are missing the `\efi\boot\bootx64.efi` file on your NTFS partition (which the log would definitely tell us if that is indeed the case).

nikon22, 2016-02-12:
I test on different computers and all give the error... When i find a time i will try to test on one computer since i work with different computers every day...

pbatard, 2016-02-12:
Thanks, I appreciate that. For the record, I tested with the one official Microsoft ISO I know of that has a >4GB `install.wim`, which was one of the Windows 8 developer previews (log below). This was created as GPT for UEFI and I found no issues booting the USB on both an AMD-APU based UEFI machine (that I know doesn't have an EFI NTFS driver in its firmware) and an Intel NUC (that does have a native NTFS driver, but I made sure I used the UEIFI:NTFS boot partition).

I do know however that some HP UEFI firmwares seem to have trouble with UEFI:NTFS, though I would expect a different error then than the one you got. Provided this is not an issue with the ISO (which the log should tell us) it would be nice to know the manufacturer and model of computers you tried with.

```
Rufus version: 2.7.848 (Beta)
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0809
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'rEFInd_0.8.7'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {2E7E48E1-E5B7-4138-AE21-9C1981CCE636}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {AF3CAFBC-7C3B-478F-8851-5F6DC321324C}
  Size: 14.9 GB (16012877312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'rEFInd_0.8.7'
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 1946, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {2E7E48E1-E5B7-4138-AE21-9C1981CCE636}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {AF3CAFBC-7C3B-478F-8851-5F6DC321324C}
  Size: 14.9 GB (16012877312 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'WindowsDeveloperPreview'
  Size: 5197092864 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: Yes
  Uses Bootmgr: Yes
  Uses EFI: Yes (win7_x64)
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: en_windows_8_developer_preview_x64.iso
Local time: 13099788678
Next update check in 13053 seconds.
WIM extraction method(s) supported: 7-Zip, wimgapi.dll
WIM apply method supported: wimgapi.dll

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE8 for write access
Will use 'F:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
[... 1740 lines cut ...]
Extracting: F:\sources\sxs\x86_netfx-sbs_sys_enterprisesvc_dll_31bf3856ad364e35_6.2.8102.0_none_e8b98f74b27e9f17\sbs_system.enterpriseservices.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-sbs_vsavb7rt_dll_31bf3856ad364e35_6.2.8102.0_none_1b3f9bfcfa3d6915\sbs_vsavb7rt.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-sbs_wminet_utils_dll_31bf3856ad364e35_6.2.8102.0_none_882b27a3126a7526\sbs_wminet_utils.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-sbscmp10_dll_31bf3856ad364e35_6.2.8102.0_none_fc96368b91a8e7ba\sbscmp10.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-scripting_engine_tlb_b03f5f7f11d50a3a_6.2.8102.0_none_3460d19855f689e4\microsoft.jscript.tlb (52 KB)
Extracting: F:\sources\sxs\x86_netfx-shared_netfx_20_mscorlib_b03f5f7f11d50a3a_6.2.8102.0_none_d3aa4456df7a2cdb\sbscmp20_mscorlib.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-shared_netfx_20_mscorwks_31bf3856ad364e35_6.2.8102.0_none_073ead5ed2fed4ec\sbscmp20_mscorwks.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-shared_netfx_20_perfcounter_31bf3856ad364e35_6.2.8102.0_none_d4e1270773c4f520\sbscmp20_perfcounter.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-shared_registry_whidbey_31bf3856ad364e35_6.2.8102.0_none_ee079fc4f3c2a312\sharedreg12.dll (13.3 KB)
Extracting: F:\sources\sxs\x86_netfx-shfusion__chm_b03f5f7f11d50a3a_6.2.8102.0_none_b12d1294b88e7458\shfusion.chm (99.1 KB)
Extracting: F:\sources\sxs\x86_netfx-shfusion_dll_b03f5f7f11d50a3a_6.2.8102.0_none_d9b110a24ad64ab9\shfusion.dll (113.3 KB)
Extracting: F:\sources\sxs\x86_netfx-shfusion_res_dll_b03f5f7f11d50a3a_6.2.8102.0_none_24e9607e4ac885b0\shfusres.dll (89.8 KB)
Extracting: F:\sources\sxs\x86_netfx-sos_dll_b03f5f7f11d50a3a_6.2.8102.0_none_92c87d147970373b\sos.dll (379.8 KB)
Extracting: F:\sources\sxs\x86_netfx-sys_enterpriseservices_tlb_b03f5f7f11d50a3a_6.2.8102.0_none_9aa2a9e3e0a8dd91\system.enterpriseservices.tlb (40 KB)
Extracting: F:\sources\sxs\x86_netfx-sys_windows_forms_tlb_b03f5f7f11d50a3a_6.2.8102.0_none_16b79694bc86c7a9\system.windows.forms.tlb (92 KB)
Extracting: F:\sources\sxs\x86_netfx-system_drawing_tlb_b03f5f7f11d50a3a_6.2.8102.0_none_6110d51911507118\system.drawing.tlb (8 KB)
Extracting: F:\sources\sxs\x86_netfx-system_tlb_b03f5f7f11d50a3a_6.2.8102.0_none_51659c8b142b9f09\system.tlb (80 KB)
[... cut ...]
Extracting: F:\sources\sxs\x86_netfx-vb_compiler_ui_b03f5f7f11d50a3a_6.2.8102.0_none_1a194f0c291ddcf4\vbc7ui.dll (185.3 KB)
Extracting: F:\sources\sxs\x86_netfx-vbc_exe_config_b03f5f7f11d50a3a_6.2.8102.0_none_b1555ecc8e5a60e6\vbc.exe.config (221 bytes)
Extracting: F:\sources\sxs\x86_netfx-vsa_codedom_tlb_b03f5f7f11d50a3a_6.2.8102.0_none_a27aea63b27c0630\microsoft.vsa.vb.codedomprocessor.tlb (6 KB)
Extracting: F:\sources\sxs\x86_netfx-web_engine_dll_b03f5f7f11d50a3a_6.2.8102.0_none_b577ff73b68334c5\webengine.

nikon22, 2016-02-16:
Hi, i do it with the new 2.7 stable and dont work again. It give the same error
On the file efi on the pen/boot only have the file bootia32.efi
Where i can see the log?

pbatard, 2016-02-16:
> Where i can see the log?

There's a 'Log' button on the interface. Just click on it __after__ creating the USB. You can then select the content to paste it, or save the log.

But really, if you only have `bootia32.efi` and no `bootx64.efi` file, then the problem has nothing to do with Rufus but is with the ISO you use. What is the name of your ISO??? Is it a Windows ISO? If so, is it for Windows __32- bit__ or __Windows 64-bit__? If it's not 64-bit then you will NEVER be able to install it on a 64-bit UEFI system...

nikon22, 2016-02-16:
The iso i use is windows 10 32x 64x with update 1. If i burn to pendrive in normal 64x mode it works fine, only in uefi dont work.
Its true that my tests were all done with same iso...
I cant find log button because in my language say Story...

Rufus version: 2.7.855
Windows version: Windows 10 64-bit (Build 10586)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0816
Found USB device 'EPSON Storage USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'TOSHIBA TransMemory-Mx USB Device' (0930:6545)
Found USB 2.0 device 'EPSON Storage USB Device' (04B8:08B9)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Kingston HyperX Fury 3.0 USB Device' (0951:16A5)
Found USB device 'EPSON Storage USB Device' (????:????)
Device eliminated because it appears to contain no media
2 devices found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7591, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {2456D71A-1539-478A-BD80-9EEE3DCE7525}
Max parts: 128, Start Offset: 17408, Usable = 62442273280 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {583D8AB6-4687-43C0-9A52-E943C937E9A1}
  Size: 58.2 GB (62441242112 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'w10aiov2pt'
  Size: 4959078400 bytes
  Has a >64 chars filename: No
  Has Symlinks: No
  Has a >4GB file: Yes
  Uses Bootmgr: Yes
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: No
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
  Uses Install.wim: Yes (version 0.13.1)
Using image: Windows 10 Update 1.iso

Format operation started
Requesting disk access...
Opened drive \\.\PHYSICALDRIVE5 for write access
Will use 'K:' as volume mountpoint
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Drive has a FAT32 DOS Partition Boot Record
Deleting partitions...
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Partitioning (GPT)...
Adding extra partition
Reserved 33 tracks (1 MB) for extra partition
Writing UEFI:NTFS partition...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
[... 1012 lines cut ...]
Found USB device 'EPSON Storage USB Device' (????:????)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'TOSHIBA TransMemory-Mx USB Device' (0930:6545)
Found USB 2.0 device 'EPSON Storage USB Device' (04B8:08B9)
Device eliminated because it appears to contain no media
Found USB device 'EPSON Storage USB Device' (????:????)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Sector Size: 512 bytes
Cylinders: 7591, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {9B5EB426-4400-4783-AF55-1F87F94CD8B9}
Max parts: 128, Start Offset: 17408, Usable = 62442273280 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {B112B7DD-271E-4843-ABCD-BB04C4D78D2A}
  Size: 58.2 GB (62440164864 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'UEFI:NTFS'
  ID: {F94FD118-D6BD-4050-A47D-2AD926334AEA}
  Size: 256 KB (262144 bytes)
  Start Sector: 121955495, Attributes: 0x0000000000000000

pbatard, 2016-02-16:
`w10aiov2pt` is a custom (__non official__) All-In-One image, and as per the following:
```
Extracting: K:\efi\boot\bootia32.efi (953.8 KB)
```
it is missing the required `\efi\boot\bootx64.efi`, so it is __NOT__ EFI 64-bit compatible.
There is nothing I can do in Rufus about this (especially as, again, this is an __unofficial__ ISO), and Rufus is not at fault, so you will either have to use an __official__ 64-bit ISO, or contact the makers of that ISO so they fix it.

For the record, you can legally download __official__ Windows 10 64-bit ISOs from [Microsoft's Tech Bench](https://www.microsoft.com/en-us/software-download/techbench).

And next time, please don't hijack an existing issue when the error you get is a different one. This makes it more difficult for people seeking help with error code `Could not find volume: [3] Unsupported` to find it.

nikon22, 2016-02-16:
Next time dont hijack? Why dont you tell me this earlier? I just answer your questions lol

pbatard, 2016-02-17:
> Why dont you tell me this earlier?

Because:
1. You didn't provide enough information initially to be positive that this was a different issue.
2. All that was needed to answer your questions after that was to see your USB creation log, so I didn't want to delete your posts before I provided you with a complete answer. And now that I have done that, I will delete this series of posts since they are unrelated to the original issue.

pbatard, 2016-02-17:
For anyone who experienced the `ERROR: Could not find volume: [3] Unsupported` error with Rufus 2.6 or earlier, please test with Rufus 2.7 and report, as it contains the latest UEFI:NTFS.

If I don't receive reports after one week, I will consider the problem resolved and close this issue.

mnbf9rca, 2016-02-18:
I now get this:
`*** UEFI:NTFS (64-bit) ***

Loading NTFS driver...
  Driver did not start: [1] Load Error

Press any key to exit.`

I tried with both Windows 7 64bit and Windows 10 64bit, with ISOs downloaded from MSDN:
W10: http://pastebin.com/k0vERANx
W7: http://pastebin.com/pjZXqJLj

pbatard, 2016-02-18:
Well, somehow I like that error better than the older one, because at least it tells us that your UEFI firmware doesn't like the NTFS driver we're trying to load from the get go, rather than produce a hard to troubleshoot failure later on. Also, this is the same error that was reported in #549, with, surprise, surprise, another HP laptop...

At this stage, this would look like an incompatibility with the HP UEFI firmware and the NTFS UEFI driver, so I would appreciate if you could do the following:
1. Download `uefi_test.iso` from [here](https://rufus.akeo.ie/testing/uefi_test.iso) (1.6MB). This ISO contains x86_64 and x86_32 UEFI shells, set as boot loaders, as well as the latest [efifs](http://efi.akeo.ie) NTFS driver, which is what UEFI:NTFS uses.
2. Use Rufus to create a GPT/FAT32 bootable USB out of it
3. Boot your HP laptop in UEFI mode using this USB. You should end up with a shell prompt<br/> (__IMPORTANT NOTE:__ This is a v2.0 shell, so it requires an UEFI firmware that supports it, which should be the case with relatively recent hardware. Older UEFI machines however may not boot this image as they only support v1.0 shell).
4. If you don't know what `fs#:` your USB drive is, use `map fs*` to locate it. This command will list all the fs drives, and if you look for something that contains USB there (e.g. `PciPort(0x0)/Pci(0x11,0x0)/USB...` you should be able to locate it. Or you can simply try `dir fs0:`, `dir fs1:`, etc... until you find something that contains `NTFS_X64.EFI` and `NTFS_IA32.EFI`
5. Assuming that you are running a 64-bit UEFI, issue: `load fs#:\NTFS_X64.EFI` to load the NTFS driver, with `#` being the number you got above (eg: `load fs1:\NTFS_X64.EFI`) and report if you see an error when doing that, and if so what error you get.
6. If you don't see an error, issue: `drivers` to list all the currently loaded drivers, and report if you see a line such as:
```
157  00000010  ? - -  -  -  efifs 0.8.0 ntfs driver (GRUB 2.02~  NTFS_X64.EFI
```

congnt92, 2016-03-10:
HI pbatard,
I have the same problem.
My laptop is HP probook 4420s. I'm try to create an USB multiboot, I run Rufus choose MBR BIOS/UEFI because GPT not support grub4dos.
After boot to USB, i get this error.
Can you give me some advices. Tks

pbatard, 2016-03-10:
Well, first of all, [I don't support multiboot](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_add_multiboot_support). So if you are trying to create a multiboot USB and it doesn't work, you're pretty much on your own. So, whereas you are indicating that you are using "MBR for UEFI" and playing with gru4dos, can I please ask you to test with "GPT for UEFI" (using an official Windows ISO if you have one)  for the time being?

Also, because I have to say I am always suspicious of people who jump on an existing issue, saying "I have the same problem", and that I'll try to help, only to find out it was a different error, can you please confirm that, the output you see from UEFI:NTFS is:
```
*** UEFI:NTFS (64-bit) ***

Loading NTFS driver...
  Driver did not start: [1] Load Error

Press any key to exit.
```

If it is, then please see my last post before yours and follow the steps I indicated, and report.
If not, then please post your output.

pbatard, 2016-03-18:
Any update on this? 

Unless someone follows the steps I posted earlier to try to get some more insight about this issue, I will have to conclude that it is no longer affecting people, and close it.

closed 2016-03-21 by pbatard

pbatard, 2016-03-25:
Just a quick note to indicate that [rEFInd](http://www.rodsbooks.com/refind/) recently uncovered a similar issue with HP firmwares, and seem to have a workaround for it. I will try to apply a similar fix in UEFI:NTFS (pbatard/uefi-ntfs#5) when I have a chance.

mnbf9rca, 2016-03-28:
thanks for still looking in to this. i simply havent had time to do the
test you asked for.

r.

pbatard, 2016-04-24:
For those interested, [this test version](https://rufus.akeo.ie/testing/rufus-2.9_TEST1.exe) of Rufus 2.9 contains the latest UEFI:NTFS, which includes the HP specific workaround and other stuff. I'd be interested to confirm that it indeed addresses the HP notebooks issue.

Michaelvas17, 2016-04-26:
Hi. Trying out the test of rufus 2.9. Do I need to select anything differently, or just do MBR for UEFI & BIOS + NTFS?

Thanks.

pbatard, 2016-04-26:
You need to select "GPT for UEFI", with NTFS as the file system of course. Don't try with "MBR for UEFI or BIOS-CSM", as this mode does NOT install UEFI:NTFS, since it doesn't boot in UEFI mode.

Michaelvas17, 2016-04-26:
It gives me an error saying the drive has to be fixed in order to use GPT. Should I use MBR for UEFI instead then?

pbatard, 2016-04-26:
The only thing I am interested in with this issue is whether the latest UEFI:NTFS improved support for HP laptops, so please unselect Windows To Go if you want to report on this, as this is a different problem. Or, if you select Windows To Go, you can try "MBR for BIOS or UEFI", but please check the log to make sure you see the message:
```
Writing UEFI:NTFS partition...
```

Michaelvas17, 2016-04-26:
I saw that when I ran MBR for UEFI. I'll let you know what happens.

Michaelvas17, 2016-04-26:
I can also test without Windows To Go if you'd like.

pbatard, 2016-04-26:
It's up to you. As long as you get the UEFI:NTFS partition created and make sure you do boot from that partition in UEFI mode, the results should be the same. Eliminating Windows To Go and using GPT for UEFI is one way to ensure that the USB can only be booted in UEFI mode, but I'll assume you know what you're doing even if you use a dual BIOS + UEFI bootable USB.

Michaelvas17, 2016-04-26:
Ok. I'll keep you posted.

Michaelvas17, 2016-04-26:
Sorry for the late response. I've gotten a different (but similar) error. 
[image]

pbatard, 2016-04-26:
Thanks for the update. I'm starting to get the feeling that HP may have added something specific to their UEFI firmware that prevents non approved drivers from loading or something. Could you please also provide the exact model of the HP laptop you've been testing with?

At this stage, I think I would also appreciate if you could do the following:
1. Download `uefi_test.iso` from [here](https://rufus.akeo.ie/testing/uefi_test.iso) (1.6MB). This ISO contains x86_64 and x86_32 UEFI shells, set as boot loaders, as well as the latest [efifs](http://efi.akeo.ie) NTFS driver, which is what UEFI:NTFS uses.
2. Use Rufus to create a GPT/FAT32 bootable USB out of it ('GPT for UEFI' with FAT32 as the file system)
3. Boot your HP laptop in UEFI mode using this USB. You should end up with a shell prompt<br/> (__IMPORTANT NOTE:__ This is a v2.0 shell, so it requires an UEFI firmware that supports it, which should be the case with relatively recent hardware. Older UEFI machines however may not boot this image as they only support v1.0 shell).
4. Issue the command `set FS_LOGGING 4`
5. If you don't know what `fs#:` your USB drive is, use `map fs*` to locate it. This command will list all the fs drives, and if you look for something that contains USB there (e.g. `PciPort(0x0)/Pci(0x11,0x0)/USB...` you should be able to locate it. Or you can simply try `dir fs0:`, `dir fs1:`, etc... until you find something that contains `NTFS_X64.EFI` and `NTFS_IA32.EFI`
6. Assuming that you are running a 64-bit UEFI, issue: `load fs#:\NTFS_X64.EFI` to load the NTFS driver, with `#` being the number you got above (eg: `load fs1:\NTFS_X64.EFI`) and report if you see an error when doing that, and if so what error you get. Also report the debug output you get (if any)
7. If you don't see an error, issue: `drivers` to list all the currently loaded drivers, and report if you see a line such as:
```
157  00000010  ? - -  -  -  efifs 0.8.0 ntfs driver (GRUB 2.02~  NTFS_X64.EFI
```

pbatard, 2016-04-26:
Oh, and let me add one more step to perform regardless of the outcome above:
8. Issue the command `drivers > fs#\drivers.txt`, where `#` is the number for your USB drive, and then copy/past the content of the `drivers.txt` you should find on your USB after plugging it on a Windows machine.

Michaelvas17, 2016-04-26:
OK, some issues. First and foremost, the laptop is a HP Elitebook 820. Secondly, I cannot boot into it with the drive due to the BIOS being on Safeboot, and I

pbatard, 2016-04-26:
> due to the BIOS being on Safeboot

Do you mean _Secure Boot_? In that case, UEFI:NTFS will never work. It needs Secure Boot disabled, and that could explain part of the issue (though I don't understand how it could start executing the UEFI:NTFS bootloader, which is not signed for Secure Boot either).

Or if HP really does have their own special security layer called "Safeboot", then that could also explain why the NTFS driver does not load...

Michaelvas17, 2016-04-26:
Haha, I meant Secure Boot. It's fickle. I've used a Ubuntu and several other Linux distros formatted drive on this and it worked fine. Booted up and I was able to use it without disabling Secure Boot.

But this entire time, I've been able to get around Secure Boot to get to UEFI:NTFS, so I don't know.

pbatard, 2016-04-26:
Ubuntu and other Linux distros use signed shims. UEFI:NTFS cannot do that because the NTFS driver is GPLv3 and Microsoft, who controls the signing of drivers and executables for Secure Boot, __refuses__ to sign anything GPLv3.

Unless your Secure Boot is completely broken, you will therefore __NEVER__ get UEFI:NTFS to work, unless you disable it (and I ca

Michaelvas17, 2016-04-26:
Good News! I was able to get Secure Boot turned off, and now it boots the USB.

OK let me do what you said, and I'll get back to you.

pbatard, 2016-04-26:
Thanks - I appreciate that!

Michaelvas17, 2016-04-26:
Weird thing. It seems to get stuck when I try steps 7 or 8 above. When I did 8 though, I noticed that the read/write light on the USB was on (it wasn't when I got to the shell)

Not sure if it's just taking too long or if it's getting stuck.

[40 more replies]